esphome/issues #3042: Slimmelezer Dsmr reader Home assistant entities are unknown

bubbamester, 2022-02-13:
### The problem

Hello,

i have just got the SlimmeLezer board, installed using the documentation, but could net get it working. In the Home Assistant all  of the entities have no value (Unknown), some are working, the ones that are not related to P1 port values (sensor.esphome_version_2, sensor.ip_address, sensor.uptime, sensor.wi_fi_bssid, sensor.wi_fi_signal).

Tried to reinstall the board with ESP Home through Home Assistant, no luck, then using the firmware from the zuidwijk.com website. Always got the same results.

### Which version of ESPHome has the issue?

2022.1.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.5

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
substitutions:
  device_name: slimmelezer
  device_description: "DIY P1 module to read your smart meter"
     
esphome:
  name: ${device_name}
  comment: "${device_description}"
  platform: ESP8266
  esp8266_restore_from_flash: true
  board: d1_mini
  name_add_mac_suffix: false
  project:
    name: zuidwijk.slimmelezer
    version: "1.0"
  on_boot:
    then:
      - if:
          condition:
            lambda: return id(has_key);
          then:
            - lambda: |-
                std::string key(id(stored_decryption_key), 32);
                id(dsmr_instance).set_decryption_key(key);
          else:
            - logger.log:
                level: info
                format: "Not using decryption key. If you need to set a key use Home Assistant service 'ESPHome:  ${device_name}_set_dsmr_key'"
 
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    ap_timeout: 15s
 
captive_portal:

#improv_serial:
 
# Enable logging
[... 191 lines cut ...]
[17:50:47][D][uart_debug:158]: <<< "1-0:3.8.0(00000018.937kVarh)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:4.8.0(00001436.860kVarh)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:M\x19\xE6\x83\x86\x06\x06\x06\x06\x06\x06\x16\x16\xE6\x96F\x16\xA6\xB6\xB3\v\x93CK\xD6\n1-0:6.8.0(00000006.996kVarh)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:7.8.0(00000423.185kVarh)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:8.8.0(00001013.675kVarh)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:15.8.0(00007070.771kwh )\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:1.6.0(004.852kW)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:1.6.1(004.852kW)\r\n"
[17:50:47][D][uart_debug:158]: <<< "\xCCK\xA6\x16\xE6f\xE6\x92\x86\x06\x06F\xE6v6f\xA6\xB6\xBBK\xD6\n1-0:2.6.0(003.940kW)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:2.6.1(003.680kW)\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-0:2.6.2(003.940kW)\r\n"
[17:50:47][D][uart_debug:158]: <<< ")\r\n"
[17:50:47][D][uart_debug:158]: <<< "!AE05\r\n"
[17:50:47][D][uart_debug:158]: <<< "/HLY5/D545-METER\r\n"
[17:50:47][D][uart_debug:158]: <<< "\r\n"
[17:50:47][D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[17:50:47][D][uart_debug:158]: <<< "0-0:96.13.0(FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF\xA34444444444444444\x944\x9C\xB444444444444\x944t44444444444444444\x944444\x9444444444\xF44t4444444444444444444444444"
[17:50:47][D][uart_debug:158]: <<< "44444444444444\x94\xB44444444444444444444444444444444444444444444444444444444444444444444\x9A444\x94\x94\x9E444444444444444444\x9A\xB444\x9A444444444444444444444444444\x94444444444"
[17:50:47][D][uart_debug:158]: <<< "444444444444444444444444444\x944444444\x9A\xB4444444\x944444444444444444444444444444\x9444444444444444444\x9444444444444444444444444444444444444\x94444\x94444444444444444\x94\x9444"
[17:50:47][D][uart_debug:158]: <<< "4444444444\xB444\x9444444444444444444444\xCA33\x1933\x193333333333333333333333333333333333333333333333333333333333333333\x193333333333333333\x193\e3333333333\x19333333333\x193333"
[17:50:47][D][uart_debug:158]: <<< "333333333FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF\xA3FF\xA3FFFFFFFFFFFFFFFFFF"
[17:50:47][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF\xA3FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[17:50:47][D][uart_debug:158]: <<< "FFFFFFFFFFFFF\xA3FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF)\r\n"
[17:50:48][D][uart_debug:158]: <<< "!80D0\r\n"
```

bubbamester, 2022-02-13:
[image]
[image]

bubbamester, 2022-02-14:
Dear Maurice,

thanks for the advice. First time I tried without the description key and I got the same results. But this eveningvI am going to try it again!

bubbamester, 2022-02-14:
Dear Maurice,

I've tired again. Deleted the description key but no success. Same results. Entities are unknown

[config_entry-esphome-2512a5dbc4380f6387fd7c223c0623b0.json.txt](https://github.com/esphome/issues/files/8061750/config_entry-esphome-2512a5dbc4380f6387fd7c223c0623b0.json.txt)
.

mmakaay, 2022-02-14:
Please try the following simplified config, in which I stripped a lot of things to be sure that there are no encrypt key things lingering in the device, in which I removed some options that used default values and in which the log level was increased a bit to possibly see some more output from the dsmr component:

```yaml
esphome:
  name: slimmelezer
  comment: "DIY P1 module to read your smart meter"
  platform: ESP8266
  board: d1_mini
 
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
 
logger:
  baud_rate: 0
  level: VERBOSE
 
api:
  
ota:
 
web_server:
  port: 80
 
uart:
  rx_pin: D7
  baud_rate: 115200
  rx_buffer_size: 3000
  debug:
    direction: RX
    after:
      delimiter: "\r\n"
    sequence:
     - lambda: UARTDebug::log_string(direction, bytes);
 
dsmr:
  max_telegram_length: 3000

sensor:
  - platform: dsmr
    energy_delivered_lux:
      name: "Energy Consumed Luxembourg"
      state_class: total_increasing
    energy_delivered_tariff1:
      name: "Energy Consumed Tariff 1"
      state_class: total_increasing
    energy_delivered_tariff2:
      name: "Energy Consumed Tariff 2"
      state_class: total_increasing
    energy_returned_lux:
      name: "Energy Produced Luxembourg"
      state_class: total_increasing
    energy_returned_tariff1:
      name: "Energy Produced Tariff 1"
      state_class: total_increasing
    energy_returned_tariff2:
      name: "Energy Produced Tariff 2"
      state_class: total_increasing
    power_delivered:
      name: "Power Consumed"
      accuracy_decimals: 3
    power_returned:
      name: "Power Produced"
      accuracy_decimals: 3
    electricity_failures:
      name: "Electricity Failures"
      icon: mdi:alert
    electricity_long_failures:
      name: "Long Electricity Failures"
      icon: mdi:alert
    voltage_l1:
      name: "Voltage Phase 1"
    voltage_l2:
      name: "Voltage Phase 2"
    voltage_l3:
      name: "Voltage Phase 3"
    current_l1:
      name: "Current Phase 1"
    current_l2:
      name: "Current Phase 2"
    current_l3:
      name: "Current Phase 3"
    power_delivered_l1:
      name: "Power Consumed Phase 1"
      accuracy_decimals: 3
    power_delivered_l2:
      name: "Power Consumed Phase 2"
      accuracy_decimals: 3
    power_delivered_l3:
      name: "Power Consumed Phase 3"
      accuracy_decimals: 3
    power_returned_l1:
      name: "Power Produced Phase 1"
      accuracy_decimals: 3
    power_returned_l2:
      name: "Power Produced Phase 2"
      accuracy_decimals: 3
    power_returned_l3:
      name: "Power Produced Phase 3"
      accuracy_decimals: 3
    gas_delivered:
      name: "Gas Consumed"
      state_class: total_increasing
    gas_delivered_be:
      name: "Gas Consumed Belgium"
      state_class: total_increasing
  - platform: uptime
    name: "SlimmeLezer Uptime"
  - platform: wifi_signal
    name: "SlimmeLezer Wi-Fi Signal"
    update_interval: 60s
 
text_sensor:
  - platform: dsmr
    identification:
      name: "DSMR Identification"
    p1_version:
      name: "DSMR Version"
    p1_version_be:
      name: "DSMR Version Belgium"
  - platform: wifi_info
    ip_address:
      name: "SlimmeLezer IP Address"
    ssid:
      name: "SlimmeLezer Wi-Fi SSID"
    bssid:
      name: "SlimmeLezer Wi-Fi BSSID"
  - platform: version
    name: "ESPHome Version"
    hide_timestamp: true
    ```

bubbamester, 2022-02-14:
[17:15:02][E][dsmr:152]: Error: telegram larger than buffer (1500 bytes)
[17:15:02][E][dsmr:265]: !80D0
 ^
Checksum mismatch
[17:15:12][E][dsmr:152]: Error: telegram larger than buffer (1500 bytes)
[17:15:12][E][dsmr:265]: !80D0
 ^
Checksum mismatch
[17:15:22][E][dsmr:152]: Error: telegram larger than buffer (1500 bytes)
[17:15:22][E][dsmr:265]: !80D0
 ^
Checksum mismatch

bubbamester, 2022-02-14:
Log of Slimmelezer web-server:

[V][dsmr:142]: Header of telegram found
[V][dsmr:142]: Header of telegram found
[D][uart_debug:158]: <<< "/HLY5/D545-METER\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[D][uart_debug:158]: <<< "0-0:1.0.0(220214172620W)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.0(00003829.948*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.1(00001816.036*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.2(00002013.912*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.3(00000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.4(00000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.0(00003623.331*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.1(00002638.635*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(00000984.696*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.3(00000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.4(00000000.000*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:3.8.0(00000020.453*kvarh)\r\n"
[D][uart_debug:158]: <<< "1-0:4.8.1(00000708.190*kvarh)\r\n"
[D][uart_debug:158]: <<< "1-0:5.8.0(00000013.117*kvarh)\r\n"
[D][uart_debug:158]: <<< "1-0:6.8.0(00000007.336*kvarh)\r\n"
[D][uart_debug:158]: <<< "1-0:7.8.0(00000441.888*kvarh)\r\n"
[D][uart_debug:158]: <<< "1-0:8.8.0(00001076.095*kvarh)\r\n"
[D][uart_debug:158]: <<< "1-0:15.8.0(00007453.279*kWh )\r\n"
[D][uart_debug:158]: <<< "1-0:32.7.0(000226.60*V)\r\n"
[D][uart_debug:158]: <<< "1-0:52.7.0(000230.70*V)\r\n"
[D][uart_debug:158]: <<< "1-0:72.7.0(000230.90*V)\r\n"
[D][uart_debug:158]: <<< "1-0:31.7.0(000000.55*A)\r\n"
[D][uart_debug:158]: <<< "1-0:51.7.0(000001.10*A)\r\n"
[D][uart_debug:158]: <<< "1-0:71.7.0(000000.20*A)\r\n"
[D][uart_debug:158]: <<< "1-0:13.7.0(0.637)\r\n"
[D][uart_debug:158]: <<< "1-0:33.7.0(0.861)\r\n"
[D][uart_debug:158]: <<< "1-0:53.7.0(0.439)\r\n"
[D][uart_debug:158]: <<< "1-0:73.7.0(0.919)\r\n"
[D][uart_debug:158]: <<< "1-0:14.7.0(49.97*HZ)\r\n"
[D][uart_debug:158]: <<< "1-0:1.7.0(0000.220*kW)\r\n"
[D][uart_debug:158]: <<< "1-0:2.7.0(0000.000*kW)\r\n"
[D][uart_debug:158]: <<< "1-0:5.7.0(0000.000*kvar)\r\n"
[D][uart_debug:158]: <<< "1-0:6.7.0(0000.000*kvar)\r\n"
[D][uart_debug:158]: <<< "1-0:7.7.0(0000.000*kvar)\r\n"
[D][uart_debug:158]: <<< "1-0:8.7.0(0000.266*kvar)\r\n"
[D][uart_debug:158]: <<< "0-0:17.0.0(00050.000)\r\n"
[D][uart_debug:158]: <<< "0-0:98.1.0\r\n"
[D][uart_debug:158]: <<< "1-0:1.8.1(00001705.795*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:2.8.2(00000934.609*kWh)\r\n"
[D][uart_debug:158]: <<< "1-0:7.8.0(00000423.185*kVarh)\r\n"
[D][uart_debug:158]: <<< "1-0:1.6.1(004.852*kW)\r\n"
[D][uart_debug:158]: <<< "1-0:2.6.2(003.940*kW)\r\n"
[D][uart_debug:158]: <<< "\r\n"
[D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[V][dsmr:105]: Start reading data from P1 port

[image]

mmakaay, 2022-02-14:
Tip: when posting logs, then use three backticks before and after the text to get it formatted correctly.
\`\`\`
so
like this
\`\`\`
In the github editor, you can also select the log text, and then click the "< >" button from the editor menu.

Looking at the debug output, it looks like bytes are being dropped in there.
For example a fragment like this looks crippled:
```
[17:18:52][D][uart_debug:158]: <<< "0-0:98.1.0\r\n"
[17:18:52][D][uart_debug:158]: <<< "(\r\n"
[17:18:52][D][uart_debug:158]: <<< "0-0:1.0.0(220201000000T)\r\n"
```

When bytes are lost, then it's to be expected that errors like this pop up:
```
[17:18:52][E][dsmr:265]: !D755
^
Checksum mismatch
```

This often happens when using software UART, but based on the settings I would expect that hardware UART is used.
You do have the super big `0-0:96.13.0` key in your DSMR output. That one might push the size of the telegrams over 3000 bytes, possibly causing bytes to be lost because they can't be read quickly enough from the UART buffer.

What you can try, based on this, is to change the two `3000` numbers to for example `3500`. So the relevant bits for this are the uart `rx_buffer_size` and the dsmr `max_telegram_length`.

bubbamester, 2022-02-15:
Unfortunately after changing the numbers I've had no success. Any other idea?

```
[17:23:38][E][dsmr:265]: !BDA0
 ^
Checksum mismatch
```

bubbamester, 2022-02-15:
If I turn off CRC check, then the result:

```
[17:40:26][V][dsmr:142]: Header of telegram found
[17:40:26][D][uart_debug:158]: <<< "/HLY5/D545-METER\r\n"
[17:40:27][D][uart_debug:158]: <<< "\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:1.0.0(220215174025W)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.0(00003844.917*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.1(00001826.031*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.2(00002018.886*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.3(00000000.000*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.4(00000000.000*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.0(00003640.901*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.1(00002656.205*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.2(00000984.696*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.3(00000000.000*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.4(00000000.000*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:3.8.0(00000020.453*kvarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:4.8.1(00000712.285*kvarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:5.8.0(00000013.117*kvarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:6.8.0(00000007.336*kvarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:7.8.0(00000443.494*kvarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:8.8.0(00001080.631*kvarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:15.8.0(00007485.818*kWh )\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:32.7.0(000231.60*V)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:52.7.0(000221.00*V)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:72.7.0(000219.10*V)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:31.7.0(000006.24*A)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:51.7.0(000001.12*A)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:71.7.0(000000.21*A)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:13.7.0(0.987)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:33.7.0(1.000)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:53.7.0(0.499)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:73.7.0(0.917)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:14.7.0(49.94*HZ)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.7.0(0001.570*kW)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.7.0(0000.000*kW)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:5.7.0(0000.000*kvar)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:6.7.0(0000.000*kvar)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:7.7.0(0000.000*kvar)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:8.7.0(0000.255*kvar)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:42.0.0(0830812100004965)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:96.1.0(0830812100004965)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:96.14.0(01)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:96.50.68(03)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:17.0.0(00050.000)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-31:4.0.0(00050.000)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-51:4.0.0(00050.000)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-71:4.0.0(00050.000)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:98.1.0\r\n"
[17:40:27][D][uart_debug:158]: <<< "(\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:1.0.0(220201000000T)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.0(00003624.904*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.1(00001705.795*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.8.2(00001919.109*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.0(00003445.867*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.1(00002511.258*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.8.2(00000934.609*kWh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:3.8.0(00000018.937*kVarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:4.8.0(00001436.860*kVarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:5.8.0(00000011.941*kVarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:6.8.0(00000006.996*kVarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:7.8.0(00000423.185*kVarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:8.8.0(00001013.675*kVarh)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:15.8.0(00007070.771*kwh  )\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.6.0(004.852*kW)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.6.1(004.852*kW)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:1.6.2(004.736*kW)\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-0:2.6.0(003.940*kW)\r\n"
[... 1 lines cut ...]
[17:40:27][D][uart_debug:158]: <<< "1-0:2.6.2(003.940*kW)\r\n"
[17:40:27][D][uart_debug:158]: <<< ")\r\n"
[17:40:27][V][dsmr:176]: Footer of telegram found
[17:40:27][D][uart_debug:158]: <<< "!F22D\r\n"
[17:40:27][V][dsmr:257]: Trying to parse telegram
[17:40:27][V][dsmr:118]: Stop reading data from P1 port
[17:40:27][D][uart_debug:158]: <<< "/HLY5/D545-METER\r\n"
[17:40:27][D][uart_debug:158]: <<< "\r\n"
[17:40:27][D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:96.13.0(FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[17:40:27][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[17:40:27][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[... cut ...]
[17:40:28][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF)\r\n"
[17:40:28][D][uart_debug:158]: <<< "!80D0\r\n"
[17:40:28][E][dsmr:265]: 1-0:1.8.0(00003624.904*kWh)
         ^
Duplicate field
[17:40:28][V][component:199]: Component dsmr took a long time for an operation (1.07 s).
[17:40:28][V][component:200]: Components should block for at most 20-30ms.
[17:40:28][V][dsmr:105]: Start reading data from P1 port
[17:40:36][V][dsmr:142]: Header of telegram found
```

mmakaay, 2022-02-15:
Turning off the CRC check has no use in this case. Data are missing. That is a different case from a meter that does not provide correct CRC values (which is what the option is meant for).

You're getting a duplicate field error, because the footer of one of the telegrams was missing. Given the header of telegram 1 and the footer of telegram 2, it's logical that there are some fields in between that got duplicated.

I have no obvious ideas about what can go wrong here.
Can you share the initial config dump that is shown when opening the log connection?

mmakaay, 2022-02-15:
It looks like my suggestion to use 3500 instead of 3000  wasn't applied. I see 3000 for Rx buffer and max telegram size.

bubbamester, 2022-02-15:
```
INFO Successfully connected to slimmelezer.local
[19:39:23][I][app:102]: ESPHome version 2022.1.4 compiled on Feb 15 2022, 19:38:16
[19:39:23][C][wifi:488]: WiFi:
[19:39:23][C][wifi:350]:   Local MAC: 44:17:93:2C:66:A9
[19:39:23][C][wifi:351]:   SSID: [redacted]
[19:39:23][C][wifi:352]:   IP Address: 192.168.0.149
[19:39:23][C][wifi:354]:   BSSID: [redacted]
[19:39:23][C][wifi:355]:   Hostname: 'slimmelezer'
[19:39:23][C][wifi:357]:   Signal strength: -56 dB ▂▄▆█
[19:39:23][V][wifi:359]:   Priority: 0.0
[19:39:23][C][wifi:361]:   Channel: 6
[19:39:23][C][wifi:362]:   Subnet: 255.255.255.0
[19:39:23][C][wifi:363]:   Gateway: 192.168.0.1
[19:39:23][C][wifi:364]:   DNS1: 192.168.0.1
[19:39:23][C][wifi:365]:   DNS2: 0.0.0.0
[19:39:23][C][logger:233]: Logger:
[19:39:23][C][logger:234]:   Level: VERBOSE
[19:39:23][C][logger:235]:   Log Baud Rate: 0
[19:39:23][C][logger:236]:   Hardware UART: UART0
[19:39:23][C][uart.arduino_esp8266:100]: UART Bus:
[19:39:23][C][uart.arduino_esp8266:102]:   RX Pin: GPIO13
[19:39:23][C][uart.arduino_esp8266:104]:   RX Buffer Size: 3500
[19:39:23][C][uart.arduino_esp8266:106]:   Baud Rate: 115200 baud
[19:39:23][C][uart.arduino_esp8266:107]:   Data Bits: 8
[19:39:23][C][uart.arduino_esp8266:108]:   Parity: NONE
[19:39:23][C][uart.arduino_esp8266:109]:   Stop bits: 1
[19:39:23][C][uart.arduino_esp8266:111]:   Using hardware serial interface.
[19:39:23][C][uptime.sensor:031]: Uptime Sensor 'SlimmeLezer Uptime'
[19:39:23][C][uptime.sensor:031]:   State Class: 'total_increasing'
[19:39:23][C][uptime.sensor:031]:   Unit of Measurement: 's'
[19:39:23][C][uptime.sensor:031]:   Accuracy Decimals: 0
[19:39:23][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[19:39:23][V][uptime.sensor:031]:   Unique ID: '4417932c66a9-uptime'
[19:39:23][C][dsmr:275]: DSMR:
[19:39:23][C][dsmr:276]:   Max telegram length: 3500
[19:39:23][C][dsmr:277]:   Receive timeout: 0.2s
[19:39:23][C][dsmr:286]:   energy_delivered_lux 'Energy Consumed Luxembourg'
[19:39:23][C][dsmr:286]:     Device Class: 'energy'
[19:39:23][C][dsmr:286]:     State Class: 'total_increasing'
[19:39:23][C][dsmr:286]:     Unit of Measurement: 'kWh'
[19:39:23][C][dsmr:286]:     Accuracy Decimals: 3
[19:39:23][C][dsmr:286]:   energy_delivered_tariff1 'Energy Consumed Tariff 1'
[19:39:23][C][dsmr:286]:     Device Class: 'energy'
[19:39:23][C][dsmr:286]:     State Class: 'total_increasing'
[19:39:23][C][dsmr:286]:     Unit of Measurement: 'kWh'
[19:39:23][C][dsmr:286]:     Accuracy Decimals: 3
[19:39:23][C][dsmr:286]:   energy_delivered_tariff2 'Energy Consumed Tariff 2'
[19:39:23][C][dsmr:286]:     Device Class: 'energy'
[19:39:23][C][dsmr:286]:     State Class: 'total_increasing'
[19:39:23][C][dsmr:286]:     Unit of Measurement: 'kWh'
[19:39:23][C][dsmr:286]:     Accuracy Decimals: 3
[19:39:23][C][dsmr:286]:   energy_returned_lux 'Energy Produced Luxembourg'
[19:39:23][C][dsmr:286]:     Device Class: 'energy'
[19:39:23][C][dsmr:286]:     State Class: 'total_increasing'
[19:39:23][C][dsmr:286]:     Unit of Measurement: 'kWh'
[19:39:23][C][dsmr:286]:     Accuracy Decimals: 3
[19:39:23][C][dsmr:286]:   energy_returned_tariff1 'Energy Produced Tariff 1'
[19:39:23][C][dsmr:286]:     Device Class: 'energy'
[19:39:23][C][dsmr:286]:     State Class: 'total_increasing'
[19:39:23][C][dsmr:286]:     Unit of Measurement: 'kWh'
[19:39:23][C][dsmr:286]:     Accuracy Decimals: 3
[19:39:23][C][dsmr:286]:   energy_returned_tariff2 'Energy Produced Tariff 2'
[19:39:23][C][dsmr:286]:     Device Class: 'energy'
[19:39:23][C][dsmr:286]:     State Class: 'total_increasing'
[19:39:23][C][dsmr:286]:     Unit of Measurement: 'kWh'
[19:39:23][C][dsmr:286]:     Accuracy Decimals: 3
[19:39:23][C][dsmr:286]:   power_delivered 'Power Consumed'
[19:39:23][C][dsmr:286]:     Device Class: 'power'
[19:39:23][C][dsmr:286]:     State Class: 'measurement'
[... 182 lines cut ...]
[19:40:58][D][uart_debug:158]: <<< "1-0:2.6.2(003.940*kW)\r\n"
[19:40:58][D][uart_debug:158]: <<< ")\r\n"
[19:40:58][V][dsmr:176]: Footer of telegram found
[19:40:58][D][uart_debug:158]: <<< "!D07C\r\n"
[19:40:58][V][dsmr:257]: Trying to parse telegram
[19:40:58][V][dsmr:118]: Stop reading data from P1 port
[19:40:58][D][uart_debug:158]: <<< "/HLY5/D545-METER\r\n"
[19:40:58][D][uart_debug:158]: <<< "\r\n"
[19:40:58][D][uart_debug:158]: <<< "1-3:0.2.8(50)\r\n"
[19:40:58][D][uart_debug:158]: <<< "0-0:96.13.0(FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[19:40:58][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[19:40:58][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[19:40:59][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[19:40:59][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[19:40:59][D][uart_debug:158]: <<< "FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF"
[... cut ...] FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF)\r\n"
[19:40:59][D][uart_debug:158]: <<< "!80D0\r\n"
[19:40:59][E][dsmr:265]: !D07C
 ^
Checksum mismatch
[19:40:59][V][component:199]: Component dsmr took a long time for an operation (1.09 s).
[19:40:59][V][component:200]: Components should block for at most 20-30ms.
[19:40:59][V][dsmr:105]: Start reading data from P1 port
[19:41:07][V][dsmr:142]: Header of telegram found
```

mmakaay, 2022-02-16:
One thing that I see that looks off, but that keeps popping up quite consistently:
```
[17:40:27][D][uart_debug:158]: <<< "0-0:98.1.0\r\n"
[17:40:27][D][uart_debug:158]: <<< "(\r\n"
[17:40:27][D][uart_debug:158]: <<< "0-0:1.0.0(220201000000T)\r\n"
```
That is plain wrong output from the meter it seems. We've seen meters that split up some measurements over two lines, and the fix for that might be causing the CRC error that you see because it strips the first `\r\n` from the data, changing the CRC value.
But with that fix for the other meters, we'd still end up with:
```
0-0:98.1.0(\r\n
0-0:1.0.0(220201000000T)\r\n
```
as the data to parse. And that first line simply is not a complete measurement.
The followup line with 0-0:1.0.0 is the same line that is sent right after the header. Fields are repeating there, and there is no new header for a new telegram. It looks like two telegrams are mashed together here.
The dsmr telegram parser trips over this data as we've seen, because a duplicate measurement key is found.

I'm starting to believe that the smart meter is sending broken data here. Normally, the patterns are a sign of bytes being dropped, but you've posted quite a few examples now in which the pattern from above consistently shows at the same location in the telegram. That is no longer random bytes, but likely actual data from the meter.

What brand and type of meter do you have? I'm curious if we can find reports with other smart meter readers that show the same behavior, or that this is specific for your meter.

I am working on new code for dsmr, in which it might be feasible to clean up this mess to make your meter work. I would like to know though, if this happens for every meter of your brand / type, or if your meter is somehow broken.

bubbamester, 2022-02-16:
My meter: Holley DTSD545.

https://www.eon.hu/content/dam/eon/eon-hungary/documents/Lakossagi/aram/muszaki-ugyek/fogyasztasmerok/Holley-DDSD285-DTSD545-v01.pdf)

bubbamester, 2022-02-19:
Here is the telegram from another user who use same Holley DTSD545 meter:

```
/HLY5/D545-METER

1-3:0.2.8(50)
0-0:1.0.0(210731132435S)
1-0:1.8.0(00008247.549*kWh)
1-0:1.8.1(00003407.883*kWh)
1-0:1.8.2(00004839.666*kWh)
1-0:1.8.3(00000000.000*kWh)
1-0:1.8.4(00000000.000*kWh)
1-0:2.8.0(00006513.786*kWh)
1-0:2.8.1(00003694.763*kWh)
1-0:2.8.2(00002819.023*kWh)
1-0:2.8.3(00000000.000*kWh)
1-0:2.8.4(00000000.000*kWh)
1-0:3.8.0(00000001.000*kvarh)
1-0:4.8.1(00001006.137*kvarh)
1-0:5.8.0(00000000.525*kvarh)
1-0:6.8.0(00000000.475*kvarh)
1-0:7.8.0(00001610.547*kvarh)
1-0:8.8.0(00002518.158*kvarh)
1-0:15.8.0(00014761.335*kWh )
1-0:32.7.0(000234.50*V)
1-0:52.7.0(000237.80*V)
1-0:72.7.0(000243.50*V)
1-0:31.7.0(000009.47*A)
1-0:51.7.0(000004.78*A)
1-0:71.7.0(000001.35*A)
1-0:13.7.0(0.943)
1-0:33.7.0(0.992)
1-0:53.7.0(0.885)
1-0:73.7.0(0.318)
1-0:14.7.0(49.95*HZ)
1-0:1.7.0(0003.097*kW)
1-0:2.7.0(0000.000*kW)
1-0:5.7.0(0000.000*kvar)
1-0:6.7.0(0000.000*kvar)
1-0:7.7.0(0000.000*kvar)
1-0:8.7.0(0001.092*kvar)
0-0:42.0.0(0812345678901247)
0-0:96.1.0(0812345678901247)
0-0:96.14.0(01)
0-0:96.50.68(03)
0-0:17.0.0(00050.000)
1-31:4.0.0(00050.000)
1-51:4.0.0(00050.000)
1-71:4.0.0(00050.000)
0-0:98.1.0
(
0-0:1.0.0(210701000000T)
1-0:1.8.0(00006239.792*kWh)
1-0:1.8.1(00002737.486*kWh)
1-0:1.8.2(00003502.306*kWh)
1-0:2.8.0(00004215.019*kWh)
1-0:2.8.1(00002500.112*kWh)
1-0:2.8.2(00001714.907*kWh)
1-0:3.8.0(00000000.934*kVarh)
1-0:4.8.0(00003695.456*kVarh)
1-0:5.8.0(00000000.472*kVarh)
1-0:6.8.0(00000000.462*kVarh)
1-0:7.8.0(00001440.663*kVarh)
1-0:8.8.0(00002254.793*kVarh)
1-0:15.8.0(00010454.811*kwh  )
1-0:1.6.0(007.352*kW)
1-0:1.6.1(006.920*kW)
1-0:1.6.2(007.352*kW)
1-0:2.6.0(005.800*kW)
1-0:2.6.1(005.800*kW)
1-0:2.6.2(004.568*kW)
)
!5261
/HLY5/D545-METER

1-3:0.2.8(50)
0-0:96.13.0(FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF)
!80D0
```

mmakaay, 2022-02-19:
Yowza! What a mess 😆
There's all kinds of things going on in there.
I'll see if I can make sense of those messages, since it seems to be consistent behaviour.
I'm knee deep in a home renovation, but in about a week I should be able to spend serious time on this.

bubbamester, 2022-02-19:
Good luck with the renovation. Just hang tight. 😁

bubbamester, 2022-02-22:
@mmakaay Talán ez segítség lehet neked.

Maybe this will help you.

A friend of mine found out why dsmr_parser fails. He said that the Holley DTSD545 meter uses CRC-16/X25. 
The dsmr_parser code only has the CRC-16/something, maybe ARC (Polynomial: x^16 + x^15 + x^2 + 1 (0xa001)) calculation.

https://github.com/ndokter/dsmr_parser/blob/5c838efbe1826e2c3e717b065af34a562413d587/dsmr_parser/parsers.py#L100

Could X25 be put into dsmr_parser?

mmakaay, 2022-02-22:
That could be added with the structure that I'm working on. It would allow for a configurable CRC check type to use.
When the parser would understand the telegrams, disabling CRC checks would be an option to work around this for smart meters that use this currently. But more work is required for that. But I appreciate the pointer, so I can keep that in mind during implementation 👍

bubbamester, 2022-04-02:
@mmakaay 

I would like to ask if there is still no progress on this issue?

Thank you.

mmakaay, 2022-04-04:
The "but in about a week" for my home renovation has become a lot more than anticipated.
My work / play room has served as temporary storage for stuff from other rooms in the house, making it unavailable for tinkering with hardware.
I'm now closer to "in about a week" than I was ever before 😄 
Getting back into the tinkering business soon now.

bubbamester, 2022-04-04:
I have been in a similar situation in last year. It is difficult to estimate the exact end of the renovation. 😄 Something unforeseen always pops up. Just hang tight. Good luck! 😁

jesperey, 2022-04-11:
I am getting Crazy 
Just spend the last 3 hours trying to the SlimmeLezer+ to work.
I have attached it to my meter but in Home assistant it just shows unknown to all values but IP, Uptime, ect.
I have read about the "configuration" file. Do this and do that to it, but I cannot seem to find, neither change it anywhere.
What to do about it? I am very close to throwing it into the ocean.

bubbamester, 2022-04-12:
@jesperey 

try this link:
https://community.home-assistant.io/t/how-to-set-up-slimmelezer-with-home-assistant-for-dummies/335174

gbarocsi, 2022-06-25:
@bubbamester 
Szinten zenesz :)

Could you solve the main issue? I've just received my slimmelezer and it's not working with the same device as you. DTSD545
Can you help? I'm not using homeassistant.

Foltifeki, 2022-06-29:
Helló
Nekem is ugyanez a problémám
lett azóta megoldás rá?

Hello
I have the same problem
has it become a solution since then?

bubbamester, 2022-06-29:
Sajnos nincs megoldás eddig. Várom a fejlesztő segítséget. Amit lehetett kipróbáltam de nincs előrelépés sajnos.

Unfortunately, there is no solution so far. I am waiting for developer help. I tried everything I could but no progress unfortunately.

gbarocsi, 2022-06-30:
Have you tried with  esp-link to read the raw data?
What about afarago's modification? https://github.com/afarago/dsmr

bubbamester, 2022-06-30:
I tried esp-link firmware without success. Afarago modification I haven't tried yet. 
I will look at it one day.  Maybe you tried?

mmakaay, 2022-06-30:
@gbarocsi Do you have the same kind of telegrams as @bubbamester ?
There are various meters that don't produce clean telegrams and @bubbamester 's meter consistently has this bit in there:
```
0-0:98.1.0
(
0-0:1.0.0(210701000000T)
```
If your meter is the same meter / produces the same output, then it's on the radar. Otherwise, I would like one or more telegram examples to work with.

In the renovation department: after some domino renovation issues (you fix one thing, you find the next), last week the last big thing (placing a rotten window frame) was done. Now there's only a leaky chimney left, but a big plastic container in the attic takes care of that until a contracter has time to remove the complete chimney from the roof. Phew :-D This means: more time for working on this. Going to dust off the new codebase for dsmr that I got so far, to see if I can crack this nut.

gbarocsi, 2022-07-11:
> @gbarocsi Do you have the same kind of telegrams as @bubbamester ? There are various meters that don't produce clean telegrams and @bubbamester 's meter consistently has this bit in there:
>
> ``` [...]

@mmakaay 
I have the same meter, same issue.
This is what I get with esplink:

.7.0(0.901)
1-0:33.7.0(0.898)
1-0:53.7.0(0.904)
1-0:73.7.0(1.000)
1-0:14.7.0(49.91*HZ)
1-0:1.7.0(00.337*kW)
1-0:2.7.0(00.000*kW)
1-0:5.7.0(00.000*kvar)
1-0:6.7.0(00.000*kvar)
1-0:7.7.0(00.000*kvar)
1-0:8.7.0(00.161*kvar)
0-0:42.0.0(30383330383132313030303130363735)
0-0:96.1.0(30383330383132313030303130363735)
0-0:96.14.0(0002)
0-0:96.50.68(03)
0-0:17.0.0(00050.000)
1-31:4.0.0(00050.000)
1-51:4.0.0(00050.000)
1-71:4.0.0(00050.000)
0-0:98.1.0
(
0-0:1.0.0(220701000000T)
1-0:1.8.0(000649.892*kWh)
1-0:1.8.1(000263.709*kWh)
1-0:1.8.2(000386.183*kWh)
1-0:2.8.0(001832.812*kWh)
1-0:2.8.1(001294.255*kWh)
1-0:2.8.2(000538.557*kWh)
1-0:3.8.0(000008.921*kVarh)
1-0:4.8.1(000140.924*kVarh)
1-0:5.8.0(000003.289*kVarh)
1-0:6.8.0(000005.632*kVarh)
1-0:7.8.0(000137.126*kVarh)
1-0:8.8.0(000179.420*kVarh)
1-0:15.8.0(002482.704*kwh  )
1-0:1.6.0(004.568*kW)
1-0:1.6.1(004.568*kW)
1-0:1.6.2(003.536*kW)
1-0:2.6.0(004.616*kW)
1-0:2.6.1(004.616*kW)
1-0:2.6.2(004.308*kW)
)
0-0:96.13.0()
!9486

I hope this helps. It would be really good to make it work.

afarago, 2022-09-14:
Some local fixes for D545 and Sanxing SX6x1, though not a complete rewrite of dsmr.
It works.

https://github.com/afarago/dsmr
https://github.com/afarago/dsmr/tree/main/specific_versions

gbarocsi, 2022-09-14:
> Some local fixes for D545 and Sanxing SX6x1, though not a complete rewrite of dsmr. It works.
> 
> https://github.com/afarago/dsmr https://github.com/afarago/dsmr/tree/main/specific_versions

How can I install it without HomeAssistant?

afarago, 2022-09-14:
> 

You can install esphome via python/pip to your local computer and pysically connect/upload via USB serial.
https://esphome.io/guides/installing_esphome.html

Foltifeki, 2022-09-14:
Hello 
Does not work
can anyone help? 🥺🥺
[image]

soadzoor, 2022-12-14:
For the record, I managed to get proper values with this description here (comment #34945):
 (Hungarian only)
[https://prohardver.hu/tema/okos_otthon_smart_home/hsz_34901-34950.html](https://prohardver.hu/tema/okos_otthon_smart_home/hsz_34901-34950.html)

Basically I installed the ESPHome addon in Home Assistant, then edited&saved&run the associated yaml file. It took some time, but it started to provide the numbers instead of "unknown".

djsomi, 2023-04-27:
> Hello Does not work can anyone help? 🥺🥺 [image]

You have to upload the components folder also to homeassistant config\esphome\ from the repo https://github.com/afarago/dsmr