esphome/issues #3043: HTTPClient crash on get()

peetereczek, 2022-02-14:
### The problem

Calling HTTP GET causes crash on Internet connectivity and continious restarts, which leads into safe mode booting.

used:
- HTTPClient library in lambda function
- http_request.get() function from yaml

No error captured in the logfile at lowest level (VERY_VERBOSE). I have also tried to call method in try/catch block, but couldn't find exception definition name.

### Which version of ESPHome has the issue?

2022.1.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.6

### What platform are you using?

ESP32

### Board

nspanel

### Component causing the issue

HTTPClient

### Example YAML snippet

```yaml
esphome:
  name: nspanel-1
  comment: Sonoff NSPanel 01

esp32:
  board: esp32dev

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  id: api_id
  services:
    # Expose "send_json" to Home Assistant for scripts/testing
    - service: send_json
      variables:
        my_type: int
        my_json: string
      then:
        - lambda: 'id(nspanel1).send_json_command(my_type,my_json);'
    - service: play_rtttl
      variables:
        song_str: string
      then:
        - rtttl.play:
            rtttl: !lambda 'return song_str;'

ota:
  password: ""

wifi:
[... 790 lines cut ...]
[14:30:20][D][homeassistant.binary_sensor:026]: 'light.light_ikea_sypialnia': Got state OFF
[14:30:20][D][binary_sensor:034]: 'dim_light_sypialnia': Sending initial state OFF
[14:30:20][D][homeassistant.binary_sensor:026]: 'light.sonoff_1000ee2034': Got state OFF
[14:30:20][D][binary_sensor:034]: 'wled_light_sypialnia': Sending initial state OFF
[14:30:21][D][sensor:125]: 'nsp01_ntc_source': Sending state 1.68700 V with 2 decimals of accuracy
[14:30:21][D][resistance:037]: 'nsp01_resistance_sensor' - Resistance 11713.8Ω
[14:30:21][D][sensor:125]: 'nsp01_resistance_sensor': Sending state 11713.82617 Ω with 1 decimals of accuracy
[14:30:21][D][ntc:026]: 'nsp01 Temperature' - Temperature: 23.4°C
[14:30:21][D][sensor:125]: 'nsp01 Temperature': Sending state 23.43503 °C with 1 decimals of accuracy
[14:30:22][D][nspanel:349]: Sending JSON command: {"temperature":23.43503,"tempUnit":0}
[14:30:22][D][main:290]: [nspanel_boot] updating Wi-Fi state
[14:30:22][D][nspanel:349]: Sending JSON command: {"wifiState":"connected","rssiLevel":3}
[14:30:22][D][main:351]: [nspanel_boot] Home Assistant API is connected!
[14:30:22][D][main:355]: [nspanel_boot] starting weather service
[14:30:22][D][main:644]: [nspanel_weather] started
[14:30:22][D][nspanel_weather:653]: Init Complete
[14:30:22][D][nspanel_weather:659]: http.begin OK
[14:30:22][D][nspanel_weather:662]: http.useHTTP10 flag set
[14:30:22][D][nspanel_weather:663]: http.GET
INFO nspanel-1.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for nspanel-1.local
WARNING Disconnected from API

and continiously restarting NSPanel screen
```

peetereczek, 2022-02-14:
I don't understand why, but with static IP configuration, above issue does not exist.

manual_ip:
    static_ip: 192.168.50.151
    gateway: 192.168.50.1
    subnet: 255.255.255.0
    dns1: 8.8.8.8
    dns2: 192.168.50.165

[14:51:06][D][main:296]: [nspanel_boot] updating Wi-Fi state
[14:51:06][D][nspanel:349]: Sending JSON command: {"wifiState":"connected","rssiLevel":2}
[14:51:06][D][main:357]: [nspanel_boot] Home Assistant API is connected!
[14:51:06][D][main:361]: [nspanel_boot] starting weather service
[14:51:06][D][main:650]: [nspanel_weather] started
[14:51:06][D][nspanel_weather:659]: Init Complete
[14:51:06][D][nspanel_weather:665]: http.begin OK
[14:51:06][D][nspanel_weather:668]: http.useHTTP10 flag set
[14:51:06][D][nspanel_weather:669]: http.GET
[14:51:07][D][nspanel_weather:674]: http.GET OK
[14:51:07][D][nspanel_weather:678]: HTTP response code is 200
[14:51:07][D][nspanel_weather:684]: JSON Parse OK
[14:51:07][D][nspanel_weather:686]: weather code received: 113
[14:51:07][D][nspanel:349]: Sending JSON command: {"HMI_weather":1,"HMI_outdoorTemp":{"current":8,"range":"0,7"}}
[14:51:07][D][main:755]: [nspanel_weather] weather update completed
[14:51:07][D][main:365]: [nspanel_boot] boot sequence completed.

OttoWinter, 2022-02-15:
If the issue also happens with the `HTTPClient` class that would suggest the error is in the arduino framework, not directly in esphome.

closed 2022-06-30 by github-actions[bot]