esphome/issues #3044: MQTT discovery for new fan component is broken in esphome/home assistant

andreasmwestberg, 2022-02-14:
### The problem

Home assistant doesn't add fans from MQTT discovery because of deprecation error.

`voluptuous.error.Invalid: The 'speed_command_topic' option has been removed, please remove it from your configuration`

### Which version of ESPHome has the issue?

2022.1.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.6

### What platform are you using?

ESP8266

### Board

Weemos D1 mini

### Component causing the issue

MQTT/fan

### Example YAML snippet

```yaml
esphome:
  name: ikea-fornuftig

esp8266:
  board: d1_mini
  
substitutions:
  friendly_name: Förnuftig

# Enable logging
logger:

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

# Enable Home Assistant API
#api:

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ikea-Fornuftig Fallback Hotspot"
    password: "xxx"

output:
  - platform: template
    id: fan_out_internal
    type: float
    write_action:
      lambda: 'return;'

fan:
 - platform: speed
   output: fan_out_internal
   id: fan_internal
   name: "$friendly_name Fan"
   speed_count: 3
```

### Anything in the logs that might be useful for us?

```txt
2022-02-14 16:54:54 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_fan_mqtt': ({'speed_level_command_topic': 'ikea-fornuftig/fan/frnuftig_fan/speed_level/command', 'speed_level_state_topic': 'ikea-fornuftig/fan/frnuftig_fan/speed_level/state', 'speed_command_topic': 'ikea-fornuftig/fan/frnuftig_fan/speed/command', 'speed_state_topic': 'ikea-fornuftig/fan/frnuftig_fan/speed/state', 'name': 'Förnuftig Fan', 'state_topic': 'ikea-fornuftig/fan/frnuftig_fan/state', 'command_topic': 'ikea-fornuftig/fan/frnuftig_fan/command', 'availability_topic': 'ikea-fornuftig/status', 'unique_id': 'ESPfanfrnuftig_fan', 'device': {'identifiers': 'a848facd4dd5', 'name': 'ikea-fornuftig', 'sw_version': 'esphome v2022.1.4 Feb 14 2022, 16:54:15', 'model': 'd1_mini', 'manufacturer': 'espressif'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/helpers/config_validation.py", line 813, in validator
    raise vol.Invalid(warning % arguments)
voluptuous.error.Invalid: The 'speed_command_topic' option has been removed, please remove it from your configuration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 211, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__
    raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: The 'speed_command_topic' option has been removed, please remove it from your configuration
```

kquinsland, 2022-06-08:
Please let me know if I should open a new Issue for this or if it's not an issue at all.

----

This PR **does** solve the MQTT -> Home Assistant auto configuration problems  (yay!) but it introduces another issue:

The fan entity shows up in HA with a number slider that goes between 0 and 100 but there is nothing in ESPHome that allows for 'mapping' this back for legacy entities.

I have a tuya fan that supports 4 speeds. This yaml renders a 'correct' control on the ESPHome web UI: a toggle + slider that can go from 1 to 4.

But in Home Assistant, any time the slider is moved  to some value that is larger than 4 (like, the value 43), this error is emitted:
`[mqtt.fan:080] | Invalid speed level 43`.

The `on_speed_set` lambda appears to be called only after the mqtt component has done it's validation: https://esphome.io/api/mqtt__fan_8cpp_source.html#L80 so there is no way to intercept the raw percentage and map it back to a suitable value.

```yaml
fan:
  - name: ${friendly_name} Speed
    id: device_fan
    platform: "tuya"
    switch_datapoint: 1
    speed_datapoint: 3
    speed_count: 4
    icon: "mdi:ceiling-fan"
    on_speed_set:
      then:
        - logger.log: "Fan Speed was changed!"
        # Docs are not super clear on this. You don't get the speed passed in as an `x` arg.. try to get it from id().speed instead.
        - lambda: |
            ESP_LOGD("speed", "(raw) speed set to: %f", id(device_fan).speed);
```

Should I open an issue or is this intentional? The only workaround that I can think of is to make the `tuya` fan internal and then implement my own template fan which should allow me to implement my own map logic.

kquinsland, 2022-06-09:
After working with @ssieb on discord, he was able to track down the issue. I was able to verify complete/proper functionality once the `spd_rng_max` field was added to the MQTT auto discovery payload.

This commit: https://github.com/esphome/esphome/commit/c6df476a847ae94fb648b5dfd164efa3ce5fc9a0 fixed the issue

ssieb, 2022-06-09:
https://github.com/esphome/esphome/pull/3537