esphome/issues #3046: Docker fresh install "Can’t connect to ESP. Please make sure your YAML file contains an ‘api:’ line."

gvdhoven, 2022-02-15:
### The problem

I spun up a fresh install of ESPHome next to my home assistant docker setup in order to create my ESPHome devices. The docker-compose part is this:

```
automation-esphome:
    image: esphome/esphome:latest
    container_name: automation-esphome
    restart: unless-stopped
    environment:
      - TZ=Europe/Amsterdam
    networks:
      f43lan:
        ipv4_address: 192.168.1.108
    ports:
      - 6052:6052
      - 6123:6123
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /data/docker/automation/esphome/config:/config
```

as you can see nothing special; All my containers which i should be able to reach either via SSL (Traefik) or internally on the network i have it setup on a separate IP since i don't want to expose my entire server + ports to the world.

The error message i get while installing the integration is:
> "Can’t connect to ESP. Please make sure your YAML file contains an ‘api:’ line."

What am i doing wrong to get the dashboard working in HA?

Changing `networks:` to `network_mode: host` doesn't resolve the issue since HA still needs to be on a traefik network in order for me to host it myself with SSL.

### Which version of ESPHome has the issue?

latest docker image

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.0.dev20220204

### What platform are you using?

ESP32

### Board

n/a

### Component causing the issue

n/a

### Example YAML snippet

```yaml
n/a
```

### Anything in the logs that might be useful for us?

```txt
n/a
```

### Additional information

n/a

mmakaay, 2022-02-15:
When installing the ESPHome integration in Home Assistant, I presume that you are trying to integrate a device that was flashed with ESPHome firmware. Am I right? If not, I might be misunderstanding the exact steps you're taking here.

If ESPHome is not running on the same network segment as the ESPHome devices, then the normal behavior is that you won't be able to connect based on the default naming scheme of `somedevice.local`. For that scheme to work, mDNS traffic must be visible to ESPHome. Were you trying this?

Solutions would be:

- to make mDNS work between the devices and the ESPHome container somehow (which might be a bit of a hassle, since an mDNS relay has to be active on the docker bridge, ander docker manages its own iptables rules for that one).
- to have host networking connect to the LAN segment on which your ESPHome devices reside. You're telling that things are behind Traefik, but that has no relation to the ESPHome devices. The ESPHome dashboard is the part that would be accessible through Traefik, but the ESPHome devices and the dashboard would communicate via different means on the LAN segement that they are on.
- to use fixed IP-addresses for the ESPHome devices, and to let the dashboard make use of ping instead of mDNS for checking if devices are online or not (there's an environment var for that that you could use in the docker-compose config).

Also note that the ESPHome container is not needed for the HA integration to work. The dashboard is a nice tool to hold your configs, to view the devices on the network and to compile firmware. But once the ESPHome device gets its firmware installed, it's an independent device on the network. It provides an HTTP based API server (that's the one that is activated using `api:` in the YAML config) and HA will login to that API server as a client.
So for the actual HA integration to work, it's important that HA can reach the API server that is running on the device. The ESPHome container is not involved at all here.
For Home Assistant to be able to see the ESPHome devices, the same kind of things apply as mentioned above. Basically: either make mDNS work or use  static IP addresses and add devices based on their IP-adres instead of their name.

If this doesn't cover the issue, then please clarify what steps you take for "installing the integration in HA".

OttoWinter, 2022-02-15:
Just to make sure: When in the HA interface you try to add the device, you're putting in the IP/hostname of the esphome _device_ (as in the Ip of the esp itself), not the ip of the container running the esphome dashboard.

gvdhoven, 2022-02-15:
@mmakaay and @OttoWinter i was indeed 'assuming' i would need to fill in the IP of the ESPHome container (since that also exposes the same port, 6052). Maybe the integration setup dialog can be made more clear to mention that you have to fill in the IP of a pre-flashed ESPHome _**device**_ instead of the IP of the container. I will assume this is the fix (going to try when i am home).

closed 2022-02-15 by gvdhoven