esphome/issues #3047: Cannot disable MQTT logging

Semmu, 2022-02-15:
### The problem

Hi,

I'm trying to create some ESP based simple "weather station" and report its sensors' values to some cloud MQTT provider. Since I`m on the free tier I have a limited amount of MQTT messages I can send in a month, so I have to absolutely minimize it.

The problem is that since I have added the MQTT client to my ESPhome configuration it is sending all its log messages and sensor readings to the broker and I cannot turn it off. Read the docs and tried many things but none of them seem to have any effect.

My goal is to only trigger sending MQTT messages in my lambdas and events, where I have complete control over them.

### Which version of ESPHome has the issue?

2022.1.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

not applicable

### What platform are you using?

ESP8266

### Board

Wemos/Lolin D1 mini

### Component causing the issue

MQTT client

### Example YAML snippet

```yaml
I have my MQTT configured like this:

mqtt:
  broker: [broker IP]
  username: [username]

And I am trying to disable MQTT logging like this:

```yaml
logger:
  logs:
    mqtt.client: NONE
    mqtt.component: NONE
    mqtt: NONE
```

I copied this snippet from the docs and tried other stuff but I still get sensor readings and Home Assistant-related messages sent to the broker, which I want to disable.
```

Semmu, 2022-02-15:
Fortunately it seems I have managed to solve my problem, I will share my solution below, but I think some kind of global switch for disabling automatic MQTT reporting and messages would be very useful.

---

The stuff I changed:

First, MQTT discovery needs to be set to false, so the system does not publish `homeassistant/*` messages automatically, and also the `log_topic` level needs to be set to `NONE`. I also set the `will_message` to nothing just to further minimize the noise.

```yaml
mqtt:
  ...
  discovery: false
  ...
  will_message: # empty object
  ...
  log_topic:
    topic: "doesnotmatter" # literally does not matter
    level: NONE
```

Then for the individual sensors I had to set them to `internal`, so now they do not publish their state changes automatically either.

```yaml
sensor:
  - platform: dht
    ...
    temperature:
      internal: true
      ...
    humidity:
      internal: true
      ...
  - platform: bh1750
    internal: true
    ...
  - platform: wifi_signal
    internal: true
    ...
```

And as for the manual MQTT message sending I used `cron`-style scheduling, lambdas, global variables and the `mqtt.publish_json` action, for example:

```yaml
time:
  - platform: sntp
    on_time:
      - cron: '* * * * * *'
        then:
          - mqtt.publish_json:
              topic: your/topic/here
              payload: |-
                root["last_reading"] = id(last_reading);

globals:
  - id: last_reading
    type: float

sensor:
  - platform: ...
    ...
    on_value:
      then:
        - lambda: id(last_reading) = x;
```

In the end this does exactly what I needed but I have to admit it took me hours to figure out all these settings and disable automatic logging.

OttoWinter, 2022-02-15:
You can disable logging by setting the log topic to null

```yaml
mqtt:
  log_topic: null
```

Contributions to the docs are welcome 😉

closed 2022-06-30 by github-actions[bot]