esphome/issues #3048: APA102 NeoPixelBus example

pruwait, 2022-02-15:
### The problem

I can use esphome with fastled_spi light platform.
But effect speed too low.

I want to use NeoPixelBus, but apa102 is not working with my code.

hardware:
WT32-ETH01 + 74AHCT125N

### Which version of ESPHome has the issue?

2022.1.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.2.6

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

NeoPixelBus

### Example YAML snippet

```yaml
esphome:
  name: stairs2ch
  platform: ESP32
  board: esp-wrover-kit

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

debug:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

    
light:
#  - platform: fastled_spi
  - platform: neopixelbus
#    chipset: APA102  # fastled_spi
    variant: DotStar
    data_pin: GPIO04
    clock_pin: GPIO14
    type: BGR
#    rgb_order: BGR  # fastled_spi
    
    num_leds: 1140
    name: "StairStrip1"
    id: staircase1
    
    #data_rate: 400kHz  # fastled_spi
    method: 
      type: spi
      speed: 500kHz
      bus:  vspi
    effects:
      - addressable_rainbow:
```

pruwait, 2022-02-16:
May be I must to use SPI Bus component or another data/clock pins on this board... Need more information.

pruwait, 2022-02-22:
Methods esp32_i2s & esp32_rmt does not support dotstar.
Hardware SPI pins 13+14, 18+23 is using with ethernet on this board.

NeopixelBus not working for WT32-ETH01 + dotstar.

pruwait, 2022-02-23:
I changed wire to pin 13 + pin 14 (hardwere spi).
Neopixelbus works but so bad...
FastLed works normally for esp32 + apa102.

````
substitutions:
  device_name: "canopy"

esphome:
  name: canopy-light
  platform: ESP32
  board: wemos_d1_mini32

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password:  !secret wifi_password

  ap:
    ssid: "${device_name} Hotspot"
    password: "!secret ota_pass"

captive_portal:

web_server:
  port: 80

light:
#  - platform: fastled_spi
  - platform: neopixelbus
    type: BGR  # neopixelbus
#    chipset: APA102 # fastled_spi
    variant: DotStar #neopixelbus
#    data_pin: GPIO23
#    clock_pin: GPIO18

    data_pin: GPIO13
    clock_pin: GPIO14

    
#    rgb_order: BGR  # fastled_spi
    
    num_leds: 150
    name: "StairStrip"
    id: staircase
#    max_refresh_rate: 1ms

    method:  # neopixelbus
      type: spi # neopixelbus
#      speed: 500kHz # neopixelbus
      bus:  hspi # neopixelbus
````

pruwait, 2022-02-23:
With neopixelbus compiling have warnings:
````
Compiling /data/canopy-light/.pioenvs/canopy-light/FrameworkArduino/esp32-hal-spi.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
````

And almost all operations working with errors. I cant change color or brightness normally.

edwardtfn, 2025-06-01:
Do you still have this system with `APA102`? I'm looking for volunteers to give a try on PR#7980:

- esphome/esphome#7980