esphome/issues #3049: ESP does not have enough space to store OTA file

dhover, 2022-02-16:
### The problem

After update of ESPHome to 2022.2.0 the firmware cannot be updates anymore.

### Which version of ESPHome has the issue?

2022.2.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.2.7

### What platform are you using?

ESP8266

### Board

nous

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/nous-02.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing nous-02 (board: esp8285; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <noise-c> 0.1.4
|   |-- <libsodium> 1.10018.1
|-- <ArduinoJson> 6.18.5
RAM:   [====      ]  44.1% (used 36148 bytes from 81920 bytes)
Flash: [======    ]  55.2% (used 564801 bytes from 1023984 bytes)
========================= [SUCCESS] Took 1.92 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.186.26
INFO Uploading /data/nous-02/.pioenvs/nous-02/firmware.bin (568960 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

### Additional information

In previous ESPHome versions the firmware was compressed before it was sent. Now I don't see anything in the logs if it is being compressed.

dhover, 2022-02-16:
https://github.com/esphome/esphome/pull/2601

OttoWinter, 2022-02-16:
OTA compression was accidentally disabled in 2022.2.0, but will be fixed in .1 (https://github.com/esphome/esphome/pull/3206)

dhover, 2022-02-17:
I've installed 2022.2.1 but the compression is still not used.

dhover, 2022-02-17:
```
INFO Reading configuration /config/esphome/nous-01.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing nous-01 (board: esp8285; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <noise-c> 0.1.4
|   |-- <libsodium> 1.10018.1
|-- <ArduinoJson> 6.18.5
Compiling /data/nous-01/.pioenvs/nous-01/src/main.cpp.o
Linking /data/nous-01/.pioenvs/nous-01/firmware.elf
RAM:   [====      ]  44.1% (used 36148 bytes from 81920 bytes)
Flash: [======    ]  55.2% (used 564801 bytes from 1023984 bytes)
Building /data/nous-01/.pioenvs/nous-01/firmware.bin
esp8266_copy_factory_bin(["/data/nous-01/.pioenvs/nous-01/firmware.bin"], ["/data/nous-01/.pioenvs/nous-01/firmware.elf"])
========================= [SUCCESS] Took 6.69 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.186.25
INFO Uploading /data/nous-01/.pioenvs/nous-01/firmware.bin (568960 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

OttoWinter, 2022-02-17:
> I've installed 2022.2.1 but the compression is still not used.

Yes, the issue is on the device side. If the device currently has .0 flashed the compression doesn't activate.

The only way to upload a large binary when .0 is currently on the device is by creating a minimal 2022.2.1 binary, then uploading the wanted binary again

dhover, 2022-02-17:
That does the job. Thanks.