esphome/issues #3050: section `.text1' will not fit in region `iram1_0_seg'

derjoerg, 2022-02-16:
### The problem

After upgrade to 2022.2.1 (from 2022.1.3) installation aborts with 
`
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/room2-info/.pioenvs/room2-info/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [/data/room2-info/.pioenvs/room2-info/firmware.elf] Error 1
`

With version 2022.1.3 everything worked. Also the delete of the old build files didn't helped.

I already tried to delete build files but no change.

### Which version of ESPHome has the issue?

2022.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.7

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

iram (perhaps)

### Example YAML snippet

```yaml
substitutions:
  devicename: room2-info
  friendly_name: Room2 Info
  device_description: Information from Room2

esphome:
  name: $devicename
  comment: ${device_description}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "123"
  password: "123"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback Hotspot"
    password: "123"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "123"

ota:
[... 355 lines cut ...]
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/spiffs/spiffs_check.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/spiffs/spiffs_gc.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/spiffs/spiffs_hydrogen.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/spiffs/spiffs_nucleus.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/spiffs_api.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/sqrt32.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/stdlib_noniso.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/time.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/uart.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling /data/room2-info/.pioenvs/room2-info/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving /data/room2-info/.pioenvs/room2-info/libFrameworkArduino.a
Linking /data/room2-info/.pioenvs/room2-info/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/room2-info/.pioenvs/room2-info/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [/data/room2-info/.pioenvs/room2-info/firmware.elf] Error 1
========================= [FAILED] Took 327.44 seconds =========================
```

### Additional information

It worked with version 2021.1.3.

truglodite, 2022-02-17:
Same issue here, 2022.2.1 has the issue, and it was working fine with previous versions 2022.1.4. I wonder if some libraries were changed and now using too much RAM. It's not affecting all of my devices, but most of them (especially ones with OLED that process text).

Here's one that won't compile:
```
substitutions:
  name: "master"
  friendly_name: "Master"

esphome:
  name: ${name}
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:

api:
  password: !secret master_password
  encryption:
    key: !secret encryption_key
# Enable ota
ota:
  password: !secret ota_password

# Device Specific Configs ###########################################
# Wemos D1 Mini
#
# dallas -> GPIO16 (D0)
# sda -> GPIO5 (D1)
# scl -> GPIO4 (D2)
# soft rx -> GPIO0 (D3) [keep high on boot!!!]
# built in LED -> GPIO2 (D4) [keep high on boot!!!]
# DHT -> GPIO14(D5)
# motion -> GPIO12(D6)
# soft tx -> GPIO13(D7)
# button (to Vcc) -> GPIO15(D8) [keep low on boot!!!]

uart:
  tx_pin: D7
  rx_pin: D3
  baud_rate: 9600

i2c:
  sda: D1
  scl: D2

dfplayer:

time:
  - platform: homeassistant
    id: ha_time
    
text_sensor:
  - platform: homeassistant
    name: "ESPhome Voice"
    internal: true
    entity_id: input_text.esphome_voice
    on_value:
      then:
        - dfplayer.set_volume: 25
        - delay: 150ms
        - dfplayer.play_folder:
            folder: 1
            file: !lambda return atoi(x.c_str());
            
font:
  - file: "verdana.ttf"
[... 137 lines cut ...]
    entity_id: sensor.backyard_humid
    internal: true
  - platform: homeassistant
    name: "Inside Temp"
    id: inside_temp
    entity_id: sensor.inside_temp_avg
    internal: true
  - platform: homeassistant
    name: "Inside Humid"
    id: thermostat_humid
    entity_id: sensor.thermostat_humidity
    internal: true
```

...and here is the error out:

```
Linking /data/master/.pioenvs/master/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /data/master/.pioenvs/master/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [/data/master/.pioenvs/master/firmware.elf] Error 1
========================= [FAILED] Took 16.18 seconds =========================
```

On a side note, I noticed compiling seems to take a lot longer since this update, even for the devices that are still working. Not sure if it's related, but thought I'd mention the observation.

OttoWinter, 2022-02-18:
Hi @derjoerg,

I tested your binary and could reproduce. By some analysis I found that a particular method is taking up a lot of that IRAM space, so this PR makes the file compile for me at least: https://github.com/esphome/esphome/pull/3223

Unfortunately, I think larger configurations will still hit this issue because we're quite close to the limit (vast majority of that is due to arduino itself though, not esphome). If you're still experiencing this issue, please

1. Post your full config (with sensitive information redacted) so I can find additional large objects
2. Use the flags listed [here](https://docs.platformio.org/en/stable/platforms/espressif8266.html#mmu-adjusting-icache-to-iram-ratio) to increase this section size, at some (minor) performance cost:

```yaml
esphome:
  # ...
  platformio_options:
    build_flags:
      - -DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48
```

derjoerg, 2022-02-18:
Hi,
so the `build_flags -DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48` fixed the issue and I can compile.
I will wait for 2022.2.4 to see if https://github.com/esphome/esphome/pull/3223 will fix it without the build_flags.

closed 2022-02-19 by OttoWinter