esphome/issues #3052: Unable to configure output for button on M5 Atom Matrix (ESP32-PICO)

Xelinor, 2022-02-16:
### The problem

Pinout of device is here: https://static-cdn.m5stack.com/resource/docs/static/assets/img/product_pics/core/minicore/atom/ATOM_MATRIX_SIMPLE_CIRCUT_20200514.webp

When trying to configure a button output on pin 39, I get the following error:
`GPIO39 (34-39) does not support output pin mode.`

At least according to the pinout, this should work. It's not a switch in the device but a stateless button.

### Which version of ESPHome has the issue?

2022.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.7

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

output

### Example YAML snippet

```yaml
output:
  - platform: gpio
    pin: 39
    id: button1

button:
  - platform: output
    name: "Start Timer"
    output: button1
    duration: 500ms
```

### Anything in the logs that might be useful for us?

```txt
GPIO39 (34-39) does not support output pin mode.
```

randybb, 2022-02-17:
A button is a type of output that can be triggered from esphome (yaml, lambda), or remotely. Which means esphome device can act like as a button for some other connected device. Your physical button on your atom is from esphome perspective a binary input, you have to use [GPIO](https://esphome.io/components/binary_sensor/gpio.html) under binary_sensor component.

jesserockz, 2022-02-17:
Closing this as this should be using `binary_sensor` not `button` which is a different thing.

closed 2022-02-17 by jesserockz