esphome/issues #3057: Maxcio Diffuser unusable since ESPHome 2022.1

Nardol, 2022-02-17:
### The problem

I missed some version and tried to update my Maxcio difuser to ESPHome 2022.2.
When uploading firmware, it looks like it connects to wifi but reboot in loop and Home Assistant has timeout when trying to get entities.

### Which version of ESPHome has the issue?

2022.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2022.2.7

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

tuya/wifi/api

### Example YAML snippet

```yaml
substitutions:
  room: Bureau
  sufix: bureau
  timezone: "Europe/Paris"

esphome:
  name: diffuseur-${sufix}
  project:
    name: "Nardol.maxcio_diffuser"
    version: "1.2.2"
  #on_boot:
    #then:
      #- uart.write: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0D]

esp8266:
  board: esp01_1m
  restore_from_flash: true

#web_server:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key
  ap:
    password: !secret ap_key

#captive_portal:

api:
  password: !secret api_pass
  encryption:
    key: !secret encryption_key
  services:
    - service: morning_diffusion
      then:
        - script.execute: morning_diffusion

ota:
  password: !secret ota_pass

uart:
  baud_rate: 9600
  tx_pin: GPIO1
  rx_pin: GPIO3

#debug:

logger:
  baud_rate: 0
  level: NONE
  #esp8266_store_log_strings_in_flash: False     #  :doc:`recommanded for ESP8266 </components/sensor/custom>`

tuya:
  time_id: my_timer
  on_datapoint_update:
    - sensor_datapoint: 13
      datapoint_type: enum
      then:
        - lambda: if (x>0x0) tuya_tuya->set_enum_datapoint_value(13, 0x0);

binary_sensor:
  - platform: "tuya"
    name: Eau Diffuseur ${room}
    id: eau_diffuseur_${sufix}
    sensor_datapoint: 12
    device_class: "problem"
    on_press:
      then:
        - script.stop: planed_diffusion
        - script.stop: morning_diffusion
        - homeassistant.service:
            service: script.notify_mobile
            data:
              message: "Il n'y a plus d'eau dans le diffuseur du ${sufix}"
              title: "Capteur d'eau du diffuseur du ${sufix}"
              tag: "eau_diffuseur_${sufix}"
        - homeassistant.service:
            service: persistent_notification.create
            data:
              title: "Capteur d'eau du diffuseur du ${sufix}"
              message: "Il n'y a plus d'eau dans le diffuseur du ${sufix}"
              notification_id: "eau_diffuseur_${sufix}"
    on_release:
      then:
        - homeassistant.service:
            service: persistent_notification.dismiss
            data:
              notification_id: "eau_diffuseur_${sufix}"
        - homeassistant.service:
            service: script.notify_mobile
            data:
              message: "clear_notification"
              title: ""
              tag: "eau_diffuseur_${sufix}"
  - platform: tuya
    name: ""
    id: power
    sensor_datapoint: 1
    internal: true
    on_release:
      then:
        - script.stop: planed_diffusion
        - script.stop: morning_diffusion

fan:
  - platform: tuya
    name: "Diffuseur ${room}"
    id: diffuseur_${sufix}
    switch_datapoint: 1
    speed_datapoint: 103
    speed_count: 3

sensor:
  - platform: tuya
    name: ""
    id: turbo
    sensor_datapoint: 103
    internal: true
    on_value_range:
      above: 2
      then:
        - fan.turn_off: diffuseur_${sufix}

light:
  - platform: tuya
    name: Leds Diffuseur ${room}
    id: leds_diffuseur_${sufix}
    dimmer_datapoint: 111
    switch_datapoint: 11
    min_value: 1
    rgb_datapoint: 108

# Defining planned diffusion
time:
  - platform: homeassistant
    id: my_timer
    timezone: ${timezone}
    on_time:
      minutes: 0
      hours: 5
      seconds: 0
      then:
        - if:
            condition:
              and:
                # - binary_sensor.is_off: eau_diffuseur_${sufix}
                - binary_sensor.is_off: power
            then:
              - script.execute: planed_diffusion

script:
  - id: planed_diffusion
    then:
      - fan.turn_on: diffuseur_${sufix}
      - delay: 00:00:01
      - light.turn_off: leds_diffuseur_${sufix}
      - delay:
          hours: 1
      - fan.turn_off: diffuseur_${sufix}
  - id: morning_diffusion
    then:
      - if:
          condition:
            and:
              # - binary_sensor.is_off: eau_diffuseur_${sufix}
              - binary_sensor.is_off: power
          then:
            - fan.turn_on: diffuseur_${sufix}
            - delay: 00:00:01
            - light.turn_off: leds_diffuseur_${sufix}
            - delay:
                minutes: 30
            - fan.turn_off: diffuseur_${sufix}
```

### Anything in the logs that might be useful for us?

```txt
In home-assistant.log:
2022-02-17 17:16:52 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for diffuseur-bureau @ 192.168.22.17: Timeout while connecting to ('192.168.22.17', 6053)
2022-02-17 17:17:21 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 192.168.22.17: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>
2022-02-17 17:18:35 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for diffuseur-bureau @ 192.168.22.17: Timeout during handshake
```

### Additional information

Unfortunately I cannot solder and directly connect to debug using other than wifi.
I set the log level to debug and activated debug component but timeouts made impossible to know what was doing wrong, some times I was able to connect but all looked fine as usual.
The most recent version it works with is 2021.12, tested with 2021.12.0.
2022.1 fails.

Nardol, 2022-02-18:
With a minimal firmware uploaded I have the following characteristics:
```txt
[10:08:32][I][app:102]: ESPHome version 2022.2.1 compiled on Feb 17 2022, 19:45:08
[10:08:32][C][wifi:491]: WiFi:
[10:08:32][C][wifi:353]:   Local MAC: XXX
[10:08:32][C][wifi:354]:   SSID: 'xxx'
[10:08:32][C][wifi:355]:   IP Address: 192.168.22.17
[10:08:32][C][wifi:356]:   BSSID: xxx
[10:08:32][C][wifi:358]:   Hostname: 'diffuseur-bureau'
[10:08:32][C][wifi:360]:   Signal strength: -60 dB ▂▄▆█
[10:08:32][C][wifi:364]:   Channel: 2
[10:08:32][C][wifi:365]:   Subnet: 255.255.255.0
[10:08:32][C][wifi:366]:   Gateway: 192.168.22.254
[10:08:32][C][wifi:367]:   DNS1: 192.168.22.254
[10:08:32][C][wifi:368]:   DNS2: 0.0.0.0
[10:08:32][C][logger:233]: Logger:
[10:08:32][C][logger:234]:   Level: DEBUG
[10:08:32][C][logger:235]:   Log Baud Rate: 0
[10:08:32][C][logger:236]:   Hardware UART: UART0
[10:08:32][C][uart.arduino_esp8266:102]: UART Bus:
[10:08:32][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[10:08:32][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[10:08:32][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[10:08:32][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[10:08:32][C][uart.arduino_esp8266:109]:   Data Bits: 8
[10:08:32][C][uart.arduino_esp8266:110]:   Parity: NONE
[10:08:32][C][uart.arduino_esp8266:111]:   Stop bits: 1
[10:08:32][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[10:08:32][C][mdns:084]: mDNS:
[10:08:32][C][mdns:085]:   Hostname: diffuseur-bureau
[10:08:32][C][ota:085]: Over-The-Air Updates:
[10:08:32][C][ota:086]:   Address: diffuseur-bureau.local:8266
[10:08:32][C][ota:089]:   Using Password.
[10:08:32][C][api:138]: API Server:
[10:08:32][C][api:139]:   Address: diffuseur-bureau.local:6053
[10:08:32][C][api:141]:   Using noise encryption: YES
[10:08:32][C][sntp:050]: SNTP Time:
[10:08:32][C][sntp:051]:   Server 1: '0.pool.ntp.org'
[10:08:32][C][sntp:052]:   Server 2: '1.pool.ntp.org'
[10:08:32][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[10:08:32][C][sntp:054]:   Timezone: '${timezone}'
[10:08:32][C][debug:049]: Debug component:
[10:08:32][D][debug:061]: ESPHome version 2022.2.1
[10:08:32][D][debug:065]: Free Heap Size: 31184 bytes
[10:08:32][D][debug:093]: Flash Chip: Size=1024kB Speed=40MHz Mode=DOUT
[10:08:32][D][debug:250]: Chip ID: 0x00053F99
[10:08:32][D][debug:251]: SDK Version: 2.2.2-dev(38a443e)
[10:08:32][D][debug:252]: Core Version: 3.0.2
[10:08:32][D][debug:253]: Boot Version=31 Mode=1
[10:08:32][D][debug:254]: CPU Frequency: 80
[10:08:32][D][debug:255]: Flash Chip ID=0x001540C8
[10:08:32][D][debug:256]: Reset Reason: Software/System restart
[10:08:32][D][debug:257]: Reset Info: Software/System restart
[10:08:32][C][tuya:028]: Tuya:
[10:08:32][C][tuya:039]:   Datapoint 1: switch (value: OFF)
[10:08:32][C][tuya:039]:   Datapoint 11: switch (value: OFF)
[10:08:32][C][tuya:047]:   Datapoint 12: bitmask (value: 1)
[10:08:32][C][tuya:045]:   Datapoint 13: enum (value: 0)
[10:08:32][C][tuya:041]:   Datapoint 14: int value (value: 0)
[10:08:32][C][tuya:045]:   Datapoint 103: enum (value: 2)
[10:08:32][C][tuya:043]:   Datapoint 108: string value (value: )
[10:08:32][C][tuya:045]:   Datapoint 109: enum (value: 0)
[10:08:32][C][tuya:045]:   Datapoint 110: enum (value: 0)
[10:08:32][C][tuya:041]:   Datapoint 111: int value (value: 255)
[10:08:32][C][tuya:053]:   GPIO Configuration: status: pin 14, reset: pin 4 (not supported)
[10:08:32][C][tuya:056]:   Product: 'sryo5vtdwhi1r2h91.0.0'
```

Nardol, 2022-02-18:
It looks like the issue come from the rgb color data point.
If I comment it all is good, maybe if I disable restore state.
But I finally have been able to obtain an information when setting it.
On boot, I have `[11:59:32][D][tuya:263]: Datapoint 14 update to 0` which is strange as this data point should not be updated.

Nardol, 2022-02-18:
In the case it could be useful, i took the opportunity to update the code:
```yaml
substitutions:
  room: Bureau
  sufix: bureau
  timezone: "Europe/Paris"

esphome:
  name: diffuseur-${sufix}
  project:
    name: "Nardol.maxcio_diffuser"
    version: "1.2.2"
  #on_boot:
    #then:
      #- uart.write: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0D]

esp8266:
  board: esp01_1m
  restore_from_flash: true

#web_server:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key
  ap:
    password: !secret ap_key

#captive_portal:

api:
  password: !secret api_pass
  encryption:
    key: !secret encryption_key
  services:
    - service: morning_diffusion
      then:
        - script.execute: morning_diffusion

ota:
  password: !secret ota_pass

uart:
  baud_rate: 9600
  tx_pin: GPIO1
  rx_pin: GPIO3

#debug:

logger:
  baud_rate: 0
  level: ERROR
  #esp8266_store_log_strings_in_flash: False     #  :doc:`recommanded for ESP8266 </components/sensor/custom>`

tuya:
  time_id: my_timer
  on_datapoint_update:
    - sensor_datapoint: 13
      datapoint_type: enum
      then:
        - lambda: if (x>0x0) tuya_tuya->set_enum_datapoint_value(13, 0x0);

binary_sensor:
  - platform: "tuya"
    name: Eau Diffuseur ${room}
    id: eau_diffuseur_${sufix}
    sensor_datapoint: 12
    device_class: "problem"
    on_press:
      then:
        - script.stop: planed_diffusion
        - script.stop: morning_diffusion
        - homeassistant.service:
            service: script.notify_mobile
            data:
              message: "Il n'y a plus d'eau dans le diffuseur du ${sufix}"
              title: "Capteur d'eau du diffuseur du ${sufix}"
              tag: "eau_diffuseur_${sufix}"
        - homeassistant.service:
            service: persistent_notification.create
            data:
              title: "Capteur d'eau du diffuseur du ${sufix}"
              message: "Il n'y a plus d'eau dans le diffuseur du ${sufix}"
              notification_id: "eau_diffuseur_${sufix}"
    on_release:
      then:
        - homeassistant.service:
            service: persistent_notification.dismiss
            data:
              notification_id: "eau_diffuseur_${sufix}"
        - homeassistant.service:
            service: script.notify_mobile
            data:
              message: "clear_notification"
              title: ""
              tag: "eau_diffuseur_${sufix}"

fan:
  - platform: tuya
    name: "Diffuseur ${room}"
    id: diffuseur_${sufix}
    switch_datapoint: 1
    speed_datapoint: 103
    speed_count: 3
    on_turn_off:
      then:
        - script.stop: planed_diffusion
        - script.stop: morning_diffusion
    on_speed_set:
      then:
        - lambda: if (id(diffuseur_${sufix}).speed == 2) {auto call = id(diffuseur_${sufix}).turn_off();call.perform();}

light:
  - platform: tuya
    name: Leds Diffuseur ${room}
    id: leds_diffuseur_${sufix}
    dimmer_datapoint: 111
    switch_datapoint: 11
    min_value: 1
    #rgb_datapoint: 108

# Defining planned diffusion
time:
  - platform: homeassistant
    id: my_timer
    timezone: ${timezone}
    on_time:
      minutes: 0
      hours: 5
      seconds: 0
      then:
        - if:
            condition:
              and:
                 #- binary_sensor.is_off: eau_diffuseur_${sufix}
                - fan.is_off: diffuseur_${sufix}
            then:
              - script.execute: planed_diffusion

script:
  - id: planed_diffusion
    then:
      - fan.turn_on: diffuseur_${sufix}
      - delay: 00:00:01
      - light.turn_off: leds_diffuseur_${sufix}
      - delay:
          hours: 1
      - fan.turn_off: diffuseur_${sufix}
  - id: morning_diffusion
    then:
      - if:
          condition:
            and:
              # - binary_sensor.is_off: eau_diffuseur_${sufix}
              - fan.is_off: diffuseur_${sufix}
          then:
            - fan.turn_on: diffuseur_${sufix}
            - delay: 00:00:01
            - light.turn_off: leds_diffuseur_${sufix}
            - delay:
                minutes: 30
            - fan.turn_off: diffuseur_${sufix}
```

I did not update the first comment as the code I pasted is this I discovered the issue with.

Metus88, 2022-03-30:
@Nardol  Thanks a lot for sharing your code! I use it! It's fantastic!

Nardol, 2022-05-10:
On [this topic about flashed diffusers](https://community.home-assistant.io/t/esphome-smart-oil-diffuser-nightlight/155663/183?u=nardol) this issue is reproduced.

It really looks like this issue appeared since new tuya components I.E. number has been implemented.

Nardol, 2022-05-10:
If someone could investigate this issue but has not the device, I might be able to provide one.
I won't be able to do it myself because soldering is unfeasible for me.

Nardol, 2022-12-26:
I cannot test again for the moment but as soon as I'll be back I'll make some tests again even if I don't think it has been fixed automagically :slightly_smiling_face:

Nardol, 2023-02-21:
Still an issue even with Esphome 2023.3.0-dev, revision 50fbbf2d3ba32f696277529f1ea5413da330ede3
I thought esphome/esphome#4383 might fix this by setting color mode to rgbhsv but it is not the case.

Is it attended that the stall label has not been removed?

Nardol, 2023-08-18:
Stale label still on this issue, fortunately still opened. Is the label presence expected?
For information, still an issue as of today.

Nardol, 2023-12-26:
Now behavior is different, the device is usable but it is not possible to get or control the light color.