esphome/issues #3058: Error compile code which used to work on 2021.9.0

bogdanr, 2022-02-17:
### The problem

After updating from 2021.9.0 to 2022.2.1 the configuration of my irrigation controller doesn't compile anymore.

### Which version of ESPHome has the issue?

2022.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.8

### What platform are you using?

ESP8266

### Board

d1_mini_pro

### Example YAML snippet

```yaml
substitutions:
  project: Sprinklers Controller
  id: sprinklers

esphome:
  name: $id
  platform: ESP8266
  board: d1_mini_pro
  includes:
    - irrigation.h

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $project
    password: !secret wifi_pass

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

globals:
  # ============================================================================= #
  # Irrigation time remaining
  - id: remaining_time1
    type: int
    restore_value: no
    initial_value: "300"
  - id: remaining_time2
    type: int
    restore_value: no
    initial_value: "300"
  - id: remaining_time3
    type: int
[... 671 lines cut ...]
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/unordered_map:46,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:61,
                 from src/esphome/core/component.h:4,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/optional:60:11: note: candidates are: 'template<class _Tp> class std::optional'
   60 |     class optional;
      |           ^~~~~~~~
In file included from src/esphome/core/component.h:7,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/optional.h:36:28: note:                 'template<class T> class esphome::optional'
   36 | template<typename T> class optional {  // NOLINT
      |                            ^~~~~~~~
/config/esphome/sprinkers.yaml:331:61: error: expected '{' before '<' token
  331 |   # Retrieve list of times from the Home Assistant UI.
      |                                                             ^
/config/esphome/sprinkers.yaml:331:62: error: expected primary-expression before 'float'
  331 |   # Retrieve list of times from the Home Assistant UI.
      |                                                              ^    
*** [/data/sprinklers/.pioenvs/sprinklers/src/main.cpp.o] Error 1
========================= [FAILED] Took 17.12 seconds =========================
```

brianhanifin, 2022-02-21:
I am having the same issue, with very similar code. I am running an irrigation controller on a Sonoff 4ch pro. I started by running a "Clean build files" action before recompiling.

I have included a log, and both the irrigation.h as well as my irrigation.yaml.

log:
```
INFO Reading configuration /config/esphome/test.yaml...
WARNING ESP8266: Pin 9 (9-10) might already be used by the flash interface in QUAD IO flash mode.
INFO Generating C++ source...
INFO Compiling app...
Processing irrigation (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/irrigation/.pioenvs/irrigation/src/main.cpp.o
Compiling /data/irrigation/.pioenvs/irrigation/lib67b/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling /data/irrigation/.pioenvs/irrigation/lib67b/ESP8266WiFi/ESP8266WiFiAP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:615:66: error: reference to 'optional' is ambiguous
  615 |   sensor_lambdafilter = new sensor::LambdaFilter([=](float x) -> optional<float> {
      |                                                                  ^~~~~~~~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/node_handle.h:39,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/hashtable.h:37,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/unordered_map:46,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:61,
                 from src/esphome/core/component.h:4,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/optional:60:11: note: candidates are: 'template<class _Tp> class std::optional'
   60 |     class optional;
      |           ^~~~~~~~
In file included from src/esphome/core/component.h:7,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/optional.h:36:28: note:                 'template<class T> class esphome::optional'
   36 | template<typename T> class optional {  // NOLINT
      |                            ^~~~~~~~
src/main.cpp:615:74: error: expected '{' before '<' token
  615 |   sensor_lambdafilter = new sensor::LambdaFilter([=](float x) -> optional<float> {
      |                                                                          ^
src/main.cpp:615:75: error: expected primary-expression before 'float'
  615 |   sensor_lambdafilter = new sensor::LambdaFilter([=](float x) -> optional<float> {
      |                                                                           ^~~~~
/config/esphome/test.yaml:547:47: error: reference to 'optional' is ambiguous
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/node_handle.h:39,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/hashtable.h:37,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/unordered_map:46,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:61,
                 from src/esphome/core/component.h:4,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/optional:60:11: note: candidates are: 'template<class _Tp> class std::optional'
   60 |     class optional;
      |           ^~~~~~~~
In file included from src/esphome/core/component.h:7,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/optional.h:36:28: note:                 'template<class T> class esphome::optional'
   36 | template<typename T> class optional {  // NOLINT
      |                            ^~~~~~~~
/config/esphome/test.yaml:547:55: error: expected '{' before '<' token
/config/esphome/test.yaml:547:56: error: expected primary-expression before 'bool'
/config/esphome/test.yaml:265:47: error: reference to 'optional' is ambiguous
[... 597 lines cut ...]
            }

            // Update the remaining time display.
            if (id(remaining_time2_previous) != id(remaining_time2)) {
              id(irrigation_zone2_remaining).publish_state( (id(remaining_time2)/60) + 1 );
            }
          }

# Time based automations.
time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: America/Los_Angeles
    on_time:
      - seconds: 0
        minutes: /1
        then:
          - lambda: |-
              if (scheduled_runtime(id(irrigation_zone1_next).state.c_str())) {
                id(irrigation_zone1).turn_on();
              }
              if (scheduled_runtime(id(irrigation_zone2_next).state.c_str())) {
                id(irrigation_zone2).turn_on();
              }
```

qurm, 2022-03-28:
Affects me also, using irrigation automation based on Brians original work, after upgrading to ESPHome 2022.2.x.
Not had time to try and reproduce this outside of Home Assistant.

Seems to be related:
https://community.home-assistant.io/t/error-in-lambda-code-since-upgrade/397057/2

```
In file included from src/esphome/core/component.h:7,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/optional.h:36:28: note:                 'template<class T> class esphome::optional'
   36 | template<typename T> class optional {  // NOLINT
      |                            ^~~~~~~~
/config/esphome/irrig-01.yaml:458:61: error: expected '{' before '<' token
  458 |     unit_of_measurement: min
      |                                                             ^
/config/esphome/irrig-01.yaml:458:62: error: expected primary-expression before 'float'
  458 |     unit_of_measurement: min
      |                                                              ^    
/config/esphome/irrig-01.yaml:473:53: error: reference to 'optional' is ambiguous
  473 |     unit_of_measurement: min
      |                                                     ^       
```

mgorven, 2022-04-04:
I had the same problem with a custom CPP component I'd written. The problem was `using namespace std` in my component's header file (and I see that `irrigation.h` also has this). Removing this and using explicit namespaces where required (e.g. `std::string`) allowed it to compile. Having `using namespace std` in the `.cpp` file is fine.

brianhanifin, 2022-04-15:
> I had the same problem with a custom CPP component I'd written. The problem was `using namespace std` in my component's header file (and I see that `irrigation.h` also has this). Removing this and [...]

Thank you @mgorven! Here are the steps I used to correct this issue:

1. Open `Irrigation.h`.
2. Removed line 2 `using namespace std` .
3. Replaced `string` with `std::string`.
4. Replaced `vector` with `std::vector`.

qurm, 2022-04-28:
I also tried this with my irrigation code that was based on a derivative of Brians.  However after fixing the string and vector with `std:: ` I still had similar errors like `error: reference to 'optional' is ambiguous`.

I then saw there was a fix in this thread, https://community.home-assistant.io/t/error-in-lambda-code-since-upgrade/397057/2, to downgrade the Arduino version by adding the arduino_version in my `irrig-01.yaml` file

```
esphome:
  name: ${device}    
  platform: ESP8266
  board: esp01_1m
  arduino_version: 2.7.4
```
This fixes my issue in that I can now compile and install to the device, but I still dont understand the root cause - ie what change in ESPHome or Arduino has caused this.  Will need to investigate further.

qurm, 2022-04-30:
Sounds like the root cause of these issues is this issue with the Platform.io support for ESP32.  There is some on-going work as below, so I guess within a month or so there will be a new Platform.io release, and an ESPHome release.

https://community.platformio.org/t/esp32-c-17-toolchain-missing-std-optional/25850

_Hi guys, just to make it clear, all 2.0.x versions will be added gradually. We started from 2.0.0 to make sure the transition from Arduino framework v1 to v2 will be as smooth as possible as it involves many changes and bugfixes. Stay tuned._

roger-, 2024-05-11:
Experienced the same issue and can confirm that removing `using namespace std` in my custom code did the trick. Works with latest Arduino framework, so no need to downgrade.

closed 2024-05-11 by ssieb