esphome/issues #3059: Disable cover stop button

archerne, 2022-02-17:
### The problem

If no "stop_action" is set in a cover template ESPHome should pass a "supported_features: 3" instead of "supported_features: 11" so that there is no stop button in Home Assistant. 

### Which version of ESPHome has the issue?

2022.2.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.8

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

cover

### Example YAML snippet

```yaml
cover:
  - platform: template
    device_class: garage
    id: "garage_door_cover"
    name: "Garage Door"
    has_position: false
    open_action:
      - switch.turn_on: side_garage_relay
      - delay: 0.1s
      - switch.turn_off: side_garage_relay
      - cover.template.publish:
          id: garage_door_cover
          state: !lambda 'return COVER_OPERATION_OPENING;'
    close_action:
      - switch.turn_on: side_garage_relay
      - delay: 0.1s
      - switch.turn_off: side_garage_relay
      - cover.template.publish:
          id: garage_door_cover
          state: !lambda 'return COVER_OPERATION_CLOSING;'
    assumed_state: false
```

archerne, 2022-02-28:
This might be just setting the supported actions instead of setting supported features directly. Here is how MyQ does it and their integration has no stop button, just up and down. https://github.com/home-assistant/core/blob/dev/homeassistant/components/myq/cover.py

lululombard, 2022-05-08:
I have the exact same issue

poxin13, 2022-08-01:
Just ran into this today too

amomchilov, 2022-10-11:
Hey there!

Total newb here. I took a stab at implementing this. Does anybody want to help me bring it to the fruition?

Here are the PRs:

* In ESP Home: https://github.com/esphome/esphome/pull/3897
* In the Python API client: https://github.com/esphome/aioesphomeapi/pull/276
* In Home Assistant to consume this value: https://github.com/home-assistant/core/pull/80104

poberth, 2023-02-09:
Hi, I am using the cover platform to control a valve but the valve doesn't support a stop function. This issue would make my device work. Are the PRs completed?
Thanks.

amomchilov, 2023-02-10:
I think the third PR depends on the changes from the second one, and I'll probably need a bit of help polishing off the last bit of it.

amomchilov, 2023-06-18:
Woohoo, great success!

[image]

Many thanks to @jesserockz for helping, reviewing and getting this merged!

aeozyalcin, 2023-06-19:
Thanks for taking this on @amomchilov. I just looked into how I can take advantage of set_supports_stop, but looks like the only current way is to create a custom cover sensor, and use this new function in the C++ header file.

I am using a template cover entity in my setup, and I was hoping this would be a simple parameter like `supports_stop: false` in the YAML file. Any thoughts on this? Or can you think of how I can make use of this without having to create a whole new custom cover entity?

amomchilov, 2023-06-19:
It's been a while since I first wrote this code, but from I can tell, there's nothing specific to template covers here.

The Python script reading your YAML will configure the `supports_stop` trait automatically based on whether your cover [has a stop action](https://esphome.io/components/cover/index.html?highlight=stop+action#cover-stop-action).

https://github.com/esphome/esphome/pull/3897/files#diff-f62e02ea76e8404abb6bd2ba45e1d46e91c05cf31ef31d8f477479fa8853d80fR76

If you have no stop action, and you still see the stop button, double check that you're running the latest versions of Home Assistant, ESPHome and the firmware on your device (this feature required updates to all 3 of these).

wernerhp, 2023-08-12:
Another issue that's related to this.  
Cover button states are not dependent on `CoverOperation`

## Problem 1
■ button is always enabled (when `stop_action` is set) regardless of the cover state and operation (even when `COVER_OPERATION_IDLE`)
[image] 
[image] 

■ button is expected to be
- **disabled** when `COVER_OPERATION_IDLE`
- **enabled** otherwise (when `CoverOperation` is either `COVER_OPERATION_OPENING` or `COVER_OPERATION_CLOSING`)

## Problem 2
↑ button is enabled during `COVER_OPERATION_CLOSING`
[image] 

↓ button is enabled during `COVER_OPERATION_OPENING`
[image] 

↑ and ↓ buttons are expected to be
- **enabled** when `COVER_OPERATION_IDLE`
- **disabled** otherwise (when `CoverOperation` is either `COVER_OPERATION_OPENING` or `COVER_OPERATION_CLOSING`)

amomchilov, 2023-08-14:
Hey @wernerhp, good point, the stop button being enabled while idle is odd indeed.

Those are related but quite distinct issues, could you please open up a new issue for them? (This thread is full of people looking to _get rid_ of the stop button, so its state wouldn't be relevant to them.)

wernerhp, 2023-08-14:
https://github.com/esphome/issues/issues/4788