esphome/issues #3060: esp32_touch binary sensors cause esp32 to slow down significantly

fgallardograzio, 2022-02-17:
### The problem

Using more than a couple `esp32_touch` binary sensors causes an esp32 to visibly slow down.
Even transitions or effects on a single light become "discontinuous", taking visible steps between blinking on and off.

Apparently this is because each `esp32_touch` binary sensors adds some ~8ms to the app loop.

-----

I measured loop times by simply logging a message after 2,000 loops were completed, with an offset of 100 loops to give it some time for things to initialize and stabilize.
Then I took the average of four measurements for every new binary sensor I added.

These are the results:

| Number of binary sensors | Loop time (milliseconds) | Loops per second | Time to complete 2,000 loops (seconds) |
|--------------------------|--------------------------|------------------|----------------------------------------|
| 0                        | 17                       | 58.8235          | 34                                     |
| 1                        | 24                       | 41.6667          | 48                                     |
| 2                        | 32                       | 31.25            | 64                                     |
| 3                        | 40                       | 25               | 80                                     |
| 4                        | 48                       | 20.8333          | 96                                     |
| 5                        | 56                       | 17.8571          | 112                                    |
| 6                        | 64                       | 15.625           | 128                                    |
| 7                        | 72                       | 13.8889          | 144                                    |
| 8                        | 80                       | 12.5             | 160                                    |

<details>
<summary>Here's the code I used to log the loop count</summary>

```cpp
// main.cpp

// ... autogenerated code up in here ...

int count = 0;

void loop() {
  if ((count == 100) || (count == 2100)) {
    ESP_LOGW("loop", "Loop!");
  }
  count += 1;

  App.loop();
}
```

</details>

-----

I haven't found any indication that this is expected behavior, if it is, please consider adding a warning or notice in the docs, akin to the logger component warning about how setting the level to `VERY_VERBOSE` will slow down the device, or the light component warning about how setting up too many lights will use too much memory.

Thanks.

### Which version of ESPHome has the issue?

2022.2.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

esp32_touch

### Example YAML snippet

```yaml
esphome:
  name: touch-sensor-latency-test
  platform: ESP32
  board: esp32dev
  
  on_boot:
    - light.turn_on:
        id: light_onboard_led
        effect: pulse

logger:

api:

ota:
  password: "*****"

wifi:
  ssid: "*****"
  password: "*****"

esp32_touch:
  
binary_sensor:
  - platform: esp32_touch
    id: touch_3
    pin: GPIO15
    threshold: 1000
    
  - platform: esp32_touch
    id: touch_0
    pin: GPIO4
    threshold: 1000

  - platform: esp32_touch
    id: touch_4
    pin: GPIO13
    threshold: 1000

  - platform: esp32_touch
    id: touch_5
    pin: GPIO12
    threshold: 1000

  - platform: esp32_touch
    id: touch_6
    pin: GPIO14
    threshold: 1000

  - platform: esp32_touch
    id: touch_7
    pin: GPIO27
    threshold: 1000

  - platform: esp32_touch
    id: touch_8
    pin: GPIO33
    threshold: 1000

  - platform: esp32_touch
    id: touch_9
    pin: GPIO32
    threshold: 1000

output:
  - platform: ledc
    id: output_onboard_led
    pin: GPIO2

light:
  - platform: monochromatic
    id: light_onboard_led
    output: output_onboard_led
    effects:
      - pulse
```

fgallardograzio, 2022-06-21:
Any thoughts on this? :)

mmakaay, 2022-06-24:
Well, I saw this PR last week, and one thing that was striking is the difference in loop time when adding another sensor.
This is shown in the following table:

Number of binary sensors |  Time to complete 2,000 loops (seconds) | Diff with previous
 -- | -- | --
0 | 34 |
1 | 48 | +14
2 | 64 | +16
3 | 80 | +16
4 | 96 | +16
5 | 112 | +16
6 | 128 | +16
7 | 144 | +16
8 | 160 | + 16

This clearly shows how every single sensor adds quite a predictable amount of time to the loop duration.
So there's some complexity going on within the component's loop code that maybe could be improved upon.
With about 8ms extra loop time per sensor, running 8 sensors gives a loop time of 64ms, which is pretty substantial.

I haven't looked into the code, but maybe there's some low hanging fruit there to lose some of this loop time.

closed 2022-12-21 by github-actions[bot]