esphome/issues #3061: Cover Toggle not enabled for Endstop cover

terminusxx, 2022-02-18:
### The problem

Cover.toggle works great on time duration covers but is not enabled on endstop covers. Looking at endstop_cover.cpp it is missing `supports_toggle` under traits and the` if call.get_toggle` block from the control section. Since endstop cover requires a duration for gate position I can't think of a reason why toggle wouldn't work with it. 

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.8

### What platform are you using?

ESP32

### Component causing the issue

endstop_cover

### Example YAML snippet

```yaml
button:
  - platform: template
    name: "Back Gate Toggle"
    on_press:
      - cover.toggle: gate
      
output:
  - platform: gpio
    pin: 17
    id: sh_direction_pin1
    inverted: true
  - platform: gpio
    pin: 16
    id: sh_direction_pin2
    inverted: true

cover:
  - platform: endstop
    name: "Back Gate"
    device_class: gate
    id: gate
    open_action:
      - output.turn_off: sh_direction_pin1
      - output.turn_on: sh_direction_pin2
    open_duration: 27s
    open_endstop: openlimit

    close_action:
      - output.turn_off: sh_direction_pin2
      - output.turn_on: sh_direction_pin1
    close_duration: 27s
    stop_action:
      - output.turn_off: sh_direction_pin1
      - output.turn_off: sh_direction_pin2
    close_endstop: closedlimit
```

rrooggiieerr, 2022-03-28:
I can confirm the Toggle action is not supported on the Endstop Cover. In fact, it's not supported on any of the Cover implementations except the Time-Based Cover.

Why is this action not implemented in the Template Cover in the first place?

rrooggiieerr, 2022-11-17:
Actually I have added the toggle action in the endstop cover at the time and submitted a pull request. It's now merged into ESPHome, so the issue can be closed

makkie2002, 2023-05-24:
This cover toggle function is still not working for “end-stop” type.

makkie2002, 2023-05-24:
This cover toggle function is still not working for “end-stop” type. Version 2023.5.3.

> Actually I have added the toggle action in the endstop cover at the time and submitted a pull request. It's now merged into ESPHome, so the issue can be closed

makkie2002, 2023-05-24:
> Actually I have added the toggle action in the endstop cover at the time and submitted a pull request. It's now merged into ESPHome, so the issue can be closed

This cover toggle function is still not working for “end-stop” type. Version 2023.5.3

rrooggiieerr, 2023-05-24:
Can you elaborate? How are you using it? Show some YAML

makkie2002, 2023-05-24:
> Can you elaborate? How are you using it? Show some YAML

`cover:
  - platform: endstop
    name: "${devicename}"
    id: thiscover
    device_class: shutter
    open_action:
      - if:
          condition:
            lambda: 'return (id(cover_moving).state);'
          then:
            - switch.turn_on: relay1_pulse
            - delay: 1s
      - switch.turn_on: relay1_pulse
    open_duration: 168sec
    open_endstop: cover_open
    close_action:
      - if:
          condition:
            lambda: 'return (id(cover_moving).state);'
          then:
            - switch.turn_on: relay2_pulse
            - delay: 1s
      - switch.turn_on: relay2_pulse
    close_duration: 168sec
    close_endstop: cover_closed
    stop_action:
      - if:
          condition:
            lambda: 'return ((not id(cover_open).state) && (not id(cover_closed).state) && (id(cover_moving).state));'
          then:
            - switch.turn_on: relay1_pulse
sensor:`

rrooggiieerr, 2023-05-24:
Ok, and where are you calling the `cover.toggle` action?

makkie2002, 2023-05-24:
> Ok, and where are you calling the `cover.toggle` action?

ah .... that is in Home Assistant itself:

[image]

rrooggiieerr, 2023-05-24:
The `cover.toggle` action in ESPHome allows the cover to be controlled by a single physical push button connected to the ESP

https://esphome.io/components/cover/index.html#cover-toggle-action

closed 2023-05-27 by terminusxx