esphome/issues #3062: Device is shown in Esphome as online but cannot connect for logging

r100gs, 2022-02-18:
### The problem

Device is schown online in esphome, but I cannot connect wirless for logging.
It has been working fine until I updated it today with esphome 2022.2.2 and 2022.2.3

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.3

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

bme680??

### Example YAML snippet

```yaml
esphome:
  name: dachschlazitemp2
  platform: ESP8266
  board: d1_mini

substitutions:
  friendly_name: "Dach SchlaZi2"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pswd

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "DachschlazitempFbH"
    password: !secret esp_ap_fb

captive_portal:

# Enable logging
logger:
#  level: DEBUG
#  level: VERBOSE  
# Enble Home Assistant API
api:
  password: !secret esp_api_pswd
  encryption:
    key: !secret esp_api_encrypt_key
  
ota:
  password: !secret esp_ota_pswd

#web_server:
#  port: 80
#  auth:
#    username: !secret esp_wsrv_usr
#    password: !secret esp_wsrv_pswd

#deep_sleep:
#  run_duration: 30s
#  sleep_duration: 5min

i2c:
  - id: bus_a
    sda: 04 #D2
    scl: 05 #D1
    scan: True

# Library bsec bme680 python in den config/esphome/custom_components ordner kopieren
# https://github.com/trvrnrth/esphome-bsec-bme680

# neu external component am 08.11.2021 hinzugefügt
#external_components:
#  - source: github://trvrnrth/esphome-bsec-bme680

bme680_bsec:
    # i2c address override (default is 0x76)
    address: 0x77

    # Temperature offset if device is in enclosure and reads too high (default is 0)
    temperature_offset: 0.0

    # Mode for IAQ sensors if device is mobile (default is static)
    iaq_mode: static

    # Interval at which to save BSEC state (default is 6 hours)
    state_save_interval: 6h

    # Sample rate
    # -----------
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: ulp

time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  - platform: uptime
    name: ${friendly_name} Uptime sec
    id: uptime_sec
    update_interval: 60s
  - platform: wifi_signal
    name: "${friendly_name} Wifi"
    update_interval: 60s

## bme680_bsec
  - platform: bme680_bsec
    temperature:
      name: "${friendly_name} Temp"
      id: temp
      sample_rate: lp
      filters:
        - median

    pressure:
      name: "${friendly_name} Luftdruck"
      id: druck
      sample_rate: lp
      unit_of_measurement: "hPa"
      filters:
        - lambda: >-
            return x / powf(1.0 - (415 / 44330.0), 5.255);

    humidity:
      name: "${friendly_name} Luftfeuchtigkeit"
      id: feuchte
      sample_rate: lp
      filters:
        - median

    gas_resistance:
      name: "${friendly_name} CO2 Widerstand"
      id: gas
      filters:
        - median
    iaq:
      name: "${friendly_name} BME680 IAQ"
      id: iaq_range
      filters:
        - median

    co2_equivalent:
      name: "${friendly_name} BME680 CO2 Equivalent"
      filters:
        - median
  
    breath_voc_equivalent:
      name: "${friendly_name} Breath VOC Equivalent"
      filters:
        - median

    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "${friendly_name} BME680 Numeric IAQ Accuracy"

## Berechnungen 
 
 
  - platform: template
    name: "${friendly_name} Absolute Feuchtigkeit"
    lambda: |-
      const float mw = 18.01534;    // molar mass of water g/mol
      const float r = 8.31447215;   // Universal gas constant J/mol/K
      return (6.112 * powf(2.718281828, (17.67 * id(temp).state) /
        (id(temp).state + 243.5)) * id(feuchte).state * mw) /
        ((273.15 + id(temp).state) * r); // in grams/m^3
    accuracy_decimals: 2
    update_interval: 60s
    icon: 'mdi:water'
    unit_of_measurement: 'g/m³'
    
  - platform: template
    name: "${friendly_name} Luftdruck auf Meereshöhe"
    lambda: |-
      const float STANDARD_ALTITUDE = 415; // in meters, see note
      return id(druck).state / powf(1 - ((0.0065 * STANDARD_ALTITUDE) /
        (id(temp).state + (0.0065 * STANDARD_ALTITUDE) + 273.15)), 5.257); // in hPa
    update_interval: 60s
    unit_of_measurement: 'hPa'

  - platform: template
    name: "${friendly_name} Taupunkt"
    lambda: return (243.5*(log(id(feuchte).state/100)+((17.67*id(temp).state)/
     (243.5+id(temp).state)))/(17.67-log(id(feuchte).state/100)-
     ((17.67*id(temp).state)/(243.5+id(temp).state))));
    update_interval: 60s
    unit_of_measurement: '°C'
    
binary_sensor:
- platform: status
  name: "${friendly_name} Temp Status"
    
text_sensor:
#  - platform: wifi_info
#    ip_address:
#      name: ESP IP Address
#    ssid:
#      name: ESP Connected SSID
#    bssid:
#      name: ESP Connected BSSID
#    mac_address:
#      name: ESP Mac Wifi Address

# bme680_sec      
  - platform: bme680_bsec
    iaq_accuracy:
      name: "${friendly_name} BME680 IAQ Accuracy"
      
      
  - platform: template
    name: ${friendly_name} Uptime
    lambda: |-
      int seconds = (id(uptime_sec).state);
      int days = seconds / (24 * 3600);
      seconds = seconds % (24 * 3600); 
      int hours = seconds / 3600;
      seconds = seconds % 3600;
      int minutes = seconds /  60;
      seconds = seconds % 60;
      return { (String(days) +"d " + String(hours) +"h " + String(minutes) +"m "+ String(seconds) +"s").c_str() };
    icon: mdi:clock-start
    update_interval: 65s
    
    
    
  - platform: template
    name: "${friendly_name} Luftqualität"
    icon: "mdi:air-filter"
    lambda: |-
   
      ESP_LOGD("main", "Aktueller Wert %d", id(iaq_range));
      
      if ((id(iaq_range).state <= 50.0 )) {
        return {"sehr gut"};
      }
      else if ((id(iaq_range).state > 50.00)  && (id(iaq_range).state <= 100.0)) {
        return {"gut"};
      }
      else if ((id(iaq_range).state > 100.0) && (id(iaq_range).state <= 150.0)) {
        return {"Leicht verschmutz"};
      }
      else if ((id(iaq_range).state > 150.0) && (id(iaq_range).state <= 200.0)) {
        return {"mäßig verschmutz"};
      }
      else if ((id(iaq_range).state > 200.0) && (id(iaq_range).state <= 250.0)) {
        return {"stark verschmutzt"};
      }
      else if ((id(iaq_range).state > 250.0) && (id(iaq_range).state <= 350.0)) {
        return {"extrem verschmutz"};
      }
      else if ((id(iaq_range).state > 350.00)) {
        return {"gefährlich verschmutzt"};
      }
      return {};
    update_interval: 60s
```

Jenda987, 2022-02-18:
I have similar issue with BME280.
From OS console it's working (esphome logs ....).
When I will try to connect from ESPHome dashboard then it's not accessible and OS console log stop responding for a while with these messages ...

[13:24:33][D][text_sensor:067]: 'xxxx homeassistant current time': Sending state '2022-02-18 13:24:32'
INFO esphome.address.local: Error while reading incoming messages: Error while reading data: [WinError 64] The specified network name is no longer available
INFO Disconnected from ESPHome API for esphome.address.local
WARNING Disconnected from API
INFO esphome.address.local: Ping Failed: Error while reading data: [WinError 64] The specified network name is no longer available
INFO Successfully connected to esphome.address.local
[13:25:22][D][sensor:124]: 'xxxx BME280 temperature': Sending state 5.17000 °C with 1 decimals of accuracy

rmeissn, 2022-02-22:
Related issue: https://github.com/esphome/issues/issues/3083
Seems the BME680 causes this problem.

amkochaki, 2022-04-30:
I have the same problem. If I add one entity to esp32 the integration works when I add the second one it loses the integration, but it stays online and I can upload it. I have already replaced the board and the problem persists. I have other EPS32 running normally but it seems that since the last update something is wrong

the-smart-home-maker, 2022-07-30:
I have the same problem. My esphome device (ESP32) worked for a very long time perfectly. I was able to update via OTA and add functionality. Now since some of the last changes, all of a sudden the problem described in this thread appeared. The device is shown as "online" in the esphome dashboard and I can also still send firmware updates via OTA. But it can not connect to the device for logging and also does not send data to Home Assistant any more.

MisterSoandSo, 2022-12-22:
Is there no update on this issue?

davidjac, 2023-02-15:
I am getting this constantly.

GRClark, 2023-02-16:
There hasn’t been a time I’ve tried to check logging on my ESP8266’s where I haven’t gotten this same message.  Shows as online and can view device under web server but still unsure if it’s actually online/working as I haven’t added sensors to them yet.

loca5790, 2023-03-05:
I'm having the same issue on a brand new ESP32 steup.  Been struggling, it connects and I can flash but it will not bring in any sensors including text sensors.  When I updated it is not prompting me with no route to host and will not flash OTA.  Despite it showing as online and up to date with esphome.

I flashed it via download and a flashing tool.  Once it flashed the log started and updates rolled.  I then had to force integrate it but still have 0 sensors or entities other than up to date configuration.

Foamier, 2024-01-21:
Been troubleshooting [Errno 111] as well.

Ping works to the WiFi IP, although with high response times. Also booting takes a long time before it responds to ping over the WiFi. It likely reboots a few times, but is hard to see because I cannot do serial logging since I need the UART pins for other things.

In all cases this was resolved by adding a capacitor of sufficient capacity close to the ESP8266/ESP01. 680uF or more seems to do it in my case. E.g. for the ESP01 modules there is a capacitor of 1000uF on the programming dingle to alleviate the current surges.
You could also use a (better) power supply.

After providing this buffer for peak currents, the ESPs boot fast and ESPHome is able to connect as well without problems.

lishan89uc, 2024-04-16:
@Foamier can you link some pictures i like to give this a try. I have a few esp32’s that are doing the same thing.

dking484, 2024-09-18:
im also having the same problem. has anyone figured out how to resolve this? it was working a few days ago. now i can see my sensors. and the results, but i cannot access the web interface at all