esphome/issues #3064: tinypico web interface not showing status

murraythegoz, 2022-02-18:
### The problem

I just started using tinypico board, which is based on ESP-pico-D4.

When compiling on windows 10 + python 3.9, everything compiles with almost no error
with two exceptions: 
1) hash package is not installed
```
Library Manager: AsyncTCP-esphome @ 1.2.2 is already installed
Library Manager: Installing Hash
Library Manager: Warning! Could not install dependency {'name': 'Hash', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome'
Library Manager: Installing ESP8266WiFi
Library Manager: Warning! Could not install dependency {'name': 'ESP8266WiFi', 'platforms': ['espressif8266']} for package 'ESPAsyncWebServer-esphome' 
```
2) a compile error related to HAL-SPI
```
C:\Users\ivan.brunello\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\ivan.brunello\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\ivan.brunello\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

```

Uploading seems to work correctly

Sensor table is shown but sensor values are blank.

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A the local web server is not working

### What platform are you using?

ESP32

### Board

tinypico

### Component causing the issue

web server

### Example YAML snippet

```yaml
esphome:
  name: windows_sensors_hub

esp32:
  board: tinypico
  framework:
      type: arduino
....

web_server:
  port: 80
  auth:
    username: !secret web_user
    password: !secret web_password
....
binary_sensor:
  - platform: status
    name: "windows_sensors_hub_Status"
  - platform: gpio
    name: "window1_chiusura"
    pin: 
      number: GPIO33
    device_class: window
    filters:
    - invert:

....
text_sensor:
  - platform: version
    name: "windows_sensors_hub_esphome_version"
    hide_timestamp: true
  - platform: wifi_info
    ip_address:
      name: "windows_sensors_hub_IP"
      icon: "mdi:ip-network"
    mac_address:
      name: "windows_sensors_hub_mac_address"
      icon: "mdi:ethernet"
```

murraythegoz, 2022-02-19:
I meant that the Home assistant version is not related.
I connect to the web server correctly but the web server is not showing properly.
I.e. all  sensor data are not populated.

This is the output.

debug log show states going up and down if i trigger them, so the GPIO mapping seems to be correct.

[image]

OttoWinter, 2022-02-19:
Ok, thanks

Next:
1. What browser are you using?
2. If you open the browser console (also called developer console), can you see any warnings?

mb-software, 2022-02-19:
I updated my ESP32Cam node today and I have the same problem: No states shown in the webserver anymore.
Browser Dev tools show this error:
```
Uncaught TypeError: Cannot read properties of null (reading 'children')
    at EventSource.<anonymous> (webserver-v1.min.js:1:709)
(anonym) @ webserver-v1.min.js:1
```

murraythegoz, 2022-02-20:
same here:

GET http://100.64.50.147/favicon.ico 500 (Internal Server Error)
15webserver-v1.min.js:1 Uncaught TypeError: Cannot read properties of null (reading 'children')
    at EventSource.<anonymous> (webserver-v1.min.js:1:709)
(anonymous) @ webserver-v1.min.js:1

OttoWinter, 2022-02-20:
@kbickar You recently [refactored](https://github.com/esphome/esphome-docs/pull/1867/files#diff-a80bea55876230bab243fcafe2d263159362643dc8ab56dece06e89179c4d1c8) the `webserver-v1.js` codebase, and this issue seems related to that.

Could you take a look at this issue and see if you spot anything? Thanks!

kbickar, 2022-02-20:
Looks like that's coming from one of the only lines I didn't change: `document.getElementById(data.id).children[1].innerText = data.state;`

It seems the id isn't being found in the HTML - can you check the source for the page and the event message to see if the event id is found?

murraythegoz, 2022-02-21:
Could you please elaborate?

you want me to send you the html rendered page?

Thank you

m.

kbickar, 2022-02-21:
Yes please, and the event from the dev console on the network tab

jesserockz, 2022-02-21:
The state events in the eventstream are blank objects: `{}`

I have seen the same issue when using a device that has psram: https://discord.com/channels/429907082951524364/943356291177410581

I think it has to do with the ArduinoJson changes, but its weird because the logs come through just fine, just not the "state" type events

mb-software, 2022-02-21:
I can confirm this, event objects are empty.

murraythegoz, 2022-02-22:
same here. events are empty

murraythegoz, 2022-03-01:
shall we try with generic esp32dev?

that is:
`esp32:
  board: esp32dev`

correct?

murraythegoz, 2022-03-01:
set as esp32dev did the trick.

will check for peripherals in the next couple days.

closed 2022-03-02 by jesserockz

mb-software, 2022-03-03:
@jesserockz 
Unfortunately this issue has not been fixed for me with 2022.2.6. I did a clean build and update to 2022.2.6, but still the state event data objects are empty on my esp32cam board. Switching to esp32dev unfortunately is not a solution when using esp32 camera.

cybergurke, 2022-03-12:
After update to 2022.2.6 state events are empty on esp32, 2022.2.5 was fine.