esphome/issues #3065: Sensors still showing up in logs after removed from yaml file

kjoglum, 2022-02-18:
### The problem

I am running ESPHome on a RPi 4, having a ESP32 connected with a pulse_meter and dallas sensors (DS18B20). The ESP32 is configured via the Dashboard (yaml file), and is set up for MQTT. I have a Mosquitto broker on the RPi publishing the topics received from the ESP32 to openHAB. Everything has basically been working ok.

However, as part of adding some dallas sensors (DS18B20), following the guide from ESPHome, establishing a dallas hub, getting addresses and configuring sensors I have run into some problems. The initial dallas sensor I created I want to completely remove/delete, but it still shows up in logs after removing the sensor part from the yaml file. The dallas sensor has only been created internally, no further Mosquitto publishing / openHAB interaction. I have also tried deleting the device from dashboard and also deleted the ESPHome folder on the RPi and reinstalled ESPHome, including flashing memory for the ESP32, but the initial sensor still shows up.

I also see from Dashboard mainpage, that I have the v2022.2.3 version installed, but when running a fresh install for the ESP32,  the log shows ESPHome version 2022.1.2. 

Maybe some files in other directories I am overlooking, which needs to be deleted?

For the sensor I want to completely remove/delete, the dallas log reference is given below (the 'VVB Lower Temperature' is the sensor which is removed from the yaml file, which still is recognised and gets state updates published to MQTT).

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

Wemos D1 ESP32

### Example YAML snippet

```yaml
esphome:
  name: vbvarme-energy
  on_boot:
    then:
      - pulse_meter.set_total_pulses:
          id: vvbenergy_instant
          value: 9491590
          
esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

ota:
  password: "4b97b28e588d3e0756e3a9d87016ff1e"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.130.161
    gateway: 192.168.130.1
    subnet: 255.255.255.0
    
  power_save_mode: none

mqtt:
  topic_prefix: vvbenergy
  discovery: false
  broker: 192.168.130.199
  port: 1883

dallas:
  - pin: 
      number: GPIO13
      mode: INPUT_PULLUP
    id: dallas_id

sensor:
  - platform: pulse_meter
    pin: 
      number: GPIO12
      mode: INPUT_PULLUP
    unit_of_measurement: 'kW'
    name: 'VVB - Electricity Usage'
    id: vvbenergy_instant
    internal_filter: 100ms
    accuracy_decimals: 3
    timeout: 30 sec
    filters:
      - multiply: 0.06
    
    total:
      name: "VVB - Electricity Total"
      id: vvbenergy_total
      unit_of_measurement: "kWh"
      accuracy_decimals: 3
      filters:
        - multiply: 0.001
```

### Anything in the logs that might be useful for us?

```txt
[23:56:27][C][dallas.sensor:070]: DallasComponent:
[23:56:27][C][dallas.sensor:071]:   Pin: GPIO13
[23:56:27][C][dallas.sensor:072]:   Update Interval: 30.0s
[23:56:27][D][dallas.sensor:077]:   Found sensors:
[23:56:27][D][dallas.sensor:079]:     0xa301203c20c66d28
[23:56:27][D][dallas.sensor:079]:     0xd301203c3b97ef28
[23:56:28][C][dallas.sensor:084]:   Device 'VVB Lower Temperature'
[23:56:28][C][dallas.sensor:084]:     Device Class: 'temperature'
[23:56:28][C][dallas.sensor:084]:     State Class: 'measurement'
[23:56:28][C][dallas.sensor:084]:     Unit of Measurement: '°C'
[23:56:28][C][dallas.sensor:084]:     Accuracy Decimals: 1
[23:56:28][C][dallas.sensor:092]:     Address: 0xd301203c3b97ef28
[23:56:28][C][dallas.sensor:093]:     Resolution: 12
```

OttoWinter, 2022-02-20:
Just to make sure, did you press the install button in the dashboard after you made the yaml changes? the device doesn't magically receive those, they have to be uploaded to it first

kjoglum, 2022-02-20:
Yes, always pressing "clean build files" and "install" as part of yaml updates. The strangest thing is the fact that the dallas component shows up in log (as shown above) even after deleting ESPHome folder / reinstalling ESPHome. Seems to be some cached files in other directories, related to MQTT..?

OttoWinter, 2022-02-20:
Ok, especially after a clean, there are no cached files it could mistakenly re-use (even after not doing one it shouldn't be an issue).

I think the device is simply not completing the OTA completely - probably the bootloader doesn't accept the new partition and so rolls back to the old build still stored on the device.

If possible, please try connecting to the device via a USB cable and viewing the logs while doing an _OTA_ update (I'm guessing shortly after the reboot it will show an error in the logs). Also, with what software (i.e. esphome-flasher, web.esphome.io, etc) did you initially flash the device?

kjoglum, 2022-02-20:
Appreciate the guidance.

I have the ESP32 connected via USB to RPi (i.e. getting power supply from the RPi). Initial installation was carried out via ESPHome dashboard/USB, while later clean installs have been carried out OTA (for some reason experiencing faster response OTA compared to USB, which might support your theory of incomplete OTA). 

As mentioned, the RPi also is set up as MQTT broker (Mosquitto). And I did the initial dallas sensor installation following the ESPHome guidelines (hub --> sensor address). Experiencing problems with the dallas component I wanted to start fresh, and when the dallas component appeared in logs after yaml changes (removing dallas reference), I tried both deleting the device from ESPHome dashboard, as well as removing ESPHome folder from the RPi/reinstalling ESPHome without any luck. Also tried factory reset/flash of the ESP32 using esptool, still without any luck.

Becoming impatient, suspecting issues related to cached MQTT items, I ended up doing a fresh install of the RPi. Now, after a fresh install of my RPi, a fresh Mosquitto install (RPi) a fresh ESPHome/yaml install, everything is working as expected.

Following your comment, having the device connected via RPi USB, I will continue using USB installs rather than OTA installs for potential future yaml updates.

closed 2022-07-05 by github-actions[bot]