esphome/issues #3067: New ESPHome update on Feb 16 is too big to fit on Shelly 1  via OTA

AlohaFlyer, 2022-02-19:
### The problem

The new version of ESPHome on Feb 16 throws this error when doing OTA updates to my Shelly 1 and Shelly 2.5.

"ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)"

Updating via USB would be very challenging for my 60 Shellys.

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

Shelly

### Component causing the issue

arduino; platform: platformio/espressif8266 @ 3.2.0)

### Example YAML snippet

```yaml
esphome:
  platform: ESP8266
  board: esp01_1m
```

### Anything in the logs that might be useful for us?

```txt
Update fails to install and throws this error:

"ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)"
```

AlohaFlyer, 2022-02-20:
The best solution I have found, as a work around. Is to download the newly compiled bin file. gzip it, then load it. It was small enough to work that way. Maybe we need to consider building gzip into the ESPHome update process in Home Assistant to stream line that.

OttoWinter, 2022-02-20:
Likely a duplicate of https://github.com/esphome/issues/issues/3049

closed 2022-07-05 by github-actions[bot]