esphome/issues #3068: Gain for TSL2591 ignored unless set to 'auto'

jkfranks9, 2022-02-20:
### The problem

When using the TSL2591 sensor with a specified gain, such as 'medium', the setting is ignored. This results in very low lux readings. Changing the setting to anything other than 'auto' results in the identical low readings.

The problem is in TSL2591Component::setup() in tsl2591.cpp. The gain_ variable is only set if gain 'auto' was specified. This (unset) variable is then passed to set_integration_time_and_gain().

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

tsl2591

### Example YAML snippet

```yaml
# Device specific information
substitutions:
  device_name:        office-sensor
  friendly_name:      Office
  prefix:             office
  loglevel:           DEBUG
  update_interval:    600s
  window_interval:    60s
  window_size:        "10"
  calibrate_measured: "71.0"
  calibrate_truth:    "65.0"

# Pull in common configuration
packages:
  nodemcu_base:   !include common/nodemcu_base.yaml
  wifi:           !include common/wifi.yaml
  bme280_sensor:  !include common/bme280_sensor.yaml
  tsl2591_sensor: !include common/tsl2591_sensor.yaml

# Enable Home Assistant API
api:

# Enable I2C, needed for the BME280 and TSL2591 sensors
i2c:
  sda: D2
  scl: D1
  scan: True

--------------- tsl2591_sensor.yaml -------------------

# Adafruit TSL2591 ambient light sensor
sensor:
  - platform: tsl2591
    #integration_time: 600ms
    gain: medium
    update_interval: $window_interval
    power_save_mode: false
    visible:
      name: ${friendly_name} Ambient Light
      unit_of_measurement: lux
      filters:
        - filter_out: 65535
```

wjcarpenter, 2022-02-20:
I guess this must be a regression in the recent tsl2591 update since it added the "auto" setting. Perhaps the author of that update (@azimath) will fix it. Meanwhile, a workaround would be to use the previous generation of the tsl2591 code.

Azimath, 2022-02-21:
I'll have a fix sometime in the next 24 hours

Azimath, 2022-02-21:
Created a PR for a fix:
https://github.com/esphome/esphome/pull/3249

closed 2022-02-22 by jesserockz

jkfranks9, 2022-02-22:
Thanks for the quick turnaround!