esphome/issues #3070: Unable to compile ESP32 boards

jeffborg, 2022-02-21:
### The problem

Seems some standard c++ libs are missing in esp32

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

Generic ESP32

### Component causing the issue

core

### Example YAML snippet

```yaml
# any esp32 board will cause this

esphome:
  name: esp32test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "20bc381de5e44446eac8ff7fd5f5db23"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "0zBMioaWSoVn"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esp32test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing esp32test (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling .pioenvs/esp32test/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/api/api_connection.h:3:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/component.h:3:18: fatal error: string: No such file or directory
compilation terminated.
Compiling .pioenvs/esp32test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp32test/src/esphome/components/api/api_pb2.cpp.o
In file included from src/esphome/components/api/api_frame_helper.cpp:1:0:
src/esphome/components/api/api_frame_helper.h:2:19: fatal error: cstdint: No such file or directory
compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/esp32test/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
Compiling .pioenvs/esp32test/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/api/proto.h:3:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/component.h:3:18: fatal error: string: No such file or directory
compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_pb2.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:3:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/component.h:3:18: fatal error: string: No such file or directory
compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================== [FAILED] Took 4.32 seconds ==========================
```

### Additional information

All esp8266 boards I can compile with 2022.2.4 but none of the esp32's will build

nagyrobi, 2022-02-21:
Works for me...
Maybe the same as https://github.com/esphome/issues/issues/3072 ?

jeffborg, 2022-02-22:
on x86 host getting a different error

```
INFO Reading configuration /config/esp32test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32test (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling .pioenvs/esp32test/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp32test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp32test/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esp32test/src/esphome/components/api/api_pb2_service.cpp.o
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/wchar.h:6:0,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/cwchar:44,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/postypes.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/char_traits.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/string:40,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/sys/reent.h:14:20: fatal error: stddef.h: No such file or directory

****************************************************************
* Looking for stddef.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stddef.h"
* Web  > https://registry.platformio.org/search?q=header:stddef.h
*
****************************************************************

compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/wchar.h:6:0,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/cwchar:44,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/postypes.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/char_traits.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/string:40,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/sys/reent.h:14:20: fatal error: stddef.h: No such file or directory

****************************************************************
* Looking for stddef.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stddef.h"
* Web  > https://registry.platformio.org/search?q=header:stddef.h
*
****************************************************************

compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_pb2.cpp.o] Error 1
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/wchar.h:6:0,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/cwchar:44,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/postypes.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/char_traits.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/string:40,
                 from src/esphome/core/component.h:3,
[... 14 lines cut ...]
compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/wchar.h:6:0,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/cwchar:44,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/postypes.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/char_traits.h:40,
                 from /config/.esphome/platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/string:40,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
[... cut ...]
****************************************************************
* Looking for stddef.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stddef.h"
* Web  > https://registry.platformio.org/search?q=header:stddef.h
*
****************************************************************

compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 2.46 seconds ==========================

```

jeffborg, 2022-02-22:
removed the entire `/config/.esphome/platformio` folder now getting different error

```
INFO Reading configuration /config/esp32test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32test (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/config/.esphome/platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/config/.esphome/platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing WiFi
Library Manager: Warning! More than one package has been found by WiFi requirements:
 - arduino-libraries/WiFi @ 1.2.7
 - mbed-zrem/WiFi @ 0.0.0+sha.a446a6e2d3ce
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts

Unpacking  [------------------------------------]    0%
Unpacking  [#-----------------------------------]    3%
Unpacking  [##----------------------------------]    6%  00:00:42
Unpacking  [###---------------------------------]    9%  00:00:41
Unpacking  [####--------------------------------]   12%  00:00:39
Unpacking  [#####-------------------------------]   15%  00:00:38
Unpacking  [######------------------------------]   18%  00:00:36
Unpacking  [#######-----------------------------]   21%  00:00:35
Unpacking  [#########---------------------------]   25%  00:00:34
Unpacking  [##########--------------------------]   28%  00:00:32
Unpacking  [###########-------------------------]   31%  00:00:31
Unpacking  [############------------------------]   34%  00:00:29
Unpacking  [#############-----------------------]   37%  00:00:28
Unpacking  [##############----------------------]   40%  00:00:26
Unpacking  [###############---------------------]   43%  00:00:25
Unpacking  [################--------------------]   46%  00:00:24
Unpacking  [##################------------------]   50%  00:00:22
Unpacking  [###################-----------------]   53%  00:00:21
Unpacking  [####################----------------]   56%  00:00:19
Unpacking  [#####################---------------]   59%  00:00:18
Unpacking  [######################--------------]   62%  00:00:17
Unpacking  [#######################-------------]   65%  00:00:15
Unpacking  [########################------------]   68%  00:00:14
Unpacking  [#########################-----------]   71%  00:00:12
Unpacking  [###########################---------]   75%  00:00:11
Unpacking  [############################--------]   78%  00:00:09
Unpacking  [#############################-------]   81%  00:00:08
Unpacking  [##############################------]   84%  00:00:07
Unpacking  [###############################-----]   87%  00:00:05
Unpacking  [################################----]   90%  00:00:04
Unpacking  [#################################---]   93%  00:00:02
Unpacking  [##################################--]   96%  00:00:00
Unpacking  [####################################]  100%          
Library Manager: WiFi @ 1.2.7 has been installed!
Library Manager: Installing FS
Warning! Could not find the package with 'FS' requirements for your system 'linux_x86_64'
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_x86_64'
Library Manager: Installing DNSServer

Unpacking  [------------------------------------]    0%
Unpacking  [###---------------------------------]    8%
Unpacking  [######------------------------------]   16%
Unpacking  [#########---------------------------]   25%
Unpacking  [############------------------------]   33%
Unpacking  [###############---------------------]   41%
Unpacking  [##################------------------]   50%
Unpacking  [#####################---------------]   58%
Unpacking  [########################------------]   66%
[... 61 lines cut ...]
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_pb2.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:4:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/log.h:14:21: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

compilation terminated.
*** [.pioenvs/esp32test/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================= [FAILED] Took 17.69 seconds =========================
```

nagyrobi, 2022-02-23:
Remove the addon too, restart HA, reinstall...

jeffborg, 2022-02-23:
> Remove the addon too, restart HA, reinstall...

Just using plan docker in Kubernetes. So have restarted. Cleared platformio folder ran on both arm64 and x86_64. What I haven't tried is downgrading and see if the issues go away

jeffborg, 2022-02-24:
Just updated to 2022.2.5 and was able to compile all boards on arm64 host
I did end up having to remove the /config/.esphone/platformio folder

closed 2022-02-24 by jeffborg