esphome/issues #3071: Parsing JSON failed error on version 2022.2.3

fpovoski, 2022-02-21:
### The problem

I have two exact same devices (two CWWW lights) and upgraded on to Version 2022.2.3 from 2021.10.3 and now I am getting a "Parsing JSON failed" on the upgraded device. The JSON I am sending via MQTT is:
{
  "color_mode": "cwww",
  "state": "ON",
  "brightness": 255,
  "color": {
    "c": 0,
    "w": 255
  }
}
This was perfectly fine with all prior versions except 2022.2.3, so I think you  introduced a "BUG" into this version.

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

MQTT/NODE-RED

### What platform are you using?

ESP32

### Board

ESP32Cam

### Component causing the issue

Your JSON parser

### Example YAML snippet

```yaml
esphome:
  name: family1
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "hill"
  password: "trek6666"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "family1 Fallback Hotspot"
    password: "trek6666"

captive_portal:

mqtt:
  broker: 192.168.0.100
#api:
#  password: "trek6666"

web_server:
  port: 80

# Enable logging
logger:
  level: VERBOSE

ota:
  password: "trek6666"
  
switch:
  - platform: restart
    name: "family1 Restart"
  
sensor:
  - platform: wifi_signal
    name: "family1 WiFi"
    update_interval: 30s
  - platform: uptime
    name: "family1 Uptime"
    update_interval: 30s
    
#binary_sensor:
# - platform: gpio
#    pin:
#      number: GPIO4
#      mode: INPUT
#      inverted: False
#    name: "family1 Button"

text_sensor:
  - platform: template
    name: "Fam1 Device Sensor"
    lambda: |-
      return {"ESP32 Cam"};
    update_interval: 600s
  - platform: version
    name: "family1 Version"
  - platform: wifi_info
    ip_address:
      name: "family1 IP Address"
    ssid:
      name: "family1 Connected SSID"
    bssid:
      name: "family1 Connected BSSID"
    mac_address:
      name: "family1 Mac Wifi Address"

status_led:
  pin:
    number: GPIO02
    inverted: yes

output:

  - platform: ledc
    id: white_channel
    frequency: "1220"
    pin:
      number: GPIO12
      inverted: False
      
  - platform: ledc
    id: soft_channel
    frequency: "1220"
    pin:
      number: GPIO13
      inverted: False

light:
  - platform: cwww
    name: "family1 light"
    default_transition_length: 2s
    cold_white: white_channel
    warm_white: soft_channel
    cold_white_color_temperature: 7000 K
    warm_white_color_temperature: 2000 K
    constant_brightness: false
    restore_mode: RESTORE_DEFAULT_ON
```

### Anything in the logs that might be useful for us?

```txt
Mon Feb 21 2022 01:08:09 GMT-0500 (Eastern Standard Time):  [W][json:055]: Parsing JSON failed.

Mon Feb 21 2022 01:08:11 GMT-0500 (Eastern Standard Time):  [W][json:055]: Parsing JSON failed.

Mon Feb 21 2022 01:08:13 GMT-0500 (Eastern Standard Time):  [W][json:055]: Parsing JSON failed.

Mon Feb 21 2022 01:08:14 GMT-0500 (Eastern Standard Time):  [W][json:055]: Parsing JSON failed.
```

### Additional information

Example JSON sent:
{
  "state": "ON",
  "brightness": "255"
}

Is there an issue with the number being in quotes?

fpovoski, 2022-02-22:
I found the issue.  When compiling using the board: esp32cam, JSON DOES NOT WORK.  I changed the board type to generic esp32dev , recompiled, and IT WORKED.  Due to the smaller form factor I sometimes use the ESP32 Camera board without the camera for projects.

mb-software, 2022-02-28:
I guess it is the same problem like non working web server on boards with psram, see here: #3064

closed 2022-12-18 by github-actions[bot]