esphome/issues #3072: ESP8266 update error after version 2022.2.3

dsimop, 2022-02-21:
### The problem

I have a smart switch device that is flashed with ESPHome and is running stable, getting all the updates.
Since 2022.2.3, I get the following error during the update process (wireless).
You can have a look at the respective logs.

### Which version of ESPHome has the issue?

2022.2.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/myDevice.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing myDevice (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/myDevice/.pioenvs/myDevice/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 3.46 seconds ==========================
```

klaxonheroku, 2022-02-21:
@dsimop Were you able to resolve the issue yet?

dsimop, 2022-02-21:
No, the issue can be reproduced by simply trying to update all devices through the ESPHome Interface.

klaxonheroku, 2022-02-21:
@dsimop i also get the error when setting up a new device – also via usb directly on my raspi... hence no solution so far

johannes281, 2022-02-21:
xtensa-lx106-elf-g++: fatal error: cannot execute 'as': execvp: No such file or directory

I have an error too regarding xtensa-lx106-elf-g++ while compiling. It isn't possible at the moment with 
2022.2.4

nagyrobi, 2022-02-21:
try renaming the .yaml and/or change the name of the node in it

klaxonheroku, 2022-02-21:
@nagyrobi at least in my case this didn't help (renamed fornuftig to fornuftig2 and also renamed the .yaml from fornuftig.yaml to fornuftig2.yaml
[image]

nagyrobi, 2022-02-21:
I just updated all my nodes to 2022.2.4 one by one and didn't run into this. There should be an issue with ESPHome installation on your system, it's missing some required files. 

Perhaps make a backup of all your configs, remove the ESPHome Add-on complelely, delete the entire esphome directory from HA's config share, restart HA, re-install the addon and put back the yaml configs.

klaxonheroku, 2022-02-22:
@nagyrobi this solved it for me! 

I copied my /config/esphome folder to the desktop
removed the esphome Add-on via HA
rebooted HA
reinstalled the esphome Add-on via HA
and copied over all my files from before

now i can successfully update... beware maybe this was my issue before... be patient while updating this took me well over 10 minutes per node... also if the log wont spit out anything for e few minutes, it's still updating... it will get back to you with a status update once it's complete or failed...

dsimop, 2022-02-22:
The issue is solved after reinstalling the add-on, as it re-downloads the necessary libraries.

closed 2022-02-22 by dsimop