esphome/issues #3074: api.connection,connection closed ,when "on_code_received" event be triggered

shzlww, 2022-02-21:
### The problem

I use a sonoff rf bridge (R2 v1.0 - Black model) to control RF433 devices, it works well with esphome old version-2021.12.x. But the new 2022.2.x version can't work well. the error log is :
[I][rf_bridge:055]: Received RFBridge Code: sync=0x2E7C low=0x017C high=0x0488 code=0xC94302
[W][api.connection:083]: Home Assistant 2022.2.8 (192.168.1.2): Connection closed
[I][ota:105]: Boot seems successful, resetting boot loop counter.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

core-2022.2.9

### What platform are you using?

ESP8266

### Board

sonoff rf bridge (r2 v1.0)

### Component causing the issue

rf_bridge

### Example YAML snippet

```yaml
substitutions:
  node_name: "rfbridge-livingroom"
  node_id: rfbridge_livingroom
  node_name_friendly: ${node_name}

packages:
  common: !include common/common.yaml

esphome:
  on_boot:
    priority: -100
    then:
      #初始化晾衣架的灯为关闭状态
      - rf_bridge.send_raw:
          raw: "AAB029030800F002801AF428190819081909090819090818181818181908181819081818181908181818181855"
      - switch.turn_on: ${node_id}_pair_switch
#默认重启之后 灯是关闭状态，应该无需重新发布状态
#      - lambda: |-
#          id(${node_id}_cover_light).publish_state(false);
api:
  services:
    - service: send_rf_code
      variables:
        sync: int
        low: int
        high: int
        code: int
      then:
        - rf_bridge.send_code:
            sync: !lambda 'return sync;'
            low: !lambda 'return low;'
            high: !lambda 'return high;'
            code: !lambda 'return code;'

    - service: send_rf_raw
      variables:
        raw: string
      then:
        - rf_bridge.send_raw:
            raw: !lambda 'return raw;'

    - service: learn
      then:
        - rf_bridge.learn

# with Portisch firmware
    - service: beep
      then:
        - rf_bridge.beep:
            duration: 100

    - service: send_rf_advanced_code
      variables:
        length: int
        protocol: int
        code: string
      then:
        - rf_bridge.send_advanced_code:
            length: !lambda 'return length;'
            protocol: !lambda 'return protocol;'
            code: !lambda 'return code;'

    - service: learn_raw
      then:
        - rf_bridge.start_bucket_sniffing

    - service: learn_advanced_code_start
      then:
        - rf_bridge.start_advanced_sniffing

    - service: learn_advanced_code_stop
      then:
        - rf_bridge.stop_advanced_sniffing

#this is important as the RF chip talks to esp on these pins at this baud rate
uart:
  tx_pin: GPIO01
  rx_pin: GPIO03
  baud_rate: 19200
#need to disable uart logging as it will interfere
logger:
  baud_rate: 0

#this allows ha to when RF signals are received, I'm not using this for anything yet.
rf_bridge:
  on_code_received:
    #if:
      #condition:
        #switch.is_on: ${node_id}_pair_switch
      then:
        - homeassistant.event:
            event: esphome.rf_code_received
            data:
              sync: !lambda 'char buffer [10];return itoa(data.sync,buffer,16);'
              low: !lambda 'char buffer [10];return itoa(data.low,buffer,16);'
              high: !lambda 'char buffer [10];return itoa(data.high,buffer,16);'
              code: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'

#with Portisch firmware
  on_advanced_code_received:
    if:
      condition:
        switch.is_on: ${node_id}_pair_switch
      then:
        - homeassistant.event:
            event: esphome.rf_advanced_code_received
            data:
              length: !lambda 'char buffer [10];return itoa(data.length,buffer,16);'
              protocol: !lambda 'char buffer [10];return itoa(data.protocol,buffer,16);'
              code: !lambda 'return data.code;'

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO00
      inverted: True
    name: "${node_name_friendly}-PairButton"
    id: ${node_id}_pair_button
    on_multi_click:
    #长按（按键时长为x秒到y秒之间，松开按键0.2秒之后发送状态）
    - timing:
        - ON for at least 0.7s
        - OFF for at least 0.2s
      then:
        - rf_bridge.beep:
            duration: 1000
        - delay: 200ms
        - switch.toggle: ${node_id}_restart
      invalid_cooldown: 0s
    #短按（x秒内按一下，松开按键0.3秒之后反馈状态）
    - timing:
        - ON for at most 0.3s
        - OFF for at least 0.2s
      then:
        - switch.turn_on: ${node_id}_pair_switch
        - delay: 5min
        - switch.turn_off: ${node_id}_pair_switch

      invalid_cooldown: 0s
# the interesting bit, controlling the awning motor
switch:
  - platform: template
    name: "${node_name_friendly}-Cover-Light"
    id: ${node_id}_cover_light

    turn_on_action:
      #发送开灯指令
      - rf_bridge.send_raw:
          raw: "AAB029030802A800C81AF428181909081818190819081819090908190908190909081908190819081909081955"
      - lambda: |-
          id(${node_id}_cover_light).publish_state(true);
    turn_off_action:
      - rf_bridge.send_raw:
          raw: "AAB029030802A800C81AF428181909081818190819081819090908190908190909081908190819081909081955"
      - lambda: |-
          id(${node_id}_cover_light).publish_state(false);
  - platform: template
    name: "${node_name_friendly}-Pair-Switch"
    id: ${node_id}_pair_switch
    optimistic: true
    on_turn_on:
        - rf_bridge.beep:
            duration: 200
        - delay: 200ms
        - rf_bridge.beep:
            duration: 200

light:
#状态指示灯
  - platform: status_led
    name: "${node_name_friendly}-LinkLed"
    id: ${node_id}_link_led
    pin: 
      number: GPIO13
      inverted: True

cover:
  - platform: time_based
    name: "${node_name_friendly}-Cover"
    device_class: garage
    assumed_state: true
    has_built_in_endstop: true

    open_action:
      - rf_bridge.send_raw:
          raw: "AAB029030802A800BE1AFE28181909090819090819081818190818190819081818190818190819090908181955"
    open_duration: 1s

    close_action:
      - rf_bridge.send_raw:
          raw: "AAB029030802B200BE1AF428190818190909081819090909081819081908190908181908181909090818190955"
    close_duration: 1s

    stop_action:
      - rf_bridge.send_raw:
          raw: "AAB029030802A800BE1AF428181909081818190819081819090908190819081818181818190819081909081955"
```

### Anything in the logs that might be useful for us?

```txt
[I][rf_bridge:055]: Received RFBridge Code: sync=0x2E68 low=0x01A4 high=0x04B0 code=0xC94302
[W][api.connection:083]: Home Assistant 2022.2.8 (192.168.1.2): Connection closed
[I][rf_bridge:055]: Received RFBridge Code: sync=0x2E7C low=0x017C high=0x0488 code=0xC94302
[W][api.connection:083]: Home Assistant 2022.2.8 (192.168.1.2): Connection closed
[I][ota:105]: Boot seems successful, resetting boot loop counter.
```

sgofferj, 2022-02-26:
Identical to #3075

jesserockz, 2022-02-27:
Even though this was in first going to close in favour of #3075 due to more comments existing there.

closed 2022-02-27 by jesserockz