esphome/issues #3075: Sonoff RF Bridge Connection closed

sanyadez, 2022-02-21:
### The problem

Connection reset errors after receiving RF codes appeared after the latest updates

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

Sonoff RF Bridge

### Component causing the issue

rf_bridge

### Example YAML snippet

```yaml
substitutions:
  devicename: sonoff_rf_bridge
  upper_devicename: Sonoff RF Bridge
  
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret esp_second_wifi_ssid
    password: !secret esp_second_wifi_pass
  - ssid: !secret esp_main_wifi_ssid
    password: !secret esp_main_wifi_pass
  manual_ip:
    static_ip: !secret esp_rf_bridge_ip
    gateway: !secret esp_gateway
    subnet: !secret esp_subnet
  ap:
    ssid: ${upper_devicename} Fallback
    password: !secret esp_fallback_ap_password

logger:
  baud_rate: 0

web_server:
  port: 80

ota:
  id: ota_pass
  password: !secret esp_ota_password

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200
  
api:
  # password: !secret esp_api_password
  services:
    - service: send_rf_code
      variables:
        sync: int
        low: int
        high: int
        code: int
      then:
        - rf_bridge.send_code:
            sync: !lambda 'return sync;'
            low: !lambda 'return low;'
            high: !lambda 'return high;'
            code: !lambda 'return code;'
    - service: learn
      then:
        - rf_bridge.learn

rf_bridge:
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'char buffer [10];return itoa(data.sync,buffer,16);'
            low: !lambda 'char buffer [10];return itoa(data.low,buffer,16);'
            high: !lambda 'char buffer [10];return itoa(data.high,buffer,16);'
            code: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'
      - text_sensor.template.publish:
          id: rf_code
          state: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'

text_sensor:
  - platform: template
    name: "Latest Code"
    id: rf_code
```

### Anything in the logs that might be useful for us?

```txt
[12:05:34][I][app:102]: ESPHome version 2022.2.4 compiled on Feb 21 2022, 11:52:08
[12:05:34][C][wifi:491]: WiFi:
[12:05:34][C][wifi:353]:   Local MAC: 60:01:94:..:..:..
[12:05:34][C][wifi:354]:   SSID: [redacted]
[12:05:34][C][wifi:355]:   IP Address: 192.168.....
[12:05:34][C][wifi:356]:   BSSID: [redacted]
[12:05:34][C][wifi:358]:   Hostname: 'sonoff_rf_bridge'
[12:05:34][C][wifi:360]:   Signal strength: -62 dB ▂▄▆█
[12:05:34][C][wifi:364]:   Channel: 11
[12:05:34][C][wifi:365]:   Subnet: 255.255.255.0
[12:05:34][C][wifi:366]:   Gateway: 192.168.....
[12:05:34][C][wifi:367]:   DNS1: 0.0.0.0
[12:05:34][C][wifi:368]:   DNS2: 0.0.0.0
[12:05:34][C][logger:233]: Logger:
[12:05:34][C][logger:234]:   Level: DEBUG
[12:05:34][C][logger:235]:   Log Baud Rate: 0
[12:05:34][C][logger:236]:   Hardware UART: UART0
[12:05:34][C][uart.arduino_esp8266:102]: UART Bus:
[12:05:34][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[12:05:34][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[12:05:34][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[12:05:34][C][uart.arduino_esp8266:108]:   Baud Rate: 19200 baud
[12:05:34][C][uart.arduino_esp8266:109]:   Data Bits: 8
[12:05:34][C][uart.arduino_esp8266:110]:   Parity: NONE
[12:05:34][C][uart.arduino_esp8266:111]:   Stop bits: 1
[12:05:34][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[12:05:34][C][template.text_sensor:021]: Template Sensor 'Latest Code'
[12:05:34][C][rf_bridge:184]: RF_Bridge:
[12:05:34][C][web_server:179]: Web Server:
[12:05:34][C][web_server:180]:   Address: 192.168.....:80
[12:05:34][C][mdns:084]: mDNS:
[12:05:34][C][mdns:085]:   Hostname: sonoff_rf_bridge
[12:05:34][C][ota:085]: Over-The-Air Updates:
[12:05:34][C][ota:086]:   Address: 192.168.....:8266
[12:05:34][C][ota:089]:   Using Password.
[12:05:34][C][api:138]: API Server:
[12:05:34][C][api:139]:   Address: 192.168.....:6053
[12:05:34][C][api:143]:   Using noise encryption: NO
[12:05:37][I][rf_bridge:055]: Received RFBridge Code: sync=0x26FC low=0x014A high=0x03E8 code=0x55550C
[12:05:37][D][text_sensor:067]: 'Latest Code': Sending state '55550c'
[12:05:37][W][api.connection:083]: Home Assistant 2022.2.9 (192.168.1.3): Connection closed
[12:05:37][D][api:102]: Accepted 192.168.1.3
[12:05:37][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:38][I][rf_bridge:055]: Received RFBridge Code: sync=0x26FC low=0x014A high=0x03F2 code=0x555530
[12:05:39][D][text_sensor:067]: 'Latest Code': Sending state '555530'
[12:05:39][W][api.connection:083]: Home Assistant 2022.2.9 (192.168.1.3): Connection closed
[12:05:39][D][api:102]: Accepted 192.168.1.3
[12:05:39][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:39][I][rf_bridge:055]: Received RFBridge Code: sync=0x2710 low=0x014A high=0x03F2 code=0x555503
[12:05:39][D][text_sensor:067]: 'Latest Code': Sending state '555503'
[12:05:40][W][api.connection:083]: Home Assistant 2022.2.9 (192.168.1.3): Connection closed
[12:05:40][D][api:102]: Accepted 192.168.1.3
[12:05:40][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:40][I][rf_bridge:055]: Received RFBridge Code: sync=0x271A low=0x014A high=0x03F2 code=0x5555C0
[12:05:40][D][text_sensor:067]: 'Latest Code': Sending state '5555c0'
[12:05:41][W][api.connection:083]: Home Assistant 2022.2.9 (192.168.1.3): Connection closed
[12:05:41][D][api:102]: Accepted 192.168.1.3
[12:05:41][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:41][I][rf_bridge:055]: Received RFBridge Code: sync=0x2724 low=0x014A high=0x03F2 code=0x55550C
[12:05:41][D][text_sensor:067]: 'Latest Code': Sending state '55550c'
[12:05:41][W][api.connection:083]: Home Assistant 2022.2.9 (192.168.1.3): Connection closed
[12:05:41][D][api:102]: Accepted 192.168.1.3
[12:05:41][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:42][I][rf_bridge:055]: Received RFBridge Code: sync=0x2738 low=0x0154 high=0x03E8 code=0x55550C
[12:05:42][D][text_sensor:067]: 'Latest Code': Sending state '55550c'
[12:05:42][W][api.connection:081]: Home Assistant 2022.2.9 (192.168.1.3): Connection reset
[12:05:42][D][api:102]: Accepted 192.168.1.3
[12:05:42][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:44][I][rf_bridge:055]: Received RFBridge Code: sync=0x2724 low=0x014A high=0x03E8 code=0x55550C
[12:05:44][D][text_sensor:067]: 'Latest Code': Sending state '55550c'
[12:05:44][W][api.connection:083]: Home Assistant 2022.2.9 (192.168.1.3): Connection closed
[12:05:44][D][api:102]: Accepted 192.168.1.3
[12:05:44][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.3): Connected successfully
[12:05:54][D][text_sensor:067]: 'Latest Code': Sending state '55550c'
```

### Additional information

I've used that in the HA as automations trigger:
trigger:
  - platform: event
    event_type: esphome.rf_code_received
    
Now it doesn't work. By default, I have an API password set, but I've tried both with and without it. I also tried using a different access point. Everywhere the same result.

sanyadez, 2022-02-21:
The issue with the 
`
  data:
    sync: !lambda 'char buffer [10];return itoa(data.sync,buffer,16);'
    low: !lambda 'char buffer [10];return itoa(data.low,buffer,16);'
    high: !lambda 'char buffer [10];return itoa(data.high,buffer,16);'
    code: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'
`         
after 
`
  homeassistant.event:
    event: esphome.rf_code_received
`
Without this all going ok. Any idea what it's about?

DarwinData, 2022-02-21:
I am having the exact same problem as the OP.

jant90, 2022-02-22:
I think I have the same issue. The big change in 2022.2 was the upgrade of the default Arduino framework from 2.7.4 to 3.0.2. **You can solve all issues with the RF Bridge Component for for now by downgrading to 2.7.4:**
```yaml
esphome:
  name: sonoff_rf_bridge
esp8266:
  board: esp01_1m
  framework:
    version: 2.7.4
```
I hope this will eventually get fixed but I'm afraid it might not because RF doesn't get much love anymore.

filipotava, 2022-02-22:
I have exactly the same issue.
Downgrade to 2.7.4 fixed the issue as @jant90 wrote above (thanks!)

DarwinData, 2022-02-22:
Downgrade to 2.7.4 fixed the issue for me as well. thanks!

jesserockz, 2022-02-22:
I will add this to the list to investigae, but as there is a workaround it will not be a high priority right now.

I do use the `rf_bridge` still so I may come across the issue myself if I simply upgrade it..haha

sgofferj, 2022-02-26:
I have the same issue. I tried replacing itoa in the sample code with sprintf(buffer, "%x", data.xxx) and that doesn't cause the disconnect. It does, however, create a wrong value which is a single digit number.

@jesserockz I'm not sure the issue lies with the integration. It might be a bug in the lambda handling too.
I tried to leg the value in the on_... action and that works fine:
```
      - logger.log:
          format: "%x"
          args: [ 'data.code' ]
```
Also, removing the hex conversion in the lambda and just returning the integer value works fine too, which is another hint that the problem lies somewhere else than with the integration.

I tried to look at the created source code, but I don't understand it as I'm not familiar with the whole ESPHome architecture.

jesserockz, 2022-02-27:
Can you try:

```yaml
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'return format_hex(data.sync);'
            low: !lambda 'return format_hex(data.low);'
            high: !lambda 'return format_hex(data.high);'
            code: !lambda 'return format_hex(data.code);'
```

sgofferj, 2022-02-27:
Works.

jant90, 2022-02-27:
Thanks @jesserockz that works for me as well! :) I personally have all my automations in HA based on the decimal values so I use this instead:
```yaml
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'return data.sync;'
            low: !lambda 'return data.low;'
            high: !lambda 'return data.high;'
            code: !lambda 'return data.code;'
```
Great to see that the fix was so simple, and the lambda is now easier to read and understand.

jesserockz, 2022-02-27:
Cool. Must have been something in the Arduino 3 upgrade that broke atoi.

I will update the documentation example

closed 2022-02-27 by jesserockz