esphome/issues #3076: ESP32-IDF framework fails to compile on ARMv7

DanteBC, 2022-02-21:
### The problem

Ever since updating to ESPHome 2022.2.x my ESP32-C3 fails to compile on my ARMv7 docker installation.
The way to circumvent it is to state the framework version in the config, like so:
 
```
esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
    platform_version: 3.3.2
```
That way it compiles just fine.

### Which version of ESPHome has the issue?

2022.2.1+

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.8

### What platform are you using?

ESP32-IDF

### Board

ESP32-C3

### Anything in the logs that might be useful for us?

```txt
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/esp01c3/.pioenvs/esp01c3/CMakeFiles/CMakeOutput.log".
See also "/data/esp01c3/.pioenvs/esp01c3/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc"

  is not able to compile a simple test program.
  It fails with the following output:

    Change Dir: /data/esp01c3/.pioenvs/esp01c3/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_fc39c && [1/2] Building C object CMakeFiles/cmTC_fc39c.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_fc39c.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc   -march=rv32imc -o CMakeFiles/cmTC_fc39c.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:311 (__project)
  CMakeLists.txt:3 (project)
```

DanteBC, 2022-03-17:
Trying to figure out what I can do about my esp32c3 esp-idf not being able to compile on my amv7 board. It's a docker installation. if I ssh into the container and try to execute 
_#/data/cache/platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc_ directly, I get "No such file or directory" but 
_#/data/cache/platformio/packages/toolchain-riscv-esp/bin/riscv32-esp-elf-gcc_ returns "no input files" which means it at least runs the compiler fine.
As far as I understand, it's something to do with the toolchain itself. It says it supports armv7 in the package.json but apparently it doesn't?

DanteBC, 2022-04-06:
I also feel like this issue is related to mine. Gonna try the workaround that is mentioned in that thread later and see if it helps. https://github.com/platformio/platform-espressif32/issues/753#

DanteBC, 2022-04-16:
https://github.com/espressif/esp-idf/issues/7857 this issue is definitely related to the problem at hand. Espressif promises to fix the issue in the next major toolchain release which will happen "soon"

Resterday, 2022-05-21:
I am having the same issue. I rried setting to an older framework as mentioned, but im still getting an error. Any help would be apreciated!

Config:

`esp32:
  board: esp32dev
  framework:
    type: esp-idf
    platform_version: 3.3.2`

Error Output:

`Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /config/esphome/.esphome/build/emportiavue/sdkconfig.emportiavue
-- Configuring incomplete, errors occurred!
See also "/data/emportiavue/.pioenvs/emportiavue/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/crosstool_version_check.cmake:37 (message):
  

  Toolchain:
  /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc,
  crosstool-ng version esp-2020r3 doesn't match supported version esp-2021r2

  Please try to run 'idf.py fullclean' to solve it quickly.

  Check Getting Started documentation if the error continues.

  You can override this error and proceed with build by defining the
  IDF_MAINTAINER environment variable.

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/components/esp32/project_include.cmake:31 (crosstool_version_check)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:315 (include)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:466 (__build_process_project_includes)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:399 (idf_build_process)
  CMakeLists.txt:3 (project)`

DanteBC, 2022-05-21:
> I am having the same issue. I rried setting to an older framework as mentioned, but im still getting an error. Any help would be apreciated!
>
> Config:
>
> esp32: board: esp32dev framework: type: esp-idf [...]

If you're using home assistant addon, try removing and reinstalling the esphome addon. It will install required toolchain versions, cause right now you have version mismatch.

Resterday, 2022-05-21:
> If you're using home assistant addon, try removing and reinstalling the esphome addon. It will install required toolchain versions, cause right now you have version mismatch.

Thanks for the suggestion and quick reply. realized that I was using the deprecated homeassistant community addon so I switched over, but Im still getting a similar error but with some more detail. Tried uninstalling and re-installing 2 times.

Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /config/esphome/.esphome/build/emportiavue/sdkconfig.emportiavue
-- Configuring incomplete, errors occurred!
See also "/data/emportiavue/.pioenvs/emportiavue/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/crosstool_version_check.cmake:37 (message):
  

  Toolchain:
  /data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gcc,
  crosstool-ng version esp-2020r3 doesn't match supported version esp-2021r2

  Please try to run 'idf.py fullclean' to solve it quickly.

  Check Getting Started documentation if the error continues.

  You can override this error and proceed with build by defining the
  IDF_MAINTAINER environment variable.

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/components/esp32/project_include.cmake:31 (crosstool_version_check)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:315 (include)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:466 (__build_process_project_includes)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:399 (idf_build_process)
  CMakeLists.txt:3 (project)

DanteBC, 2022-05-21:
Can I see your device Config?

DanteBC, 2022-06-25:
Update: according to my rough tests, updating toolchain to the latest [(esp-2022r1-RC1)](https://github.com/espressif/crosstool-NG/releases/tag/esp-2022r1-RC1) availible version fixes the issue and compiling is able to start. This [commit](https://github.com/espressif/esp-idf/commit/bf43076c044a58429ebd90ac508e32dd82321af1) adds the required toolchain. Now we just have to wait till it trickle through platformio over into esphome

bearpawmaxim, 2022-07-07:
Hi @DanteBC, I'm facing the same problem.
Is there any way to specify in esphome the version of toolchain to use?

DanteBC, 2022-07-07:
> Hi @DanteBC, I'm facing the same problem. Is there any way to specify in esphome the version of toolchain to use?

Hello @bearpawmaxim
Try specifying this version of the framework. Works for me at least:
```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    platform_version: 3.4.0
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
```
If it won't work for you please show me your compile output so I could maybe see what went wrong

bearpawmaxim, 2022-07-07:
> If it won't work for you please show me your compile output

Thank you! But, still not working, but the error differs.
The compilation output is in my gist here https://gist.github.com/bearpawmaxim/cb0528a96aa1eca94427abd9cd45b1ff#file-1_logs_water-cutoff_upload-log

jesserockz, 2022-07-07:
You can override the packages required by the framework as below:
```yaml
esphome:
  ...
  platformio_options:
    platform_packages:
	  - <package_name> @ <version>
```

DanteBC, 2022-07-07:
> You can override the packages required by the framework as below:
> 
> ```yaml
> esphome:
>   ...
>   platformio_options:
>     platform_packages:
>       - <package_name> @ <version>
> ```

Not sure how exactly I should specify the toolchain version? Maybe I'm setting the version number incorrectly. Can you help us to set the toolchain version to 2021r1? or even to esp-2022r1-RC1. For me any combination of package name and version results in getting an error that there is no such package for my system 'linux_armv7l'

bearpawmaxim, 2022-07-08:
> Try this in your config

The error changed
https://gist.github.com/bearpawmaxim/cb0528a96aa1eca94427abd9cd45b1ff#file-2_logs_water-cutoff_upload-log

DanteBC, 2022-07-08:
> > Try this in your config
> 
> The error changed https://gist.github.com/bearpawmaxim/cb0528a96aa1eca94427abd9cd45b1ff#file-2_logs_water-cutoff_upload-log

As far as I understand, you're trying to use ESP32 board, right? not ESP32-C3, just generic ESP32?
I'll try your config on my installation, I'll see if I'll get the same error

bearpawmaxim, 2022-07-08:
> just generic ESP32?

Yes!

DanteBC, 2022-07-08:
> > just generic ESP32?
> 
> Yes!

This config addition allowed me to start compiling. Please check.

```yaml
esphome:
  name: ${device_name}
  platformio_options:
    platform_packages:
     - framework-espidf @ 3.40300.0

esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    platform_version: 3.4.0
```

bearpawmaxim, 2022-07-08:
Thanks @DanteBC! Finally it started compilation!

mmakaay, 2022-10-12:
Reopened the issue, because it keeps popping up, and possibly we can fix this from within ESPHome.
I did some investigation and found a possible way to get the toolchain working.
My comments on this can be found at https://github.com/mmakaay/esphome-xiaomi_bslamp2/issues/93#issuecomment-1275865228

TLDR: the toolchain ELF binaries use /lib/d-linux.so.3 as their interpreter, but the docker container only has /lib/ld-linux-armhf.so.3 available.

mmakaay, 2022-10-13:
After some digging, I'll summarize my findings so far.
Input and thoughts on this are appreciated.

- ESPHome docker container is based on `armv7` (32 bit ARM) using architecture `armhf` (which is correct).
- PlatformIO only looks at `armv7`, ignores `armel` vs `armhf`, and installs `toolchain-xtensa-esp32` for `armel`.
- Espressif does provide an `armhf` toolchain when using their esp-idf installer.

The combination of the docker container and the PlatformIO toolchain fails, because the ELF binaries for the toolchain are not working in the `armhf` container. They depend on the `/lib/ld-linux.so.3` interpreter instead of `/lib/ld-linux-armhf.so.3`. That makes the compile process crash at the first step: compiling a test program.

I would say that the problem lies within the fact that PlatformIO does not provide the toolchain binaries for `armhf` here, but the non-functional ones.

**Possible solutions**

- Don't use an armv7 32-bit OS on the RPi, to fully circumvent the issue. Of course, this can be quite daunting when already running a production OS on there, and it's merely a work-around and not a fix.
- Maybe base the armv7 image of ESPHome on an `armel` image to make it compatible with the toolchain that PlatformIO installs.
- Publish an `armhf` toolchain and modify the esphome YAML config to point to that toolchain. I tried this route, but failed because there's no binary release available (anymore) for the `8.4.0+2021r2-patch2` release of the toolchain that is currently in use by the platform. This would also be a work-around at best, since it would require creating a new toolchain package when PlatformIO moves to a new version.

The only real solution that I see though is:

- PlatformIO honours the `armhf` architecture by installing the correct toolchain for it.

raoulbhatia, 2022-12-16:
FWIW, I stumbled across this issue as well for my `GL.iNet GL-S10` that I intend to use as a bluetooth-proxy, see [1][2][3].
I am running esphome in a docker container on a `Raspberry Pi 2 Model B Rev 1.1`.

It would be good to get that one fixed.
As a workaround, I can update from my laptop.  But this is not how I intend to use this product with esphome.

Thanks,
Raoul

[1] https://esphome.github.io/bluetooth-proxies/
[2] https://blakadder.com/gl-s10/
[3] https://github.com/esphome/bluetooth-proxies/blob/main/gl-s10.yaml

erwinschoonhoven, 2023-01-14:
Facing the same problem

Currently running 2022.12.3 on an rpi-3b (home assistant add-on): Failing with above mentioned `liblto_plugin.so: cannot open shared object file`

Would be really nice if this one is fixed ..

raoulbhatia, 2023-02-12:
Per https://github.com/platformio/platform-espressif32/issues/753#issuecomment-1355576346
this should be fixed by upgrading to `espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5`

However, manually specifying this via `platform_packages` does not pass the validation as `espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3` is expected during compile time.

Is there any way to get the new version incorporated into `esphome`?

DanteBC, 2023-02-12:
> Per [platformio/platform-espressif32#753 (comment)](https://github.com/platformio/platform-espressif32/issues/753#issuecomment-1355576346) this should be fixed by upgrading to `espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5`

Not in my case. The issue here is entirely in how toolchain-xtensa-esp32 is built after 2021r1. The only way I was able to start compiling is to use the newest availible toolchain v. 11.2.0+2022r1, which is made specifically to work with ESP-IDF version 5. Which is yet incompattible with esphome at this point. It starts compiling though.

But if you still want to check if toolchain 8.4.0+2021r2-patch5 will do it for you you can try adding the following ti your config:

```yaml
esphome:
  platformio_options:
    platform_packages:
     - framework-espidf @ 3.40402.0
esp32:
  framework:
    type: esp-idf
    platform_version: 5.3.0
```

To use the newest availible(and yet incompatible) platform and toolchain version you can use the following:

```yaml
esphome:
  platformio_options:
    platform_packages:
     - framework-espidf @ 3.50000.0
esp32:
  framework:
    type: esp-idf
    platform_version: 6.0.0
```

ssieb, 2023-02-12:
> However, manually specifying this via `platform_packages` does not pass the validation as `espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3` is expected during compile time.

Not sure what you mean by this.  You can specify that toolchain version in the config.  I've seen others use it, but I don't remember the exact syntax for it.

ssieb, 2023-02-14:
And now I see it was actually earlier in this issue...
If you're having a problem, paste what exactly you're trying to do.

DanteBC, 2023-02-21:
> Hey folks, I have stumbled on the same issue as well.

Add ISP-IDF and platformio versions to your config as stated in the bottom of the very first message. Should start compiling for you.

thanpolas, 2023-02-21:
Thank you @DanteBC, I now have a new error, pasting bellow, a few questions before:

What are the tradeoffs of using a static / older version? What is the current version of esp-idf?

Is this a permanent solution or a stub until a fix is ready? Long-term, this explicit version requirement will have to be removed, right? What's the pathway to that goal?

### New Configuration (Only the header)

```yaml
esphome:
  name: bedrooms-esphub
  friendly_name: Bedrooms ESPHub
  platformio_options:
    platform_packages:
     - framework-espidf @ 3.40300.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    platform_version: 3.4.0
```

### The New Error

```
INFO Reading configuration /config/esphome/bedrooms-esphub.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bedrooms-esphub (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40300.0 (4.3.0) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r1 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r1 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Archiving /data/bedrooms-esphub/.pioenvs/bedrooms-esphub/esp-idf/mdns/libmdns.a
Compiling /data/bedrooms-esphub/.pioenvs/bedrooms-esphub/newlib/newlib_init.o
Compiling /data/bedrooms-esphub/.pioenvs/bedrooms-esphub/newlib/time.o
Compiling /data/bedrooms-esphub/.pioenvs/bedrooms-esphub/newlib/port/esp_time_impl.o
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:85:6: error: 'struct syscall_stub_table' has no member named '_retarget_lock_init'; did you mean '_lock_init'?
     ._retarget_lock_init = &__retarget_lock_init,
      ^~~~~~~~~~~~~~~~~~~
      _lock_init
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:86:6: error: 'struct syscall_stub_table' has no member named '_retarget_lock_init_recursive'; did you mean '_lock_init_recursive'?
     ._retarget_lock_init_recursive = &__retarget_lock_init_recursive,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      _lock_init_recursive
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:87:6: error: 'struct syscall_stub_table' has no member named '_retarget_lock_close'; did you mean '_lock_close'?
     ._retarget_lock_close = &__retarget_lock_close,
      ^~~~~~~~~~~~~~~~~~~~
      _lock_close
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:87:29: warning: initialization of 'void (*)(struct __lock **)' from incompatible pointer type 'void (*)(struct __lock *)' [-Wincompatible-pointer-types]
     ._retarget_lock_close = &__retarget_lock_close,
                             ^
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:87:29: note: (near initialization for 's_stub_table._lock_close')
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:88:6: error: 'struct syscall_stub_table' has no member named '_retarget_lock_close_recursive'; did you mean '_lock_close_recursive'?
     ._retarget_lock_close_recursive = &__retarget_lock_close_recursive,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      _lock_close_recursive
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:88:39: warning: initialization of 'void (*)(struct __lock **)' from incompatible pointer type 'void (*)(struct __lock *)' [-Wincompatible-pointer-types]
     ._retarget_lock_close_recursive = &__retarget_lock_close_recursive,
[... 29 lines cut ...]
      _lock_try_acquire_recursive
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:92:45: warning: initialization of 'int (*)(struct __lock **)' from incompatible pointer type 'int (*)(struct __lock *)' [-Wincompatible-pointer-types]
     ._retarget_lock_try_acquire_recursive = &__retarget_lock_try_acquire_recursive,
                                             ^
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:92:45: note: (near initialization for 's_stub_table._lock_try_acquire_recursive')
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:93:6: error: 'struct syscall_stub_table' has no member named '_retarget_lock_release'; did you mean '_lock_release'?
     ._retarget_lock_release = &__retarget_lock_release,
      ^~~~~~~~~~~~~~~~~~~~~~
      _lock_release
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:93:31: warning: initialization of 'void (*)(struct __lock **)' from incompatible pointer type 'void (*)(struct __lock *)' [-Wincompatible-pointer-types]
     ._retarget_lock_release = &__retarget_lock_release,
                               ^
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:93:31: note: (near initialization for 's_stub_table._lock_release')
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:94:6: error: 'struct syscall_stub_table' has no member named '_retarget_lock_release_recursive'; did you mean '_lock_release_recu [... cut ...]
     ._retarget_lock_release_recursive = &__retarget_lock_release_recursive,
                                         ^
/data/cache/platformio/packages/framework-espidf@3.40300.0/components/newlib/newlib_init.c:94:41: note: (near initialization for 's_stub_table._lock_release_recursive')
*** [/data/bedrooms-esphub/.pioenvs/bedrooms-esphub/newlib/newlib_init.o] Error 1
Compiling /data/bedrooms-esphub/.pioenvs/bedrooms-esphub/nghttp/nghttp2/lib/nghttp2_buf.o
========================= [FAILED] Took 71.41 seconds =========================
```

sbrk-r, 2023-02-28:
Changing to the specific versions in the first post doesn't seem to work for me, did the old one get deprecated or something like that?

```   
INFO Reading configuration node32c3_94.yaml...
Failed config

esp32: [source node32c3_94.yaml:27]
  board: esp32-c3-devkitm-1

  Only ESP-IDF 4.0+ is supported.
  framework:
    type: esp-idf
    version: 3.4.0
```

this is on old junker armv7l
esphome v 2023.2.4

jrames, 2023-03-05:
> Thank you @DanteBC, I now have a new error, pasting bellow, a few questions before:

I'm getting the same subsequent error that you were 2 weeks ago. Any luck with that?

thanpolas, 2023-03-05:
nope, maybe time to nudge @DanteBC a bit

evan-shaw, 2023-05-10:
Is there any update on this issue? I'm getting ESP32-IDF compilation issues on x86_64 h/w as well.

DanteBC, 2023-05-10:
> Is there any update on this issue? I'm getting ESP32-IDF compilation issues on x86_64 h/w as well.

What chip are you compiling for? esp32 or esp32-c3?

evan-shaw, 2023-05-10:
> > Is there any update on this issue? I'm getting ESP32-IDF compilation issues on x86_64 h/w as well.
> 
> What chip are you compiling for? esp32 or esp32-c3?

esp32

DanteBC, 2023-05-10:
> > > Is there any update on this issue? I'm getting ESP32-IDF compilation issues on x86_64 h/w as well.
> > 
> > 
> > What chip are you compiling for? esp32 or esp32-c3?
> 
> esp32

Can you provide a sample of your config and an error that compilers gives you?

evan-shaw, 2023-05-10:
> > > > Is there any update on this issue? I'm getting ESP32-IDF compilation issues on x86_64 h/w as well.
> > > 
> > > 
> > > What chip are you compiling for? esp32 or esp32-c3?
> > 
> > 
> > esp32
> 
> Can you provide a sample of your config and an error that compilers gives you?

Sure, I;m yaing thus code "https://github.com/patrick3399/Hi-Link_mmWave_Radar_ESPHome/blob/main/LD1115H/ESP32-LD1115H-Complete.yaml"

First time I tried it, end of march, it worked fine, but after an update, compilation failed.
Here's the error part of the log: -

`framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /config/.esphome/build/esp32-test/sdkconfig.esp32-test
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/esp32-test/.pioenvs/esp32-test/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /config/.esphome/platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)`

DanteBC, 2023-05-11:
@evan-shaw you have a toolchain version mismatch. This happens cause esphome by default uses specific esp-idf version, but your config specifies it to use the latest version it can find. And considering since the moment this config was written platformio relised a couple of new platform versions that's yet incompatible with esphome, we have this picture. You can try to remove or comment out **_version: latest_** in your config. It should start using the default esphome esp-idf and toolchain versions and compile fine that way.

GaryOkie, 2023-05-12:
Having this exact error on RasPi4 32bit HASSOS(2023.5.2)  ESPHome add-on install.

I understand "remove or comment out _version: latest_ in your config" is the workaround, but I don't see that in any config files.  

So I next tried uninstalling ESPHome and then installing the Beta.   That install also failed, but with a different error about a pyparsing version conflict:

```
  Attempting uninstall: pyparsing
    Found existing installation: pyparsing 3.0.9
    Uninstalling pyparsing-3.0.9:
      Successfully uninstalled pyparsing-3.0.9
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2023.5.0b2 requires pyparsing>=3.0, but you have pyparsing 2.3.1 which is incompatible.
Successfully installed cachecontrol-0.12.11 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 future-0.18.3 idf-component-manager-1.2.2 idna-3.4 lockfile-0.12.2 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.30.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-2.0.2
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/esp32-bluetooth-proxy-578a40/.pioenvs/esp32-bluetooth-proxy-578a40/CMakeFiles/CMakeOutput.log".
See also "/data/esp32-bluetooth-proxy-578a40/.pioenvs/esp32-bluetooth-proxy-578a40/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/esp32-bluetooth-proxy-578a40/.pioenvs/esp32-bluetooth-proxy-578a40/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_8f555 && [1/2] Building C object CMakeFiles/cmTC_8f555.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_8f555
    FAILED: cmTC_8f555 
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_8f555.dir/testCCompiler.c.obj  -o cmTC_8f555   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:296 (__project)
  CMakeLists.txt:3 (project)
```

Uninstalled the beta, tried reinstalling production ESPHome, and get yet another/different error:
```
Successfully installed cachecontrol-0.12.11 certifi-2023.5.7 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 future-0.18.3 idf-component-manager-1.2.2 idna-3.4 lockfile-0.12.2 msgpack-1.0.5 packaging-23.1 pyparsing-2.3.1 pyyaml-6.0 requests-2.30.0 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.65.0 urllib3-2.0.2
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/esp32-bluetooth-proxy-578a40/.pioenvs/esp32-bluetooth-proxy-578a40/CMakeFiles/CMakeOutput.log".
See also "/data/esp32-bluetooth-proxy-578a40/.pioenvs/esp32-bluetooth-proxy-578a40/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

[... cut ...]
    : && /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_2b897.dir/testCCompiler.c.obj  -o cmTC_2b897   && :
    /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
```

DanteBC, 2023-05-12:
@GaryOkie you don't have the exact same problem. There's no toolchain version mismatch in your case. Your toolchain version is fine, it's just broken.
Show me your config please. There's probably an option to set some specific (older) working toolchain\platform version to compile it.

GaryOkie, 2023-05-13:
Thanks very much for the assist!   

The configuration for the ESPHome addon is empty - just defaults.   I've configured nothing.  It seems to install fine and the webUI works.   But when trying to update the firmware in an adopted  ESP32 Bluetooth proxy is when it tries to compile all that stuff and fails.   I'm not clear what config  manages the compile that I can show here.

DanteBC, 2023-05-13:
@GaryOkie I don't need your esphome addon configuration. I need your esphome device config that you are trying to compile. You can get it by pressing edit on the device card in your esphome addon webUI.

GaryOkie, 2023-05-13:
Ah - ok.  Again, it's just the default config.   

```
substitutions:
  name: esp32-bluetooth-proxy-578a40
  friendly_name: Bluetooth Proxy 578a40
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

DanteBC, 2023-05-14:
Adding strings to state older esp-idf and toolchain versions was enough for it to start compiling for me. Should do the trick in your case too
```yaml
substitutions:
  name: esp32-bluetooth-proxy-578a40
  friendly_name: Bluetooth Proxy 578a40

packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    platform_packages:
     - framework-espidf @ 3.40300.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    platform_version: 3.4.0

api:
  encryption:
    key: "ay38+BwUtHYhJIHNtDVuj8bKxQqD1n58AielmG4Ojic="

api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

GaryOkie, 2023-05-14:
I really appreciate your assistance with the proper library incantations.

The compile choked again not finding this package: 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r1' 
I```
NFO Reading configuration /config/esphome/esp32-bluetooth-proxy-578a40.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-bluetooth-proxy-578a40 (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.4.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 3.4.0
INFO Installing platformio/espressif32 @ 3.4.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@3.4.0 has been installed!
INFO espressif32@3.4.0 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1
INFO Installing espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r1 has been installed!
INFO toolchain-xtensa-esp32@8.4.0+2021r1 has been installed!
Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
INFO Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r1
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa-esp32s2@8.4.0+2021r1 has been installed!
INFO toolchain-xtensa-esp32s2@8.4.0+2021r1 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r1
INFO Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r1
Error: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r1' requirements for your system 'linux_armv7l'
```

DanteBC, 2023-05-16:
It looks like platformio has removed older toolchain versions from their servers. I already had those installed and therefor were able to compile with them. 
In that case you can try to compile your project with arduino framework. It's been working fine on esp32 for quite some time now. Not sure about bluetooth-proxy specifically though. You can try. Here's the config to implement bluetooth-proxy into default config:

```yaml
substitutions:
  name: esp32-bluetooth-proxy-578a40
  friendly_name: Bluetooth Proxy 578a40

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

esp32:
  board: esp32dev
  framework:
    type: arduino

api:
  encryption:
    key: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "passwordHere"

dashboard_import:
  package_import_url: github://esphome/bluetooth-proxies/esp32-generic.yaml@main

esp32_ble_tracker:
  scan_parameters:
    # We currently use the defaults to ensure Bluetooth
    # can co-exist with WiFi In the future we may be able to
    # enable the built-in coexistence logic in ESP-IDF
    active: true

bluetooth_proxy:
  active: true

button:
- platform: safe_mode
  name: Safe Mode Boot
  entity_category: diagnostic
```

GaryOkie, 2023-05-16:
Again, many thanks Dante!

I am just starting with ESP32 with BT Tracking and only have 1 BT Proxy set up so far.   HA was prompting for an important firmware update but OTA was not possible until I added ESPHome.

So rather than "try this arduino code and see..."   I decided to just plug the device into the laptop and reflash ESPGeneric from the browser.   All good now.

Going forward when I have multiple ESP's running, ESPHome management will be the way to go, and hopefully I won't be having these version conflicts any longer.

DanteBC, 2023-05-16:
Glad you were able to solve your situation. Esp32 toolchains are only broken on some architectures. You can freely use them on local pc installation and compile them that way. The only way to fix them is to wait for esphome to move to platformio version 6, which is not happening in the near future

2m, 2023-06-23:
Had a similar issue when following [Home Assistant guide for Voice Assistant on M5 Atom Echo](https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/).

esphome config: https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml

Thank you for the `arduino` hint. That fixed the issue!

```
esp32:
  board: m5stack-atom
  framework:
    type: arduino
```

edgicat, 2024-06-28:
Hi All,
I have the same issue as reported here.  Daily driver is ESPHome (2024.6.4) as a Docker container on a Raspberry PI 2, But I've also tried exactly the same code and ESPHome setup on a Raspberry PI 400 with the 32bit lite BookWorm OS install is fails in the same way.  BUT If i change to a 64Bit OS on the Raspberry PI 400 again BookWorm with Docker ESPhome Installed it works fine...  It looks to be an issue within the building compiler stage. 

ESP Code Below: (Same file used in all tests) I've replaced secure information with x's
substitutions:
  name: my-bluetooth-proxy
  board: esp32-poe-iso

esphome:
  name: esp-test
  name_add_mac_suffix: true
  friendly_name: ESP_Test

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
              
ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
                         

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

# Optional manual IP
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.0.x
    gateway: 192.168.0.x
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xx-xxxx xxxxxxxx xxxxxxx"
    password: "xxxxxxxxxxxx"

web_server:
  port: 80

sensor:
   - platform: internal_temperature
     device_class: "temperature"     
     name: "Internal Temperature"
     filters:
       - offset: "-35"
---------------------------
Here is the build process on the pi2 (Technically this output is from a different ESP Home script to above, but it does matter which script I choose the result is the same. )

[logs_boiler-condensate-fluid-level_run(pi2 32bit).txt](https://github.com/user-attachments/files/16036676/logs_boiler-condensate-fluid-level_run.pi2.32bit.txt)

Here is the build process on pi400 32bit OS

[Logs_ep-test_run (pi400 32bit).txt](https://github.com/user-attachments/files/16036684/logs_ep-test_run.pi400.32bit.txt)

The below seems to be the cause of the issue:

-- Check for working C compiler: /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/config/.esphome/build/boiler-fluid-level/.pioenvs/boiler-fluid-level/CMakeFiles/CMakeOutput.log".
See also "/config/.esphome/build/boiler-fluid-level/.pioenvs/boiler-fluid-level/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /config/.esphome/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /config/.esphome/build/boiler-fluid-level/.pioenvs/boiler-fluid-level/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/config/.esphome/platformio/packages/tool-ninja/ninja cmTC_454e6 && [1/2] Building C object CMakeFiles/cmTC_454e6.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_454e6
    FAILED: cmTC_454e6 
    : && /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_454e6.dir/testCCompiler.c.obj  -o cmTC_454e6   && :
    /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /config/.esphome/platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: cannot open shared object file: No such file or directory
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /config/.esphome/platformio/packages/framework-espidf/tools/cmake/project.cmake:381 (__project)
  CMakeLists.txt:3 (project)
___________________________

I'm guessing that ESPHome is having issues when pulling/building the frame work for the C Builder for the Project.

I really hope that some within the ESP Team can look at this be look the get this issue resolved.  I'm happy to provide more information, run more tests, try beta builds etc.

Many Thanks

edgicat, 2024-07-05:
@jesserockz-2024-262

I notice Esphome 2004.6.6 armv7 / armhf dependencies ref issues: #7017, #7022 & #7021,  I was really hopeful this was going to fix the issue I'm experiencing,  unfortunately it didn't (sad face)

Below are the log crashes from my raspberry pi400 with a 32bit os and pi2, both run the same 32bit workworm lite os and have the same docker setup's

Pi400 - 32bit os

[logs_ep-test_run (pi400 32bit)2.txt](https://github.com/user-attachments/files/16113493/logs_ep-test_run.pi400.32bit.2.txt)

_____________________________

pi2
[logs_ep-test_run.txt](https://github.com/user-attachments/files/16113490/logs_ep-test_run.txt)

________________

Really hope this helps this issue being resolved especially as there are been other fixes applied for 32bit hardware/os

thanks

jesserockz, 2024-07-05:
Hi @edgicat, I have just come across this myself as I am rebuilding the container images to avoid some other issues. But in my rebuilding I noticed this issue when trying to compile an esp-idf 4.4.7 project in the armv7 image only. 

I'm not sure how to solve this as the errors describe a file not found but I can clearly see the files referenced. 

I am surprised you are even able to compile at all using a RPi2 given it is extremely slow and very CPU intensive.

I don't have a fix for this right now, and can only suggest that you try to upgrade the machine you are compiling on to a RPi 4/5 (aarch64/arm64) or an x86_64/amd64 based host.

edgicat, 2024-07-05:
> Hi @edgicat, I have just come across this myself as I am rebuilding the container images to avoid some other issues. But in my rebuilding I noticed this issue when trying to compile an esp-idf 4.4.7 [...]

Thanks for the update, I'm aware of the potential issue of my under powered pi, which is why I take the time to replicate on my kids Pi400.  TBH my Pi2 with USB boot (well also most USB boot :-)  ) has been incredibility stable. Whilst I don't want to needlessly retire another Pi, I have been contemplating moving to a HA Yellow (just waiting on the release of CM5) or perhaps a ProxMox server, only time will tell.  keep fighting to good fight and plugging away at those bug fixes.