esphome/issues #3077: deep_sleep not working with esp32c3

LokiMetaSmith, 2022-02-21:
### The problem

Been having issue with running deep_sleep, was hoping that https://github.com/esphome/esphome/pull/3066 would resolve the issue, but still having a problem after pulling latest dev branch with pip. Not sure if it's a local dev env issue or something else.

### Which version of ESPHome has the issue?

2022.3.0-dev

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

deep_sleep

### Example YAML snippet

```yaml
esphome:
  name: deep-sleep
esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
deep_sleep:
  run_duration: 10s 
  sleep_duration: 20min
```

### Anything in the logs that might be useful for us?

```txt
src/esphome/components/deep_sleep/deep_sleep_component.cpp:78:6: error: no declaration matches 'void esphome::deep_sleep::DeepSleepComponent::set_ext1_wakeup(esphome::deep_sleep::Ext1Wakeup)'
 void DeepSleepComponent::set_ext1_wakeup(Ext1Wakeup ext1_wakeup) { this->ext1_wakeup_ = ext1_wakeup; }
      ^~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:78:6: note: no functions named 'void esphome::deep_sleep::DeepSleepComponent::set_ext1_wakeup(esphome::deep_sleep::Ext1Wakeup)'
In file included from src/esphome/components/deep_sleep/deep_sleep_component.cpp:1:
src/esphome/components/deep_sleep/deep_sleep_component.h:56:7: note: 'class esphome::deep_sleep::DeepSleepComponent' defined here
 class DeepSleepComponent : public Component {
       ^~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:79:6: error: no declaration matches 'void esphome::deep_sleep::DeepSleepComponent::set_touch_wakeup(bool)'
 void DeepSleepComponent::set_touch_wakeup(bool touch_wakeup) { this->touch_wakeup_ = touch_wakeup; }
      ^~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:79:6: note: no functions named 'void esphome::deep_sleep::DeepSleepComponent::set_touch_wakeup(bool)'
In file included from src/esphome/components/deep_sleep/deep_sleep_component.cpp:1:
src/esphome/components/deep_sleep/deep_sleep_component.h:56:7: note: 'class esphome::deep_sleep::DeepSleepComponent' defined here
 class DeepSleepComponent : public Component {
       ^~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp:80:6: error: no declaration matches 'void esphome::deep_sleep::DeepSleepComponent::set_run_duration(esphome::deep_sleep::WakeupCauseToRunDuration)'
 void DeepSleepComponent::set_run_duration(WakeupCauseToRunDuration wakeup_cause_to_run_duration) {
      ^~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/deep_sleep/deep_sleep_component.cpp:1:
src/esphome/components/deep_sleep/deep_sleep_component.h:81:8: note: candidate is: 'void esphome::deep_sleep::DeepSleepComponent::set_run_duration(uint32_t)'
   void set_run_duration(uint32_t time_ms);
        ^~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.h:56:7: note: 'class esphome::deep_sleep::DeepSleepComponent' defined here
 class DeepSleepComponent : public Component {
       ^~~~~~~~~~~~~~~~~~
src/esphome/components/deep_sleep/deep_sleep_component.cpp: In member function 'void esphome::deep_sleep::DeepSleepComponent::begin_sleep(bool)':
src/esphome/components/deep_sleep/deep_sleep_component.cpp:137:81: error: cannot convert 'bool' to 'esp_deepsleep_gpio_wake_up_mode_t'
     esp_deep_sleep_enable_gpio_wakeup(gpio_num_t(this->wakeup_pin_->get_pin()), level);
                                                                                 ^~~~~
In file included from src/esphome/components/deep_sleep/deep_sleep_component.h:9,
                 from src/esphome/components/deep_sleep/deep_sleep_component.cpp:1:
C:/Users/Loki-VR/.platformio/packages/framework-espidf/components/esp_system/include/esp_sleep.h:262:103: note:   initializing argument 2 of 'esp_err_t esp_deep_sleep_enable_gpio_wakeup(uint64_t, esp_deepsleep_gpio_wake_up_mode_t)'
 esp_err_t esp_deep_sleep_enable_gpio_wakeup(uint64_t gpio_pin_mask, esp_deepsleep_gpio_wake_up_mode_t mode);
                                                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
*** [.pioenvs\deep-sleep\src\esphome\components\deep_sleep\deep_sleep_component.o] Error 1
=================================================================================== [FAILED] Took 15.25 seconds ===================================================================================
```

MFlasskamp, 2022-02-21:
I have the same issues. My bugfix are two changes in `/esphome/components/deep_sleep/deep_sleep_component.cpp`:

1. For ESP32-C3 `set_ext1_wakeup`, `set_touch_wakeup` and `set_run_duration` are excluded in `deep_sleep_component.h`. So I did the same in `.cpp`:
```cpp
#if !defined(USE_ESP32_VARIANT_ESP32C3)
void DeepSleepComponent::set_ext1_wakeup(Ext1Wakeup ext1_wakeup) { this->ext1_wakeup_ = ext1_wakeup; }
void DeepSleepComponent::set_touch_wakeup(bool touch_wakeup) { this->touch_wakeup_ = touch_wakeup; }
void DeepSleepComponent::set_run_duration(WakeupCauseToRunDuration wakeup_cause_to_run_duration) {
  wakeup_cause_to_run_duration_ = wakeup_cause_to_run_duration;
}
#endif
```

2. According to ESP-IDF API the [`esp_deep_sleep_enable_gpio_wakeup()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/sleep_modes.html#_CPPv433esp_deep_sleep_enable_gpio_wakeup8uint64_t33esp_deepsleep_gpio_wake_up_mode_t) needs the level as `esp_deepsleep_gpio_wake_up_mode_t` so I added a static_cast:
```cpp
esp_deep_sleep_enable_gpio_wakeup(gpio_num_t(this->wakeup_pin_->get_pin()), static_cast<esp_deepsleep_gpio_wake_up_mode_t>(level));
```

MFlasskamp, 2022-02-23:
There was another problem introduced by https://github.com/esphome/esphome/pull/3066: `esp_deep_sleep_start()` is never called for ESP32-C3 variant, [see here](https://github.com/esphome/esphome/blob/6919930aaa230f0437d0975d902e98cc7dbbf11f/esphome/components/deep_sleep/deep_sleep_component.cpp#L126).
Can be fixed easily by reordering the ifdef/endif

LokiMetaSmith, 2022-02-24:
Yep, that seemed to do it.

```

------------ esphome/components/deep_sleep/deep_sleep_component.cpp ------------
index 82751b53..04ba34bf 100644
@@ -75,6 +75,8 @@ void DeepSleepComponent::set_sleep_duration(uint32_t time_ms) { this->sleep_dura
 void DeepSleepComponent::set_wakeup_pin_mode(WakeupPinMode wakeup_pin_mode) {
   this->wakeup_pin_mode_ = wakeup_pin_mode;
 }
+#endif
+#if defined(USE_ESP32) && !defined(USE_ESP32_VARIANT_ESP32C3)
 void DeepSleepComponent::set_ext1_wakeup(Ext1Wakeup ext1_wakeup) { this->ext1_wakeup_ = ext1_wakeup; }
 void DeepSleepComponent::set_touch_wakeup(bool touch_wakeup) { this->touch_wakeup_ = touch_wakeup; }
 void DeepSleepComponent::set_run_duration(WakeupCauseToRunDuration wakeup_cause_to_run_duration) {
@@ -122,7 +124,8 @@ void DeepSleepComponent::begin_sleep(bool manual) {
     esp_sleep_enable_touchpad_wakeup();
     esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
   }
-
+#endif
+#if defined(USE_ESP32) 
   esp_deep_sleep_start();
 #endif
 
@@ -134,7 +137,7 @@ void DeepSleepComponent::begin_sleep(bool manual) {
     if (this->wakeup_pin_mode_ == WAKEUP_PIN_MODE_INVERT_WAKEUP && this->wakeup_pin_->digital_read()) {
       level = !level;
     }
-    esp_deep_sleep_enable_gpio_wakeup(gpio_num_t(this->wakeup_pin_->get_pin()), level);
+    esp_deep_sleep_enable_gpio_wakeup(gpio_num_t(this->wakeup_pin_->get_pin()), static_cast<esp_deepsleep_gpio_wake_up_mode_t>(level));
   }
 #endif
```

LokiMetaSmith, 2022-02-25:
Quick update, I've been running this and there is either something wrong with my yaml snippet above, or there's a bug somewhere. I get successful upload of a single sensor reading, connection to the wifi, and it posts it to an external server and then goes to sleep for 20 minutes, but doesn't seem to wake back up ever. I'll play around with it some more, it might be the wake-up time is too short.

MFlasskamp, 2022-02-27:
You could watch the log via serial to see when it starts deep sleep. 
I configured `run_duration: 10s` for the deep_sleep component but added the following to the sensor: 
```yaml
sensor:
  - platform: bme280
    ...
    humidity:
      name: t01c3_01 humidity
      expire_after: 15min
      on_value:
       then:
        - if:
            condition:
              not:
                lambda: "return id(ota_mode);"
            then:
              - delay: 3s
              - deep_sleep.enter: deep_sleep_1
```

This is working fine so far with a LilyGo TTGO ESP32-C3 Module (T-01C3). (I added the ESP with BME280 and BH1750 into a cheap solar lamp. In deep sleep it consumes about 13µA and wakes up every 5min for about 4-5s to send sensor values via MQTT)

Troublebrewing, 2022-04-21:
Has this been fixed in the latest release?

closed 2022-05-09 by jesserockz