esphome/issues #3080: Inkplate 6 plus touchscreen ektf2232 wrong max resolution

mkocus, 2022-02-21:
### The problem

I'm using the Inkplate 6 plus and have noticed that after switching to the official release (2022.2.4) the maximum touch x/y is reported to be 800/600 - which is invalid for the Inkplate 6 plus (which should have 1024 x 758).

I have used the
```
  - source: github://pr#3027
    components: ["ektf2232"]
```
component earlier (before this refactor: https://github.com/esphome/esphome/pull/3083 done by @jesserockz) and it worked fine.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

inkplate 6 plus

### Component causing the issue

touchscreen/ektf2232

### Example YAML snippet

```yaml
mcp23017:
  - id: mcp23017_hub
    address: 0x20

touchscreen:
  platform: ektf2232
  interrupt_pin: GPIO36
  rts_pin:
    mcp23xxx: mcp23017_hub
    number: 10
  on_touch:
  - logger.log:
      format: "touch x=%d, y=%d"
      args: ['touch.x', 'touch.y']
```

jesserockz, 2022-02-22:
Are you sure you are using `model: inkplate_6_plus` in the display yaml config?
I will hook mine up and double check I didn't break anything recently

jesserockz, 2022-02-22:
```
[13:27:16][D][main:417]: Touch at (721, 1009)
```
When touching in the bottom right corner of my display on the latest dev branch. I will try the release next.

jesserockz, 2022-02-22:
Oh wow, that makes no sense whatsoever as so why it would change the touch points :smile: . I will try to reproduce that locally when I can.

closed 2022-07-07 by github-actions[bot]