esphome/issues #3083: BME680 crashes system after last update of ESPHome

rmeissn, 2022-02-22:
### The problem

I've updated my ESP8266 setups from 2021.12 to 2022.2.4 today and started wondering, because one of the devices stopped sending values to HomeAssisstant. This is an indoor environment sensor station with two DHT22, a BH1750 and a BME680.
The device is shown in the ESPHome dashboard as "Online", but the Dashboard isn't able to connect to the device to show logs (Errno 111). I was able to wirelessly reflash the device successfully, but this didn't change anything.

So I started looking at the device itself. When powering it up, the small LED near the wifi antenna blinks 10 times. I started to pull some sensors and after pulling the BME680, the device only blinked once (instead of 10 times), started successfully and is reachable again. As soon as I plug the BME680 back in (and repowering the device) it stops working again.

As I haven't touched the device before and while updating, I guess there must be some software issue related to this behaviour. The device worked as expected 1min before updating it and stopped working after the update.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

bme680

### Example YAML snippet

```yaml
esphome:
  name: bme680
  platform: ESP8266
  board: nodemcuv2

logger:
  baud_rate: 0

api:

ota:
  password: "..."

wifi:
  ssid: "..."
  password: "..."
  output_power: 10 # in db
  domain: '...'

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bme680 Fallback Hotspot"
    password: "..."

captive_portal:

i2c:
  sda: D1
  scl: D2

bme680_bsec:
    # i2c address
    # -----------
    # Common values are:
    # - 0x76
    # - 0x77
    # Default: 0x76
    address: 0x77

    # Temperature offset
    # ------------------
    # Useful if device is in enclosure and reads too high
    # Default: 0
    temperature_offset: 0.3

    # IAQ calculation mode
    # --------------------
    # Available options:
    # - static (for fixed position devices)
    # - mobile (for on person or other moveable devices)
    # Default: static
    iaq_mode: static

    # Sample rate
    # -----------
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: ulp

    # Interval at which to save BSEC state
    # ------------------------------------
    # Default: 6h
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      # Temperature in °C
      name: "BME680 Temperature"
      filters:
        - median:
            window_size: 3
            send_every: 3
    pressure:
      # Pressure in hPa
      name: "BME680 Pressure"
      filters:
        - median:
            window_size: 3
            send_every: 3
    humidity:
      # Relative humidity %
      name: "BME680 Humidity"
      filters:
        - median:
            window_size: 3
            send_every: 3
    gas_resistance:
      # Gas resistance in Ω
      name: "BME680 Gas Resistance"
      filters:
        - median:
            window_size: 3
            send_every: 3
    iaq:
      # Indoor air quality value
      name: "BME680 IAQ"
      filters:
        - median:
            window_size: 3
            send_every: 3
    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "BME680 Numeric IAQ Accuracy"
    co2_equivalent:
      # CO2 equivalent estimate in ppm
      name: "BME680 CO2 Equivalent"
      icon: "mdi:molecule-co2"
      filters:
        - median:
            window_size: 3
            send_every: 3
    breath_voc_equivalent:
      # Volatile organic compounds equivalent estimate in ppm
      name: "BME680 Breath VOC Equivalent"
      filters:
        - median:
            window_size: 3
            send_every: 3
#
  - platform: dht
    model: DHT22
    pin: D5
    temperature:
      name: "DHT1 Temperature"
      filters:
        - median:
            window_size: 3
            send_every: 3
    humidity:
      name: "DHT1 Humidity"
      filters:
        - median:
            window_size: 3
            send_every: 3
    update_interval: 5min
#
  - platform: dht
    model: DHT22
    pin: D6
    temperature:
      name: "DHT2 Temperature"
      filters:
        - median:
            window_size: 3
            send_every: 3
    humidity:
      name: "DHT2 Humidity"
      filters:
        - median:
            window_size: 3
            send_every: 3
    update_interval: 5min
#
  - platform: bh1750
    name: "BH1750 Brightness"
    address: 0x23
    resolution: 1.0
    measurement_duration: 69
    update_interval: 5min
    filters:
      - median:
          window_size: 3
          send_every: 3

text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "BME680 IAQ Accuracy"
```

trvrnrth, 2022-02-22:
Interersting. I've just updated my test device. It's still running and updating the display I have connected but as you say I can't connect to view logs and the HA API connection isn't up either.

rmeissn, 2022-02-23:
> Tell a lie. It has now connected. It just took way way longer then normal.

How much time did you spent waiting? I've waited for about 2 minutes and decided that the device won't come up...

chefhb88, 2022-02-23:
Same problem here. 
But still no connection after >24 hours...

trvrnrth, 2022-02-23:
Had a chance to do a very quick bit of testing. With 2021.1.4, WIFI connection is fast. With 2022.2.4, I get repeated 'Auth Expired' failures, before an eventual connection. So one of the core changes must be at play here. Though it wouldn't be much of a surprise if it's the extra overhead of running the BSEC comms that break's the camels back.

trvrnrth, 2022-02-23:
Looks like it might be the [arduino framework bump](https://github.com/esphome/esphome/pull/2606). Try setting `arduino_version: "2.7.4"`([docs](https://esphome.io/components/esphome.html#configuration-variables)) and see if that helps.

chefhb88, 2022-02-23:
Hi trvrnrth,
looks good. With this setting (arduino_version) the device is reachable again. ;-)

chefhb88, 2022-02-23:
Looks that the problem is not solved completely.
After quite a while the device is restarting with log message:
[17:08:52][D][debug:256]: Reset Reason: Software Watchdog
[17:08:52][D][debug:257]: Reset Info: Fatal exception:4 flag:3 (Software Watchdog) epc1:0x4026142a epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

ajpalm, 2022-02-23:
I took am having issues, since updating ESPHOME to the latest version, my ESP32 that has a BME680 connected, started crashing every few minutes. Commenting out the bme680_bsec code has resolved the rebooting issue. Happy to provide logs etc as required.

JohnnyM84, 2022-02-26:
I have the same problem, my Wemos d1 mini crashes sometimes after an hour or after a fee days but eventually it crashes.
Will try now with the old framework:
framework:
    version: 2.7.4

enurist, 2022-03-13:
Same here. I was about to throw it away when I accidently found this issue. Because I have to other similar devices which run well I was digging around and found an issue and a solution:

When setting up both multiple wifi-networks AND time platform sntp the device was not able to boot correctly. Setting up one of them leads to no error.

chefhb88, 2022-03-24:
Hi,
any news ? Workaround with arduino_version: "2.7.4" only helps to get the "functions" (webserver / mqtt)
running again. The device is still rebooting after about 60 mins..

chefhb88, 2022-03-24:
PS: I don't have multiple wifi or sntp defined...

github-iw, 2022-03-26:
Same problem here after updating mutiple esp8266 devices in combination with BME680: devices online but webserver not available and mqtt connection unstable

trvrnrth, 2022-05-12:
If anyone gets a chance to try the latest beta it would be interesting to know if the [change to queue sensor publishes](https://github.com/esphome/esphome/pull/3422) improves things for you (both with the old and new arduino versions).

chefhb88, 2022-05-12:
Hi Trevor,

i think i can try, if you would give me hint, how to implement / install the latest beta in home assistant. Is there a switch ?

Von: Trevor North
Gesendet: Donnerstag, 12. Mai 2022 13:35
An: esphome/issues
Cc: chefhb88; Comment
Betreff: Re: [esphome/issues] BME680 crashes system after last update ofESPHome (Issue #3083)

If anyone gets a chance to try the latest beta it would be interesting to know if the change to queue sensor publishes improves things for you (both with the old and new arduino versions).

trvrnrth, 2022-05-12:
@chefhb88 You should be able to install the beta addon from https://github.com/esphome/home-assistant-addon

JohnnyM84, 2022-05-12:
I have changed the 2 files in my docker container almost 2 weeks ago. Used the default framework and it behaves now the same as with framework 2.7.4. Uptimes from 1 day till a few weeks. Still some reboots but before the fix and the new framework the uptime was max 4 hours.

JohnnyM84, 2022-05-20:
Is this fix in esphome 2022.5.0?

trvrnrth, 2022-05-20:
> Is this fix in esphome 2022.5.0?

Yes.

chefhb88, 2022-05-20:
I installed 2022.5.0 today. 
But Problem not solved:    Ping ok. But API / Webserver / MQTT dead.

Von: Trevor North
Gesendet: Freitag, 20. Mai 2022 14:33
An: esphome/issues
Cc: chefhb88; Mention
Betreff: Re: [esphome/issues] BME680 crashes system after last update ofESPHome (Issue #3083)

Is this fix in esphome 2022.5.0?
Yes.

trvrnrth, 2022-05-20:
> I installed 2022.5.0 today. But Problem not solved: Ping ok. But API / Webserver / MQTT dead.
@chefhb88 You might want to try without the webserver as that's pretty memory hungry.

chefhb88, 2022-05-20:
Test:
With webserver off  no API / MQTT  too. ☹

Von: Trevor North
Gesendet: Freitag, 20. Mai 2022 14:44
An: esphome/issues
Cc: chefhb88; Mention
Betreff: Re: [esphome/issues] BME680 crashes system after last update ofESPHome (Issue #3083)

I installed 2022.5.0 today. But Problem not solved: Ping ok. But API / Webserver / MQTT dead.
@chefhb88 You might want to try without the webserver as that's pretty memory hungry.

IGNNE, 2022-07-31:
Can confirm: ESP8266 (esp_wroom_02), ping/ota works, everything else doesn't. It seems like it boot loops, and eventually ends up in safe mode. Rolling back the docker image to 2022.1.4 makes it work again. 

This is the error I get when I only add the `bme680_bsec:` and not the sensor component:
`(BSEC Status: 12, BME680 Status: 0)`

According to https://github.com/BoschSensortec/BSEC-Arduino-library/blob/7a9357566c75048331c15b55a4eb20f1de14f36e/src/inc/bsec_datatypes.h#L310 it is the following error:
`/*!< No output (virtual) sensor data were requested via bsec_update_subscription() */`

jetpacktuxedo, 2022-08-30:
I just hit this too on esphome 2022.8.0 on two different devices. Both ran fine for 3+ days and then locked up, reporting as online and flashable but not pushing any values into homeassistant. after dropping any of the bosc-specific stuff and just using the basic BME680 sensors it started working again.

zevs44, 2022-09-09:
Hello, the same crushes for me on ESPHome v2022.8.3. I2C communication error, code 2. Basic BME680 sensors working. Problem on bosc-specific stuff

r100gs, 2022-09-19:
Hello, I still have the same problem with bsec

trvrnrth, 2022-09-19:
If anyone is able to try out the changes in #3550 it makes a number of improvements to resource utilisation.

r100gs, 2022-10-19:
Hello,

Thx to erybody! With 2022.10.00 Update my Sensors are working again.

Best regards, Stefan