esphome/issues #3084: RGBW Lights strange behaviour since version 2022.2.4 (possibly earlier)

sparkydave1981, 2022-02-23:
### The problem

I updated my RGBW downlights from 2021.12.x to 2022.2.4 and now they won't fully turn off, plus brightness control is erratic.

When turning the lights off from Home Assistant the light dims down to very low brightness but stays on even though HA reports it as off.  

Brightness control is not as it was previously with the control being erratic and not properly following that of the brightness slider in HA.  It seems that they stay very dim unless the brightness is set to 100%.

Other than updating the ESPhome version no other settings were changed and the lights worked perfectly previously.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

rgbw light (my9231)

### Example YAML snippet

```yaml
my9231:
 data_pin: GPIO13  # GPIO13 for AiLight
 clock_pin: GPIO15  # GPIO15 for AiLight
 num_channels: 4
 num_chips: 1

output:
  - platform: my9231
    id: output_blue
    channel: 1
  - platform: my9231
    id: output_red
    channel: 3
  - platform: my9231
    id: output_green
    channel: 2
  - platform: my9231
    id: output_white
    channel: 0

light:
  - platform: rgbw
    name: Living Room Downlight 3
    red: output_red
    green: output_green
    blue: output_blue
    white: output_white
    color_interlock: true
    id: living
```

### Anything in the logs that might be useful for us?

```txt
No errors in log
```

sparkydave1981, 2022-03-01:
I'm also finding some lights randomly turning on, but again, HA still thinks they are off so it's like they aren't reporting that state back.

YvesZH, 2022-03-02:
I have the same issue with all my 'my9231' lights after i updating from 2021.12.x. to 2022.2.5.
After noticing the error, i activated the web server on the affected esp.  Even there the toggle button doesn't work (same effect), although everything is displayed correctly in the log.

YvesZH, 2022-03-02:
another addition: the problems did not occur when i updated esphome to home assistant.  The problems started when I updated the esp/lamps from 2021.12.x to 2022.2.5 a few days later.

sparkydave1981, 2022-03-02:
> I have the same issue with all my 'my9231' lights after i updating from 2021.12.x. to 2022.2.5. After noticing the error, i activated the web server on the affected esp. Even there the toggle button doesn't work (same effect), although everything is displayed correctly in the log.

Well I'm glad I'm not the only person with this problem, so hopefully it gets fixed for us.

In the meantime do you know how we can install an old version of ESPhome?  I tried with Docker but couldn't access the GUI as I obviously had something wrong...

sparkydave1981, 2022-03-02:
> another addition: the problems did not occur when i updated esphome to home assistant (from 2021.12.x to 2022.2.5). The problems started when I updated the esp/lamps OTA from 2021.12.x to 2022.2.5 a few days later.

Yes, updating the version of ESPhome within HA doesn't really do anything until you send a new .bin file to the device itself.

sparkydave1981, 2022-03-03:
Also, I have a forum thread about this [here](https://community.home-assistant.io/t/esphome-rgbw-lights-behaving-strange-since-upgrade-to-2022-2-4/396147).

dylan09, 2022-03-03:
Did you use Home Assistant API integration or MQTT?

YvesZH, 2022-03-03:
> Did you use Home Assistant API integration or MQTT?

I use HA API

dylan09, 2022-03-03:
Ok. Than its not directly related to some of my problems. They began if I remember correctly with 2021.10.
There where some changes to ESPHome light integration.

maybe you could test your light with ```color_interlock: false```? Be careful with brightness if you switch on RGB and white together.

YvesZH, 2022-03-04:
> Ok. Than its not directly related to some of my problems. They began if I remember correctly with 2021.10.
> There where some changes to ESPHome light integration.

I roll back to 2021.11.4 and its work as aspect (after compiling and uploading the ESP/lamps).  I remember the led-esp change (was nice), but it had nothing to do with it.  Do you know someone you could involve in this topic who can help?

esplor, 2022-03-04:
I have the same problems with my Sonoff B1's rgbww.
After upgrading esphome to 2022.2.X it stopped working, would not change colours or even turn on/off.

The workaround for me was to set older framework version:

Complete config that works:
```
esphome:
  name: REPLACEME

esp8266:
  board: esp01_1m
# After upgrade ESPHOME 2022.2.X Sonoff B1 stopped working
  framework:
    version: 2.7.4

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

logger:
api:
ota:
web_server:

my9231:
  data_pin: GPIO12  # GPIO13 for AiLight
  clock_pin: GPIO14  # GPIO15 for AiLight
  num_channels: 6
  num_chips: 2

output:
  - platform: my9231
    id: output_blue
    channel: 0
  - platform: my9231
    id: output_red
    channel: 1
  - platform: my9231
    id: output_green
    channel: 2
  - platform: my9231
    id: output_warm_white
    channel: 4
  - platform: my9231
    id: output_cold_white
    channel: 5

light:
  - platform: rgbww
    name: REPLACEME
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold_white
    cold_white_color_temperature: 6500 K
    warm_white: output_warm_white
    warm_white_color_temperature: 2800 K
```

esplor, 2022-05-20:
@sparkydave1981

My Sonoff B1 works as before 2022.2.X
I've upgraded esphome to 2022.5.0 right now, disabled force framework version.. OTA upgrade.

Nothing in the changelog about my9231 though.

YvesZH, 2022-10-07:
any news about?

sparkydave1981, 2022-10-10:
I'm still including the framework: code in my config, having not tested without it in more recent ESPhome versions.  I do still have trouble with one light which flickers and has messed up colour selection. (it worked perfectly about a year ago.  I haven't had time to do any testing with that light.

YvesZH, 2023-03-17:
@sparkydave1981 your workaround work great (Thanks). Since the ESP-Home Update 2023.3, the compiling work anymore:

```
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/01-eingang-spot-1/.pioenvs/01-eingang-spot-1/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/01-eingang-spot-1/.pioenvs/01-eingang-spot-1/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/01-eingang-spot-1/.pioenvs/01-eingang-spot-1/src/esphome/components/api/api_connection.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [/data/01-eingang-spot-1/.pioenvs/01-eingang-spot-1/src/esphome/components/api/api_pb2.cpp.o] Error 1
========================== [FAILED] Took 1.45 seconds ==========================
```

However, the lights can be controlled as expected.

sparkydave1981, 2023-03-18:
> Since the ESP-Home Update 2023.3, the compiling work anymore:

Do you mean _doesn't_ work anymore?  In this case I'm guessing you are still actually running an older version of ESPhome on the light...?

YvesZH, 2023-03-19:
> Do you mean _doesn't_ work anymore? In this case I'm guessing you are still actually running an older version of ESPhome on the light...?

@sparkydave1981   yes, that's what I meant "doesn't work anymore" 🙈

The compiled code from ESPhome 2023.2 (with your workaround) runs on the esp lights.  My ESPhome on the server is now on 2023.3.  The esp lights can be controlled without error in this constellation. Only when I compile (same yaml configuration as before) does the above error message appear.

sparkydave1981, 2023-03-20:
> The compiled code from ESPhome 2023.2 (with your workaround) runs on the esp lights. My ESPhome on the server is now on 2023.3. The esp lights can be controlled without error in this constellation. Only when I compile (same yaml configuration as before) does the above error message appear.

Right.  Well the version of ESPhome that you have installed on HA plays no role in the function of the lights, only what has been flashed to the light (in your case 2023.2)

sparkydave1981, 2023-03-28:
> Since the ESP-Home Update 2023.3, the compiling work anymore:

Probably best to create a new Issue based on this error.  I haven't yet updated ESPhome so I haven't tried compiling on the 2023.3 version.