esphome/issues #3085: Unable to get an M5 Atom Lite working

jake404, 2022-02-23:
### The problem

I'm trying to set up an M5 Atom Lite device but ESPHome gives me the error message 'Unable to compile the configuration' when going through the wizard. If I skip the installation part of the wizard, the device is created. However, when I later click on the three dots and then Install, I get the error message 'Failed to prepare the configuration. Something went wrong.'.  If I then click on 'Something went wrong.', it is empty.

When I edit the configuration of the device, it looks correct. When I click on Validate the dialog is empty and there is a red X at the top. If I go to the Logs for the device, it is at first empty but when I reset the device it shows the startup of the existing code.

I read about similar issues here but I have tried the recommended solutions. I have uninstalled the addon in HA, restarted HA, and reinstalled the ESPHome addon.

Is this a known issue? Have I missed a step?

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP32

### Board

M5 Atom Lite

### Example YAML snippet

```yaml
esphome:
  name: m5

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "ff4192e83ecfa30efed2c7b011e59b37"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Test Fallback Hotspot"
    password: "uHR9hZEJBS98"

captive_portal:
```

khenderick, 2022-02-23:
I use a bunch M5Stack Atom Lites with the following configuration:

```
esphome:
  platform: ESP32
  board: m5stack-atom
```

jake404, 2022-02-23:
I have tried multiple boards including the one you mentioned without success.

Yorik-Code, 2022-03-11:
Try this. I've found the following syntax to work better.

```
esphome:
  name: m5
  platform: ESP32
  board: m5stack-atom

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "ff4192e83ecfa30efed2c7b011e59b37"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Test Fallback Hotspot"
    password: "uHR9hZEJBS98"

captive_portal:
```

randybb, 2022-03-11:
For ESP32-Pico you need to choose a different board. I have been using m5stack-atom on ~10 atoms and no issues.

closed 2022-03-17 by jake404