esphome/issues #3086: BMS Daly EspHome DEV

Didou30, 2022-02-23:
### The problem

Ajout des tensions des cellule de 1 à 16 ne fonctionne pas, il ne remonte que la Température

### Which version of ESPHome has the issue?

V 2022.2.5-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

Daly BMS

### Example YAML snippet

```yaml
esphome:
  name: bmsdaly

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "057e5e176a11c4f5ac1cab965ff9bf6c"

wifi:
  ssid: !secret ssid
  password: !secret mp_wifi

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bmsdaly Fallback Hotspot"
    password: "MzJE8fEg3jGy"

captive_portal:

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 9600
  
switch:
  - platform: restart
    name: "bms_daly_restart"
    
daly_bms:
  update_interval: 20s

sensor:
[... 228 lines cut ...]
[21:01:36][C][ota:085]: Over-The-Air Updates:
[21:01:36][C][ota:086]:   Address: bmsdaly.local:8266
[21:01:36][C][ota:089]:   Using Password.
[21:01:36][C][api:138]: API Server:
[21:01:36][C][api:139]:   Address: bmsdaly.local:6053
[21:01:36][C][api:143]:   Using noise encryption: NO
[21:01:36][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi puissance_BMS_Daly'
[21:01:36][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:01:36][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:01:36][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:01:36][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:01:43][D][sensor:124]: 'Max Cell Voltage': Sending state 3.32200 V with 2 decimals of accuracy
[21:01:43][D][sensor:124]: 'Max Cell Voltage Number': Sending state 13.00000  with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Min Cell Voltage': Sending state 3.31500 V with 2 decimals of accuracy
[21:01:43][D][sensor:124]: 'Min Cell Voltage Number': Sending state 1.00000  with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Max Temperature': Sending state 13.00000 °C with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Max Temperature Probe Number': Sending state 1.00000  with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Min Temperature': Sending state 13.00000 °C with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Min Temperature Probe Number': Sending state 1.00000  with 0 decimals of accuracy
[21:01:43][D][text_sensor:067]: 'BMS Status': Sending state 'Discharging'
[21:01:43][D][sensor:124]: 'Remaining Capacity': Sending state 6.93360 Ah with 2 decimals of accuracy
[21:01:43][D][sensor:124]: 'Cells Number': Sending state 16.00000  with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Temperature 1': Sending state 13.00000 °C with 0 decimals of accuracy
[21:01:43][D][sensor:124]: 'Temperature 2': Sending state -40.00000 °C with 0 decimals of accuracy
[21:01:47][D][sensor:124]: 'WiFi puissance_BMS_Daly': Sending state -56.00000 dBm with 0 decimals of accuracy

Didou30, 2022-02-23:
le code testé en Esphome V 2022.2.5-dev
esphome:
  name: bmsdaly

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "057e5e176a11c4f5ac1cab965ff9bf6c"

wifi:
  ssid: !secret ssid
  password: !secret mp_wifi

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bmsdaly Fallback Hotspot"
    password: "MzJE8fEg3jGy"

captive_portal:

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 9600
  
switch:
  - platform: restart
    name: "bms_daly_restart"
    
daly_bms:
  update_interval: 20s

sensor:
  - platform: daly_bms
    voltage:
      name: "Battery Voltage"
    current:
      name: "Battery Current"
    battery_level:
      name: "Battery Level"
    max_cell_voltage:
      name: "Max Cell Voltage"
    max_cell_voltage_number:
      name: "Max Cell Voltage Number"
    min_cell_voltage:
      name: "Min Cell Voltage"
    min_cell_voltage_number:
      name: "Min Cell Voltage Number"
    max_temperature:
      name: "Max Temperature"
    max_temperature_probe_number:
      name: "Max Temperature Probe Number"
    min_temperature:
      name: "Min Temperature"
    min_temperature_probe_number:
      name: "Min Temperature Probe Number"
    remaining_capacity:
      name: "Remaining Capacity"
      filters:
      - lambda: return x * 0.04;
    cells_number:
      name: "Cells Number"
    temperature_1:
      name: "Temperature 1"
    temperature_2:
      name: "Temperature 2"
    cell_1_voltage:
      name: "Cell 1 Voltage"
    cell_2_voltage:
      name: "Cell 2 Voltage"
    cell_3_voltage:
      name: "Cell 3 Voltage"
    cell_4_voltage:
      name: "Cell 4 Voltage"
    cell_5_voltage:
      name: "Cell 5 Voltage"
    cell_6_voltage:
      name: "Cell 6 Voltage"
    cell_7_voltage:
      name: "Cell 7 Voltage"
    cell_8_voltage:
      name: "Cell 8 Voltage"
    cell_9_voltage:
      name: "Cell 9 Voltage"
    cell_10_voltage:
      name: "Cell 10 Voltage"
    cell_11_voltage:
      name: "Cell 11 Voltage"
    cell_12_voltage:
      name: "Cell 12 Voltage"
    cell_13_voltage:
      name: "Cell 13 Voltage"
    cell_14_voltage:
      name: "Cell 14 Voltage"
    cell_15_voltage:
      name: "Cell 15 Voltage"
    cell_16_voltage:
      name: "Cell 16 Voltage"
    
 # - plateform: template
      #name: "Capacité en KW"
      #remaining_capacity*0.04
      
      
      
#intensité du signal wifi
  - platform: wifi_signal
    name: "WiFi puissance_BMS_Daly"
    update_interval: 60s

text_sensor:
  - platform: daly_bms
    status:
      name: "BMS Status"

binary_sensor:
  - platform: daly_bms
    charging_mos_enabled:
      name: "Charging MOS"
    discharging_mos_enabled:
      name: "Discharging MOS"

Didou30, 2022-02-27:
Anythink for help?

codewise-nicolas, 2022-09-09:
Me 2...not all values were populating from my bms. Specially temp and battery current.

Thanks to the insights from @matthias882, I added
`  rx_buffer_size: 512`
To my UART line and I seem to be receiving all values properly now.

I dont know if that will work in all situations or if its even the right value (who knows if im loosing a few extra bits near the end), but unlike @matthias882 that said it didnt help him (possibly he is on esp8266 which uses soft uart?), it did for me.
Im on an ESP32, I only have 8 cells, 1 temp sensor.

Digging further into the daly_bms.cpp update() function its clear that Temp is one of the last requested items, so it makes sense that a buffer of 256 (default) could possibly be loosing the bits related to the last few requested data.

Ive briefly looked thru @matthias882 pull request for a state machine change and overall it looks good. I will see about trying it later. Let me know if there is any way I can help with this issue.

closed 2023-08-10 by jesserockz