esphome/issues #3087: Initial state at reset/boot not updating in HA

jerobins, 2022-02-23:
### The problem

Existing binary sensor working well for months. Works properly with esphome 2022.1.4. No longer reports initial state at boot/reset. DEBUG log says initial state is being reported, but does not reflect in Home Assistant.

All subsequent state changes are reflected properly.

uptime sensor first updates at approx 6 seconds in HA; however, that update is not reflected in the DEBUG log.
First uptime sensor DEBUG log is in the 40 second time frame. See below.

### Which version of ESPHome has the issue?

2022.2.X

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

binary_sensor gpio platform

### Example YAML snippet

```yaml
binary_sensor:
  - platform: gpio
    name: ${upper_device_name}
    pin:
      number: GPIO13
      inverted: true
      mode:
        input: true
        pullup: true
    device_class: door
    filters:
      - delayed_off: 10s
```

### Anything in the logs that might be useful for us?

```txt
ESPHome debug outputs "Sending initial state OFF" but is not reflected in HA.
-----
[21:17:21][C][logger:233]: Logger:
[21:17:21][C][logger:234]:   Level: DEBUG
[21:17:21][C][logger:235]:   Log Baud Rate: 115200
[21:17:21][C][logger:236]:   Hardware UART: UART0
[21:17:21][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Mailbox Door Sensor'
[21:17:21][C][gpio.binary_sensor:015]:   Device Class: 'door'
[21:17:21][C][gpio.binary_sensor:016]:   Pin: GPIO13
[21:17:21][C][uptime.sensor:031]: Uptime Sensor 'Mailbox Door Sensor Uptime'
[21:17:21][C][uptime.sensor:031]:   State Class: 'total_increasing'
[21:17:21][C][uptime.sensor:031]:   Unit of Measurement: 's'
[21:17:21][C][uptime.sensor:031]:   Accuracy Decimals: 0
[21:17:21][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[21:17:21][C][mdns:084]: mDNS:
[21:17:21][C][mdns:085]:   Hostname: mailbox-door-sensor
[21:17:21][C][ota:085]: Over-The-Air Updates:
[21:17:21][C][ota:086]:   Address: mailbox-door-sensor.local:8266
[21:17:21][C][api:138]: API Server:
[21:17:21][C][api:139]:   Address: mailbox-door-sensor.local:6053
[21:17:21][C][api:143]:   Using noise encryption: NO
[21:17:24][D][api:102]: Accepted 192.168.1.168
[21:17:24][D][api.connection:826]: Home Assistant 2022.2.9 (192.168.1.168): Connected successfully
[21:17:24][D][binary_sensor:034]: 'Mailbox Door Sensor': Sending initial state OFF
[21:17:50][D][sensor:124]: 'Mailbox Door Sensor Uptime': Sending state 41.80500 s with 0 decimals of accuracy
```

jerobins, 2022-03-03:
Still present on 2022.2.6

closed 2022-04-29 by jerobins

Drealine, 2022-06-05:
I've the same problem