esphome/issues #3088: Update All Crash all my esphome device

djiesr, 2022-02-23:
### The problem

After the last update, I do a "update all" and now, all my device are offline because they crash during the update. 
[image]

### Which version of ESPHome has the issue?

2022.2.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

all bord, esp32 8266 sonoff, etc

djiesr, 2022-02-23:
They still work, but appear Offline in ESPHome

Plampi64, 2022-02-26:
I seems to have same issue. After updated (luckily one one) ESP32, Wifi connection was not vivible anymore. Odd thing is that if i connect updated ESP32 via USB cable to HA (Raspberry), Wifi comes back visible and working. Only error i can see after getting it back online was: [12:49:06][W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts. Now if i agin disconnect ESP32 from HA and connect ESP32 to power source only, no wifi visible thus offline state. Odd....