esphome/issues #3092: xtensa-lx106-elf-g++: not found

stevemutch, 2022-02-24:
### The problem

When compiling new nodes the following error arises. Worked prior to 2022.2.5 upgrade.

|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <AsyncMqttClient-esphome> 0.8.6
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ArduinoJson> 6.18.5
Compiling /data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [/data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_pb2.cpp.o] Error 127
Compiling /data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/15a-powr2-02/.pioenvs/15a-powr2-02/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 2.77 seconds ==========================

### Which version of ESPHome has the issue?

2022.2.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

All boards have this issue ESP8266 and ESP32

### Anything in the logs that might be useful for us?

```txt
See above
```

Kyrunix, 2022-03-04:
Hi, i had the same problem.

After searching a little bit in system logs, i could find out that my update process of platformio, xtensa (and so on) crashed. Platformio was successfully installed, but xtensa and everything else was missing. On device install, esphome check only the version of platformio. Thats the real problem i think.

I reinstall esphome to clean it up. After this, it works normal. Maybe it is possible to delete only platformio, i don't know.

You can backup your yaml files for this, but it should not be necessary, because they are in a different directory. 

Hope this is helpfull for you

thoompje, 2022-03-09:
> Hi, i had the same problem.
>
> After searching a little bit in system logs, i could find out that my update process of platformio, xtensa (and so on) crashed. Platformio was successfully installed, [...]

I reinstalled the addon in Home Assistant but still no luck. What did you else?

INFO Reading configuration /config/esphome/shutters.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing shutters (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/shutters/.pioenvs/shutters/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/shutters/.pioenvs/shutters/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/shutters/.pioenvs/shutters/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/shutters/.pioenvs/shutters/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/shutters/.pioenvs/shutters/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/shutters/.pioenvs/shutters/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/shutters/.pioenvs/shutters/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/shutters/.pioenvs/shutters/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 2.35 seconds ==========================

doabigcheese, 2022-04-14:
i had same issue, and just deleted the .platformio folder ... with the next compile, everything got downloaded again and now it woks...

juavinas, 2022-04-24:
Hello, 
    where can I find .platformio folder?
I get only these:
[image]

Thank you

lambu76, 2022-04-29:
same problem. any info ?

juavinas, 2022-04-29:
I uninstalled esphome from home assistant & installed. It was fixes. No more issues.

lambu76, 2022-04-29:
what about projects during uninstall ? are they deleted or kept ?

juavinas, 2022-04-29:
Left, create back-up for any issues

Tobix99, 2022-08-09:
The problem is probably that the OS is a 64Bit OS. So the PlatformIo Tools are 32Bit compatible. The only solution at the moment is to prefix any command with

```setarch arm``` (for example: ```setarch arm esphome compile FILE.yml```)

A clue to see if you are running a 32 or 64 Bit OS is to execute this command: ```uname -m```

If it is ```aarch64```, then it is a 64Bit OS. 

But I don't know how to use the Webinterface with this "fix", because every command there is executed automatically. I always have to go into the docker with ```exec``` and execute the command in this environment.

KlausHans, 2022-08-22:
You are correct, i am using 64 bit.
Simply switching back to 32 bit seems not possible, Home Assistant won't start anymore.
Why cant ESPHome recognize the architecture by itself to prefix the commands accordingly? Sounds like a trivial if-else to me. ;)
It seems ESPHome is just not usable on a Raspberry with 64 bit if you want to use ESP8266. That would be very strange to say it diplomatic.

ssieb, 2022-08-22:
It's nothing to do with esphome.  Esphome passes the compiling to platformio which can only use the compilers provided by Espressiv.  You also have some other misconceptions.  The https requirement is from the browser, not esphome.  The browser will not let you use web serial unless it's over https.

breimann, 2022-09-13:
> Hello, where can I find .platformio folder? I get only these: [image]
> 
> Thank you

I also have the same issue and am getting "Error 1". I also can't find the .platformio folder and get similar response to "find | grep platformio"

breimann, 2022-09-13:
I have uninstalled and reinstalled esphome.
This is so frustrating as I was really getting into creating nodes this way!

KlausHans, 2022-09-13:
> It's nothing to do with esphome. Esphome passes the compiling to platformio which can only use the compilers provided by Espressiv. You also have some other misconceptions. The https requirement is from the browser, not esphome. The browser will not let you use web serial unless it's over https.

Well, i have the issue as an enduser with esphome. And in this state esphome is not usable **for me**.
Granted, the https thing is browser politics. Still very convenient.

kjh81512, 2022-10-01:
I had same problem in docker Esphome.
But I solved this, with installing 32bit docker image (esphome/esphome-armv7:latest).

mkress, 2023-01-23:
I still have the same problem.
Raspberry Pi 64 bit (rapsbian) with esp8266 (wemos d1 mini)
logging into the esp docker bash and running the command
setarch arm esphome compile esphome-xyz.yml solves the problem.
but not really a handy solution - I'm really wondering that only a small amount of people are facing this issue.

Samu84, 2023-01-28:
I found a way to fix this problem, when you install a docker on RPI with 32bit OS but kernel is modificated to 64bit (like me), usually you install esphome docker using name image: esphome/esphome:2022.11, is going to download the image for 32bit, instead you need to force the install of the 64bit image using SHA256 for example :
image: esphome/esphome:2022.11@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d

kjh81512, 2023-01-28:
Thanks a lots.

I install 64 bit esphome with your advise

   esphome:

        container_name: esphome

           image: esphome/esphome-aarch64:latest

thank you.

Samu84, 2023-01-28:
happy that is working also for you, i was becoming crazy for 2 months

thoompje, 2023-01-28:
How do I remove and reinstall the container to the right one? And keeping my configuration. Thanks a lot for the answer.

Samu84, 2023-01-28:
I think to be safe you need to backup all the YAML Configurations, better all the file inside in the CONFIG directory, then to be sure, after you stop the docker, rename the directory config to configBAK, so you have all the files if it happen some problem, you just need to change only the image then restart the docker and a new config directory will be generated (if is not generated the new directory, press on recreate up on the right), then inside CONFIG directory you put the Yaml configurations. 
If you have problems and you need to delete everything, IS IMPORTANT TO SAVE ALL THE YAML FILE.

thoompje, 2023-01-30:
> I think to be safe you need to backup all the YAML Configurations, better all the file inside in the CONFIG directory, then to be sure, after you stop the docker, rename the directory config to configBAK, so you have all the files if it happen some problem, you just need to change only the image then restart the docker and a new config directory will be generated (if is not generated the new directory, press on recreate up on the right), then inside CONFIG directory you copy the YAML configurations you just made backup. If you have problems and you need to delete everything, IS IMPORTANT TO SAVE ALL THE YAML FILE.

Hi, thanks for all the information. I am not able to start the new container.. Does anyone know what i did wrong?

```
@raspberrypi:/var/lib/docker/containers# mv 23c749573aba1b2834a3695c2fd8d7e0862440abd41d7d521148b117503dacf2 23c749573aba1b2834a3695c2fd8d7e0862440abd41d7d521148b117503dacf2.bak
@raspberrypi:/var/lib/docker/containers# docker pull esphome/esphome:2022.11@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d
docker.io/esphome/esphome@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d: Pulling from esphome/esphome
2203022c5aa9: Pull complete
4407a1cd79bc: Pull complete
f587014541a1: Pull complete
adeff0f283a1: Pull complete
3955cc8fcd29: Pull complete
fe6b21e35dec: Pull complete
c2f4b085eabd: Pull complete
67c94b5d3861: Pull complete
9f371c88fe4d: Pull complete
Digest: sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d
Status: Downloaded newer image for esphome/esphome@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d
docker.io/esphome/esphome:2022.11@sha256:ebfbdd11f29c1a2bacd554476d5d8f2981123fe8f7e3636005bff032d293ea3d

@raspberrypi:/var/lib/docker/containers# docker start esphome
Error response from daemon: No such container: esphome
Error: failed to start containers: esphome
@raspberrypi:/var/lib/docker/containers# docker start esphome/esphome:2022.11
Error response from daemon: No such container: esphome/esphome:2022.11
Error: failed to start containers: esphome/esphome:2022.11
@raspberrypi:/var/lib/docker/containers# docker start esphome/esphome-aarch64:latest
Error response from daemon: No such container: esphome/esphome-aarch64:latest
```

What is the command to start the new container?

thoompje, 2023-01-31:
Can someone please help? Many thanks!

Samu84, 2023-01-31:
do you have portainer CE? there it would be more simple

thoompje, 2023-01-31:
> do you have portainer CE? there it would be more simple you are updating an image on a container that is running, you need to stop it, then update, then run it, probably the name of the container is different. You need t find the name of the container
> 
> you need to setup the place where are all the config files if you use command, i advice to use portainer ce because you can setup everything from browser
> 
> to be faster i created SMB roofs share so i can see all the system files from SMB, copy files then with portainer CE make all the changings

I now have portainer and got the message: Stopped for 2 minutes with exit code 159

What i can find is that my arcitecture is not 64x, but if i check my raspberry i see the following:

```
@raspberrypi:~# uname -m
aarch64
r@raspberrypi:~#
```

Samu84, 2023-01-31:
Please sens the sceen with the error code

thoompje, 2023-01-31:
[image]

Samu84, 2023-01-31:
I need to see the log screen

thoompje, 2023-01-31:
> […](#)

There are now logs because it won't start the container... 

[image]

Samu84, 2023-01-31:
It doesn't look like an error for the arcitecture, try to kill this (esphome), create a new, with a new name and try with the 64bit image

thoompje, 2023-01-31:
Got it working after duplicate the original with the new image... But now the following message in Esphome

[image]

Samu84, 2023-01-31:
Have you used the last version 64bit?

thoompje, 2023-01-31:
Oke what is the case. I had the DEPRECATED version of ESPHome but now installed the new one. Every time when i start the add on in Home Assistant the original image is going back to the container. What can i do in this case? Sorry for this noob question, but its not working at all at the moment :)

[image]
[image]
[image]

thoompje, 2023-02-01:
> Have you used the last version 64bit?
> 
> 
> 
> hold the BAK folder and delete the original config folder
> 
> try copy only the yaml files and not all the files inside the config folder, i think the platformio you had is the old version, and you copied all the file in the bak folder to the new

You have any idea?

thoompje, 2023-02-01:
Everytime when i stop the add-on in Home Assistant the container automatically be deleted... And when i duplicate the container and change it to the other image then it start up but does nothing, it hangs on `2023-02-01 10:50:23,601 INFO Starting dashboard web server on http://0.0.0.0:6052 and configuration dir /config...` And from within Home Assistant i cannot manage anything...

thoompje, 2023-02-01:
Let it go over, i fixed it myself to without a new docker container. I made an docker cp script to update the util.py script which has the problem in it. It is a workaround but good enough.

New part in util.py

```
def get_systype():
    system = platform.system().lower()
    arch = platform.machine().lower()
    if system == "windows":
        if not arch:  # issue #4353
            arch = "x86_" + platform.architecture()[0]
        if "x86" in arch:
            arch = "amd64" if "64" in arch else "x86"
    if arch == "aarch64":
        if platform.architecture()[0] == "32bit":
            arch = "armv7l"
    return "%s_%s" % (system, arch) if arch else system
```

Old part in util.py

```
def get_systype():
    type_ = platform.system().lower()
    arch = platform.machine().lower()
    if type_ == "windows" and "x86" in arch:
        arch = "amd64" if "64" in arch else "x86"
    if arch == "aarch64":
        if platform.architecture()[0] == "32bit":
            arch = "armv7l"
    return "%s_%s" % (type_, arch) if arch else type_

```

Cheers and thanks.

Samu84, 2023-02-01:
good you get it fixed, i didn't have time at all today

thoompje, 2023-02-02:
Yes happy that it is fixed with the workaround!

lexcorp, 2023-03-02:
> 

Hi, where is locate this file?

lexcorp, 2023-03-03:
Aquí también hay otra referencia:

https://github.com/esphome/issues/issues/1386#issuecomment-1073117862

lexcorp, 2023-03-31:
Use this command to search files, from root of file system /

find | grep platformio/util.py

sheminasalam, 2023-04-01:
@lexcorp  that worked. thanks for help.

Florian-Schmidt, 2023-05-11:
Ok, this exactly the error message I get after I've made a OS and docker reboot. Tried to follow every hint but wasn't able to solve the issue.

Has anybody a final step by step description what solves this issue?

iZac85, 2023-05-12:
I also have the same issue after reinstall the EspHome add-on to get the lastest version (discovered that I had an old version since I first got an error saying I needed to update the PlatformIO core).

sheminasalam, 2023-05-12:
My issue was solved only after reinstalling 64 bit OS

Florian-Schmidt, 2023-05-12:
> My issue was solved only after reinstalling 64 bit OS

My system is a 64Bit OS. This is why I‘m surprised that it suddenly came up.

sheminasalam, 2023-05-12:
> > My issue was solved only after reinstalling 64 bit OS
> 
> My system is a 64Bit OS. This is why I‘m surprised that it suddenly came up.

what is your system and OS?

Florian-Schmidt, 2023-05-12:
> > > My issue was solved only after reinstalling 64 bit OS
> > 
> > 
> > My system is a 64Bit OS. This is why I‘m surprised that it suddenly came up.
> 
> what is your system and OS?

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

sheminasalam, 2023-05-12:
did you upgrade the 32 bit to 64 bit OS or did you do a full fresh install of 64 bit with this system?

Florian-Schmidt, 2023-05-12:
System was installed years ago and get‘s regular updates. Only rebooting the OS is done irregular. So there was no upgrade to 64bit OS.

I see the error since I updated ESPHome container and rebooted OS.

sheminasalam, 2023-05-12:
If you are using raspberry pi os, the official stable version of 64 bit was launched only in FEB last year. I also was running 32 bit for a long time and then upgraded it to 64 bit without fresh install. That was what caused the problem for me. Then i made a fresh install of 64 bit and problem got solved. Also one more thin i notices is that you are running an older firmware version in Rpi.

Florian-Schmidt, 2023-05-12:
Ok, strange as I installed ESPHome at the end of last year and compiled/installed several version on my esp8266.

So it actually seems as I need to wait and hope that further updates will solve this again.

iZac85, 2023-05-12:
> Let it go over, i fixed it myself to without a new docker container. I made an docker cp script to update the util.py script which has the problem in it. It is a workaround but good enough.
>
> New [...]

I tried these instructions and now I can compile again! Awesome work!

PaulUitWijchen, 2023-09-03:
Hi, I'm trying to understand this. I also have an aarch64 (raspberry pi4). And I also have this issue with xtensa ... not found. 
But I'm not sure what the problem is. 
Moreover, if I do a find in the root: 
$  cd /
$  sudo find | grep platformio/util.py

I find 5 occasions. And now I'm not sure which one to modify. 
Can someone explain ?

-- Paul

petriojk, 2023-09-16:
I had this same issue with ESPHome stand-alone installation (different computer than HA). Installation was made to Win10 that ran on VMWare virtual machine. I was able to solve the issue by uninstalling platformio (pip uninstall platformio) and then I tried to uninstall ESPHome (pip uninstall esphome) but that failed. Then I re-installed ESPhome and it started to work again. Not sure why this issue started, but I assume it was related either bad connection between the computer and device or then some yaml configuration issue?

mobilutz, 2023-10-11:
I am still having this problem.

I tried to use all of these docker images, but non of them make things work on my raspberry pi4
* https://hub.docker.com/r/esphome/esphome
* https://hub.docker.com/r/esphome/esphome-armv7
* https://hub.docker.com/r/esphome/esphome-aarch64

I did not add the workaround mentioned in this [comment](https://github.com/esphome/issues/issues/3092#issuecomment-1412223981), as my original platformio code is different than what is written there - version is different.

I will research this a bit more, and hope that something will be done here, I would like to switch from tasmota to esphome with using the dashboard to generate the firmware etc.

PaulUitWijchen, 2023-10-13:
Hi LutzDid you figure it out in the meantime? I presume you have a arch64 system. I’m also running on R pi4. My esphome units work fine now.— PaulSent from my iPhone

https://hub.docker.com/r/esphome/esphome
https://hub.docker.com/r/esphome/esphome-armv7
https://hub.docker.com/r/esphome/esphome-aarch64

ssieb, 2023-10-13:
> https://hub.docker.com/r/esphome/esphome

This is the only valid one.  Maybe you need to specify that you want the `linux/arm64` arch.

mobilutz, 2023-10-16:
@PaulUitWijchen @ssieb 
I unfortunately could not get esphome to compile the firmware correctly.

I will provide (hopefully) all needed information.

## What host platform am I using?
-> Raspberry Pi4 4GB:
```
> uname -a
Linux PiSmart 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

> docker --version
Docker version 24.0.6, build ed223bc

> docker compose version
Docker Compose version v2.21.0
```

## How is esphome running?
-> As docker container with the help of this docker compose file:
```yaml
version: '3'

services:
  esphome:
    container_name: esphome
    image: esphome/esphome
    volumes:
      - ./config:/config
      - /etc/localtime:/etc/localtime:ro
    restart: always
    privileged: true
    network_mode: host
```

```
> docker compose up -d
...
> docker compose exec esphome /bin/bash
root@PiSmart:/config# uname -a
Linux TruPiSmart 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

## What esphome config is used?
```yaml
esphome:
  name: esp8285
  friendly_name: ESP8285

esp8266:
  board: esp8285

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

### Firmware manual install output
```
INFO ESPHome 2023.9.3
INFO Reading configuration /config/test-esp8266-esp8285.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp8285 (board: esp8285; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/esp8285/src/esphome/components/esp8266/core.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/esp8285/src/esphome/components/esp8266/gpio.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/esp8285/src/esphome/components/esp8266/preferences.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/esp8285/src/esphome/components/esp8266/core.cpp.o] Error 127
Compiling .pioenvs/esp8285/src/esphome/components/mdns/mdns_component.cpp.o
*** [.pioenvs/esp8285/src/esphome/components/esp8266/gpio.cpp.o] Error 127
*** [.pioenvs/esp8285/src/esphome/components/esp8266/preferences.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/esp8285/src/esphome/components/mdns/mdns_component.cpp.o] Error 127
========================== [FAILED] Took 3.03 seconds ==========================
```

I don't know what else I could try to make things work directly through the dashboard 😦

PaulUitWijchen, 2023-11-03:
Hi Lutz, 
 
Sorry for the late reply. 
So I have the very same Pi4 with 4GB. 
***@***.***:~ $ uname -a
Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux
 
 
docker-compose version 1.29.2, build unknown
docker-py version: <module 'docker.version' from '/usr/local/lib/python3.9/dist-packages/docker/version.py'>
CPython version: 3.9.2
OpenSSL version: OpenSSL 1.1.1n 15 Mar 2022
 
esphome:
container_name: esphome
# image: esphome/esphome:stable
image: esphome/esphome:2022.10
# image: esphome/esphome:2023.3
# image: esphome/esphome:latest
# image: esphome/esphome:dev
volumes:
- /opt/esphome/config:/config
- /etc/localtime:/etc/localtime:ro
restart: always
privileged: true
network_mode: host
 
So indeed I have been struggling with getting a good version for Esphome. 
I have a hard time remembering what the exactly issue was. 
 
 
In the yaml code I have: 
 
esphome:
name: "kwh-monitor"

esp8266:
board: nodemcuv2

# Enable logging
logger:
 
 
I have the older version: 8266. But I'm not sure if that causes the issue. 
For the fun you could maybe use the nodemcuv2 option for the board. 
Do you have another board that does work? 
 
Sorry if I'm not able to help you further. 
 
 
-- Paul

[16 more replies]