esphome/issues #3093: Unable to compile for ESP32S3 core

lollokara, 2022-02-25:
### The problem

``Hi Everyone, 

I'm facing issues working on a custom board based on the ESP32-S3-WROOM-1 Module.

I'm able to program said board with arduino using the branch https://github.com/espressif/arduino-esp32/tree/esp32-s3-support. 

Since i've read that Esphome is adding support to it i've gone into the rabbithole to get it working.
Up until now i've tried every combination possible and so far the only one that produces at least an output is the following:
esp32:
  board: esp32-s3-devkitm-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: latest

With a custom esp32-s3 board file based on the tasmota one. 

Attatched is the output for the run command.
I'll need some help to figure this out since there is no support for it yet. 

Thanks a lot for everyones time.

### Which version of ESPHome has the issue?

2022.3.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: servoboard

esp32:
  board: esp32-s3-devkitm-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: latest
```

### Anything in the logs that might be useful for us?

```txt
`INFO Reading configuration ESP32Main.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing servoboard (board: esp32-s3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 512KB RAM, 8MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.9.0 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s3 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling .pioenvs/servoboard/src/esphome/components/api/api_connection.o
Compiling .pioenvs/servoboard/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/servoboard/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/servoboard/src/esphome/components/api/api_pb2_service.o
Compiling .pioenvs/servoboard/src/esphome/components/api/api_server.o
Compiling .pioenvs/servoboard/src/esphome/components/api/list_entities.o
Compiling .pioenvs/servoboard/src/esphome/components/api/proto.o
Compiling .pioenvs/servoboard/src/esphome/components/api/subscribe_state.o
Compiling .pioenvs/servoboard/src/esphome/components/api/user_services.o
Compiling .pioenvs/servoboard/src/esphome/components/api/util.o
Compiling .pioenvs/servoboard/src/esphome/components/esp32/gpio_arduino.o
Compiling .pioenvs/servoboard/src/esphome/components/esp32/gpio_idf.o
Compiling .pioenvs/servoboard/src/esphome/components/esp32/preferences.o
Compiling .pioenvs/servoboard/src/esphome/components/logger/logger.o
Compiling .pioenvs/servoboard/src/esphome/components/md5/md5.o
Compiling .pioenvs/servoboard/src/esphome/components/mdns/mdns_component.o
Compiling .pioenvs/servoboard/src/esphome/components/mdns/mdns_esp32_arduino.o
Compiling .pioenvs/servoboard/src/esphome/components/mdns/mdns_esp8266.o
Compiling .pioenvs/servoboard/src/esphome/components/mdns/mdns_esp_idf.o
Compiling .pioenvs/servoboard/src/esphome/components/network/util.o
Compiling .pioenvs/servoboard/src/esphome/components/ota/ota_backend_arduino_esp32.o
Compiling .pioenvs/servoboard/src/esphome/components/ota/ota_backend_arduino_esp8266.o
Compiling .pioenvs/servoboard/src/esphome/components/ota/ota_backend_esp_idf.o
Compiling .pioenvs/servoboard/src/esphome/components/ota/ota_component.o
Compiling .pioenvs/servoboard/src/esphome/components/socket/bsd_sockets_impl.o
Compiling .pioenvs/servoboard/src/esphome/components/socket/lwip_raw_tcp_impl.o
Compiling .pioenvs/servoboard/src/esphome/components/socket/socket.o
Compiling .pioenvs/servoboard/src/esphome/components/wifi/wifi_component.o
Compiling .pioenvs/servoboard/src/esphome/components/wifi/wifi_component_esp32_arduino.o
Compiling .pioenvs/servoboard/src/esphome/components/wifi/wifi_component_esp8266.o
Compiling .pioenvs/servoboard/src/esphome/components/wifi/wifi_component_esp_idf.o
Compiling .pioenvs/servoboard/src/esphome/core/application.o
Compiling .pioenvs/servoboard/src/esphome/core/component.o
Compiling .pioenvs/servoboard/src/esphome/core/controller.o
Compiling .pioenvs/servoboard/src/esphome/core/helpers.o
Compiling .pioenvs/servoboard/src/esphome/core/log.o
Compiling .pioenvs/servoboard/src/esphome/core/scheduler.o
Compiling .pioenvs/servoboard/src/esphome/core/util.o
Compiling .pioenvs/servoboard/src/main.o
Linking .pioenvs/servoboard/firmware.elf
RAM:   [=         ]   5.9% (used 30972 bytes from 524288 bytes)
Flash: [====      ]  35.7% (used 655401 bytes from 1835008 bytes)
Building .pioenvs/servoboard/firmware.bin
esp32_create_combined_bin([".pioenvs/servoboard/firmware.bin"], [".pioenvs/servoboard/firmware.elf"])
Unexpected chip id in image. Expected 9 but value was 6. Is this image for a different chip model?
Wrote 0xb01b0 bytes to file /Users/lorenzokaravania/Desktop/ESP/.esphome/build/servoboard/.pioenvs/servoboard/firmware-factory.bin, ready to flash to offset 0x0
======================================================================================================== [SUCCESS] Took 15.90 seconds ========================================================================================================
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] /dev/cu.usbserial-AJ02ZJ9A (FT232R USB UART - FT232R USB UART)
  [2] Over The Air (servoboard.local)
(number): 1
esptool.py v3.2
Serial port /dev/cu.usbserial-AJ02ZJ9A
Connecting.............................
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 60:55:f9:f5:29:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00010000 to 0x000b0fff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Compressed 655792 bytes to 431070...
Wrote 655792 bytes (431070 compressed) at 0x00010000 in 11.2 seconds (effective 466.9 kbit/s)...
Hash of data verified.
Unexpected chip id in image. Expected 9 but value was 6. Is this image for a different chip model?
Compressed 20400 bytes to 12497...
Wrote 20400 bytes (12497 compressed) at 0x00001000 in 0.6 seconds (effective 255.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 305.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 459.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-AJ02ZJ9A with baud rate 115200
[14:57:49]ESP-ROM:esp32s3-20210327
[14:57:49]Build:Mar 27 2021
[14:57:49]rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
[14:57:49]SPIWP:0xee
[14:57:49]mode:DIO, clock div:1
[14:57:49]load:0x3fcd0108,len:0x43c
[14:57:49]load:0x403b6000,len:0xbd0
[14:57:49]load:0x6001f000,len:0x0
[14:57:49]Checksum failure. Calculated 0x16 stored 0x00
[14:57:49]ets_main.c 329 
[14:57:50]ESP-ROM:esp32s3-20210327
[14:57:50]Build:Mar 27 2021
[14:57:50]rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[14:57:50]Saved PC:0x40043ac8
[14:57:51]SPIWP:0xee
[14:57:51]mode:DIO, clock div:1
[14:57:51]load:0x3fcd0108,len:0x43c
[14:57:51]load:0x403b6000,len:0xbd0
[14:57:51]load:0x6001f000,len:0x0
[14:57:51]Checksum failure. Calculated 0x16 stored 0x00
[14:57:51]ets_main.c 329 
[14:57:52]ESP-ROM:esp32s3-20210327`
```

lollokara, 2022-03-29:
Not being able still to compile for that core. Is there any resolution for it?

kj4dxk, 2022-08-26:
I am having the same issue.  Is there a way to run ESPhome on the ESP32s3?

gmarcc, 2022-08-26:
I'm having the same problem.

kj4dxk, 2022-08-26:
gmarcc: what board do you have?

gmarcc, 2022-08-28:
It was a ESP32-S3-DevKitC-1-N8R2

kj4dxk, 2022-08-29:
Use ESPhome-Dev.  It is a different Add-On in HA.

Also use this:

esp32:
  board: adafruit_qtpy_esp32s3_nopsram  #5.0.1
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1

kj4dxk, 2022-08-29:
or substitute your board.  But neopixel still won't work for me.

Zettaga, 2022-09-17:
Also suffering from this issue, using Adafruit ESP32-S3 Feather with STEMMA QT / Qwiic - 8MB Flash No PSRAM

ssieb, 2022-09-17:
@Zettaga, what issue?  The title here isn't even correct.  The only compile log shows that it compiled successfully.  Several people have successfully compiled and installed for both S3 and C3 boards.  It would be better if you join the esphome Discord server for more interactive help.

pszober, 2022-10-11:
Hi, I'm in the same boat. I recently purchased 15pcs of ESP32-S3-WROOM-1 boards with an intention to flash them all with esphome and now I'm stuck with them being useless. 

I cannot even find any info on the net that would tell me if there is an active development to include these chips into the esphome project. I'm very disappointed ;(

kj4dxk, 2022-10-11:
This is working for me right now.  Does it work for you?

esp32:
  board: adafruit_qtpy_esp32s3_nopsram  #5.0.1
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1

pszober, 2022-10-12:
Thanks kj4dxk. I tried your config but with same results. I must be doing something wrong.

Here's what I did:
1. I installed ESPHome (dev) in HA (without uninstalling standard ESPHome add-on but it seems to be working fine).
2. Created a new ESPhome device and replaced the "esp32:" section with your code
3. Tried to install using "Plug into this computer" option but got a reply saying "Expected an ESP32S3 device."
4. Tried to compile the code and download .bin file in both formats - legacy and modern.
5. Used ESPHome-Flasher-1.4.0-Windows-x64 to flash the bin file onto the device and it seems to flash successfully regardless of which bin file I use - modern or legacy work fine but when I restart the ESP into normal mode, I get "Invalid header 0xffffffff" loop.

Any ideas?

pszober, 2022-10-12:
UPDATE:

I managed to flash it using this code (and only after plugging in directly to RPi running HA):
```
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.1.0
  variant: esp32s3
```
Unfortunately, it won't boot, it's in a continuous boot loop.

pszober, 2022-10-15:
UPDATE 2:
I have to admit to a little bit of laziness and trying to take shortcuts. Turns out that, for some reason, I can only flash the ESP32S3 board by having it connected directly to RPi running HA, otherwise it will give the compatibility error (which I already knew the last time). The thing is, that when I unplugged the chip from my PC and connected it to the RPi, I did not plug in the external 3.3v power supply as I was assuming that the CH340G I'm using, will give sufficient power, as it was fine flashing all sorts of other ESP chips. Well, it wasn't giving sufficient power. 

Now I have it flashed successfully using the following config:
```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.1.0
  variant: esp32s3
```

This leads to a new problem :) 

I can only flash it as I described above but OTA updates won't work. I am nearly certain it's because there is less memory on the chip than esphome expects. When flashing, the terminal window displays the following line:

`HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash`

My chip has only 4MB of flash memory and I believe, that is why subsequent OTA updates fail. How do I fix that?

kj4dxk, 2022-10-15:
Mine does OTA well.

Paul

pszober, 2022-10-15:
Can I ask you if your ESP is a 4 or 8MB version?

ark02, 2022-10-29:
Having similar issue on both DevkitC-1 and DevkitM-1

my baords.json file: 

```
{
    "build": {
        "arduino": {
            "ldscript": "esp32s3_out.ld",
            "partitions": "max_app_8MB.csv"
        },
        "core": "esp32",
        "extra_flags": [
            "-DARDUINO_ESP32S3_DEV",
            "-DARDUINO_USB_MODE=1",
            "-DARDUINO_RUNNING_CORE=1",
            "-DARDUINO_EVENT_RUNNING_CORE=1",
            "-DBOARD_HAS_PSRAM"
        ],
        "f_cpu": "240000000L",
        "f_flash": "80000000L",
        "flash_mode": "dio",
        "hwids": [
            [
                "0x303A",
                "0x1001"
            ]
        ],
        "mcu": "esp32s3",
        "variant": "esp32s3"
    },
    "connectivity": [
        "wifi"
    ],
    "debug": {
        "default_tool": "esp-builtin",
        "onboard_tools": [
            "esp-builtin"
        ],
        "openocd_target": "esp32s3.cfg"
    },
    "frameworks": [
        "arduino",
        "espidf"
    ],
    "name": "Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, 2MB PSRAM)",
    "upload": {
        "flash_size": "8MB",
        "maximum_ram_size": 327680,
        "maximum_size": 8323072,
        "require_upload_port": true,
        "speed": 460800
    },
    "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html",
    "vendor": "Espressif"
}
```

pszober, 2022-10-30:
What is baords.json file and where is it located? Can I edit that file to modify the amount of flash ROM available on the board so that it's matching my actual esp32 board?

cvjensen, 2022-11-02:
Hi guys , I also have a ESP32S3 which I'm having aloooot of troubles with.. 

Its a ESP32-S3-WROOM-1-N16R8

Have any of you experienced these errors ? 
I'm not even talking about uploading to the device , just building the firmware ready for download!

```
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32s3-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32s3-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32s3-elf-g++: not found
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-esp32s3-elf-g++: not found
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================= [FAILED] Took 166.50 seconds =========================
```

I have also been trying the config @pszober posted:

```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 5.1.0
  variant: esp32s3
```

kj4dxk, 2022-11-02:
Install the dev esphome in ha.

Paul

haplm, 2022-11-02:
try switching to esp-idf framework:

`type: esp-idf`

cvjensen, 2022-11-03:
> try switching to esp-idf framework:
> 
> `type: esp-idf`

I just tried with IDF with the following config on the DEV version:

```
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    version: latest
    platform_version: 5.2.0
  variant: esp32s3
```

But I received this error:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
esphome 2022.11.0.dev0 requires colorama==0.4.5, but you have colorama 0.4.6 which is incompatible.
Successfully installed certifi-2022.9.24 charset-normalizer-2.1.1 click-8.1.3 colorama-0.4.6 contextlib2-21.6.0 future-0.18.2 idf-component-manager-1.1.4 idna-3.4 pyparsing-2.3.1 pyyaml-6.0 requests-2.28.1 requests-toolbelt-0.10.1 schema-0.7.5 six-1.16.0 tqdm-4.64.1 urllib3-1.26.12
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/data/test/.pioenvs/test/CMakeFiles/CMakeOutput.log".
See also "/data/test/.pioenvs/test/CMakeFiles/CMakeError.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /data/test/.pioenvs/test/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/data/cache/platformio/packages/tool-ninja/ninja cmTC_8a4a2 && [1/2] Building C object CMakeFiles/cmTC_8a4a2.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_8a4a2.dir/testCCompiler.c.obj 
    /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc   -mlongcalls -o CMakeFiles/cmTC_8a4a2.dir/testCCompiler.c.obj   -c testCCompiler.c
    /bin/sh: 1: /data/cache/platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gcc: not found
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:311 (__project)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 176.74 seconds =========================
```

ssieb, 2022-11-03:
It's not.  It's most likely the wrong type of binary.  Are you running a 32-bit OS?

cvjensen, 2022-11-03:
> It's not. It's most likely the wrong type of binary. Are you running a 32-bit OS?

Guess its gonna be a 'yes' on that question...

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```

haplm, 2022-11-03:
There was a problem that some idf-esp libraries were not available for arm. Therefore I'm compiling it on a PC.

cvjensen, 2022-11-03:
> There was a problem that some idf-esp libraries were not available for arm. Therefore I'm compiling it on a PC.

Do you know if its anything thats in the pipeline and is going to get fixed?

BoxedBrain, 2022-11-03:
It was fixed but looks like it is not released yet: https://github.com/espressif/esp-idf/issues/6432

cvjensen, 2022-11-03:
> It was fixed but looks like it is not released yet: [espressif/esp-idf#6432](https://github.com/espressif/esp-idf/issues/6432)

Thanks for the link. Any idea when it will be released ?

BoxedBrain, 2023-01-13:
As the esp-idf was released recently, I was able to flash my S2 mini without a problem now 🥳
I used a config like this:
```
# [...]
esp32:
  board: lolin_s2_mini
  framework:
    type: arduino
# [...]
```

@lollokara can you verify and close/update the issue?

closed 2023-02-22 by lollokara