esphome/issues #3094: GPIO switch toggles on boot (ESP32, various pins)

stefesciberras, 2022-02-25:
### The problem

Setup: ESP32 NodeMCU, with ESPHome 2022.2.4 installed. GPIO 33 connected to Wemos D1 relay shield. Active High, attached to NO. This is to open a door on HA command.

The same setup on a ESP8266 worked perfectly. I had to expand this project to more sensors, including bluetooth, so opted for ESP32.

On ESP32, at boot or reset, the GPIO pin is going high, then low. This activates the relay, so opens the door (this becomes a security issue)

CODE:
```
# Relay pin
switch:
  - platform: output
    id: doorrelay
    #pin: GPIO32
    output: doorPIN
    name: "Door relay"
    restore_mode: ALWAYS_OFF
    icon: "mdi:gate"
    on_turn_on:
    #- output.turn_on: doorPIN
    - output.turn_on: doorrelayLED
    - delay: 500ms
    #- output.turn_off: doorPIN
    - switch.turn_off: doorrelay
    - output.turn_off: doorrelayLED
    - homeassistant.service:
        service: notify.mobile_app_stefe_iphone
        data:
          title: "Button was pressed"
          message: "opening door"

# Relay pin
output:
  - platform: gpio
    pin: GPIO02
    id: doorrelayLED
    inverted: true
    
  - platform: gpio
    pin: GPIO33
    id: doorPIN
```

I have tried the following, with no success

1) Initial code used switch platform GPIO, set on pin 32. I tried changing this to an output switch.
2) Tried GPIO 33 as well
3) Placed a pulldown resistor 4k7 between relay pin and ground.
4) changed ALWAYS_OFF to restore_DEFAULT_OFF
5) tried inverting pins to get an inverse logic

The only thing that worked was setting a different GPIO in thee code, but keeping the physical connection. That is, soldering input to GPIO 33, but using GPIO 32 in code. At that point, nothing happens on boot, but off course, this is useless, as I cannot activate the relay any more.

It seems that at boot, GPIO outputs that are set up in ESPhome toggle on & off. 

Help welcome, as now I have to take everything down again.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.2.9, even core-2022.2.8

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

GPIO switch, output

OttoWinter, 2022-03-09:
Sometimes that is due to the ESP32's internal pullup resistors (you'll have to consult the technical reference manual for that), and sometimes due to the board having external pullups on some pins. Additionally, some special pins are used by the ESP32's internal bootloader (before esphome even starts), and we can't control that because it happens even before esphome code starts.

I'd advise looking at the technical reference manual and checking with a multimeter/oscilloscope if there are pullups on the pins (and check if there are others without them)

Parschulke, 2022-03-28:
I habe the same issue. This was working with the previous version

swbouman, 2022-06-10:
For me the same. It worked like a charm till 2022.1. After that, at boot the relays chatter at boot. Had anyone made changes in the boot loader or something like that? 

Hope the issue can be fixed because it is annoying, my doorbell fires at reboot.

jshep321, 2022-07-14:
Hey,  I've successfully avoided this issue with my esp32 dev boards, but just received some super cheap ESP01's w/8266 chip.  They reuse GPIO0 & 2 from the boot process as the only outputs to my relays.  And they chatter at reset/cold boot.

Try moving your output GPIO2 to a different pin.  [https://github.com/esphome/issues/issues/640](url)

My next attempt will be to play around with stronger (external) pull-ups, (which would exclude uart flashing), but I think in the case of ESP01 - that's just life.

I'm curious if both your input button and output are  glitching or just the output,...

chaffinched, 2022-07-27:
I'm having the same problem on a Wemos D1 mini. It doesn't happen if I just set up the relays using the Arduino IDE, so it doesn't seem to be a device issue, rather an ESPHome issue.

chaffinched, 2022-08-01:
Adding the last line below fixed my ESP8266 issue.
esp8266:
  board: d1_mini
  early_pin_init: false

jshep321, 2022-08-06:
Thanks @chaffinched.  On the ESP01, early_pin_init: false reduced the chatter to a single event instead of two.  But that's still one too many.  I guess I'll move on from the ESP01 to the D1 or other.

chaffinched, 2022-08-06:
@jshep321 I swear I had this working on all GPIO pins on my D1 mini, but since the last ESPHome update it only works for some pins. You might want to try other pins before changing boards.

jshep321, 2022-08-06:
Thanks.  There are only 2 on the ESP01.  GPIO0&2, which are both boot related.

michaellsailo, 2022-11-15:
I got the same problem of relay firing on boot. Waiting for solution

FiTekno, 2023-09-02:
I'm having same problem.

nodemcu 32s
im using gpio25 and gpio26 for double 5v relay (2x normally open contacts). they are activated for a moment during power on.

Kappa71, 2023-09-09:
I have the same problem on a ESP32-C3 on (at least) GPIO6 and GPIO7 on ESPHOME 2023.8.3 
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

cormiersa, 2024-10-03:
So this may get a little long: I have three boards FREENOVE esp32, LoLin ModeMCU V3, ESP12F_RELAY_X2_V1.2
Have tried all three with esphome connecting them to different GPIOs presently using recommended GPIO5,4
using them to remotely switch relays.
have tried "early_pin_init: false" and "restore_mode: ALWAYS_OFF" in my code.
they are set as pin not inverting with the relay connected to normally open.
all three of them activate the relay on boot.
After reading many posts on this issue I needed to decide if this is actually a hardware issue or a software issue. So I loaded a sketch in Arduino and up loaded it to the ESP12F_RELAY_X2_V1.2. it sets up a web site and activates the two relays.
The Arduino sketch works perfectly with no power up on restart. So this indicates a programming issue in the ESPHome software.
It also means the problem exists over the ESP32 and ESP8266 code. Can someone help fix this?

clydebarrow, 2024-10-04:
Does it happen on a hardware reset or only on software reset (e.g. after OTA?)

swbouman, 2024-10-04:
Thuis happens on hardware reset. Software reset it happens sometimes.

cormiersa, 2024-10-04:
Both. If I remove power.and when I send an OTA update at the end when it
resets.

cormiersa, 2024-10-11:
So I have solved my issues: 
When you have a device that has a "on_turn_off" command that controls a relay and
you have that device default to off :restore_mode: RESTORE_DEFAULT_OFF"
the following happens:
when there is a power reset or over the air update the device looses state.
on boot the device sets the state for example to off. this triggers the on_turn_off routine. 
In my case this routine triggered the relay.
Solution:
us an if statement in the on_turn_off routine that blocks action if say something is true.
then use;
esphome:
   on_boot:
       priority: 500
    then:
        - delay: 10s
        - set your blocking action to false so the on_turn_off now works normally

hope this helps

dannybloe, 2024-10-11:
It still is a workaround. Would be better if it was solved somehow.

funnyfrish, 2024-11-22:
@cormiersa 
Can you send your yaml with this blocking action variable?

cormiersa, 2024-11-23:
[tailwind.txt](https://github.com/user-attachments/files/17880562/tailwind.txt)
Here is my yaml file

dgtal1, 2025-05-04:
I've got the same issue. Wemos D1 mini on ESP8266. 4 GPIOs are set as outputs controlling transistors and these in turn control relays. Relays are turned on for a short while on boot. 
I've tried the following:
1. early_pin_init: false
2. using restore mode ALWAYS_OFF
3. setting state to OFF on boot priority 800

Nothing helps :( Has anyone found something else? This is not only frustrating, but also unacceptable as these relays control other devices, which are controlled by momentary impulses and they are triggered on Wemos D1 boot :(

ssieb, 2025-05-04:
Depending on the pins, you might not be able to avoid that, at least on an 8266.

cormiersa, 2025-10-29:
> ### The problem
> Setup: ESP32 NodeMCU, with ESPHome 2022.2.4 installed. GPIO 33 connected to Wemos D1 relay shield. Active High, attached to NO. This is to open a door on HA command.
>
> The same setup [...]

So there is no issue. The device is working logically. 
You told it to default to Always off. So when it power resets it go to off. This triggers the on turn off, which triggers the relay!!