esphome/issues #3095: CWWW light not restoring state on power on

Brazen00, 2022-02-26:
### The problem

I have a bunch of CWWW light bulbs which I updated from 1.18.x to 2022.2.4 (I know...). Some of them are connected to dumb switches, and previously the colour temperature and brightness would be restored when the dumb switch was powered off and on again. Now, the state reverts to a default which is different from the previous state. Home Assistant (once it connects to the API again) is still reporting the previous state, but the bulb colour temperature is different.

I tried adding ```esp8266_restore_from_flash: true``` and ```flash_write_interval: 0s``` but it still doesn't restore the state.

### Which version of ESPHome has the issue?

2022.2.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP8266

### Board

esp01_1m

### Example YAML snippet

```yaml
esphome:
  name: $deviceName
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: true

preferences:
  flash_write_interval: 0s

output:
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO13
  - platform: esp8266_pwm
    id: output_daylight
    pin: GPIO5

light:
  - platform: cwww
    name: $friendlyName
    cold_white: output_daylight
    warm_white: output_warm_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    
    # Ensure the light turns on by default if the physical switch is actuated.
    restore_mode: ALWAYS_ON
```

OttoWinter, 2022-03-09:
You have `restore_mode: ALWAYS_ON` in your config, which makes ESPHome _not_ restore the state, and make it always turn the light on when esphome starts.

Brazen00, 2022-03-09:
Oh, I thought ```restore_mode``` only referred to on/off, not the brightness/temperature.

Brazen00, 2022-03-09:
Thanks @OttoWinter, this fixed it.

closed 2022-03-09 by Brazen00