esphome/issues #3100: ModuleNotFoundError: No module named 'SCons.Script'

ebdavison, 2022-02-28:
### The problem

Trying to compile a config on esphome from the Home Assistant UI results in the following error:

```
INFO Reading configuration /config/esphome/master_led1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing masterled1 (board: nodemcu; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
SCons import failed. Unable to find engine files in:
  /data/cache/platformio/packages/tool-scons/..
  /data/cache/platformio/packages/tool-scons/scons-local-4.3.0
Traceback (most recent call last):
  File "/data/cache/platformio/packages/tool-scons/scons.py", line 89, in <module>
    import SCons.Script
ModuleNotFoundError: No module named 'SCons.Script'
========================== [FAILED] Took 0.55 seconds ==========================
```

### Which version of ESPHome has the issue?

2022.2.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.7

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

unknown

### Example YAML snippet

```yaml
esphome:
  name: masterled1
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxx"
  manual_ip:
    static_ip: 10.1.1.2
    gateway: 10.1.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Master LED1 Fallback Hotspot"
    password: "xxxxxxxx"

captive_portal:

# Enable logging
logger:

#web_server:
#  port: 80

# Enable Home Assistant API
api:
  password: "xxxxxxxx"

ota:
  password: "xxxxxxxx"

light:
  - platform: neopixelbus
    variant: WS2812
    pin: D2
    num_leds: 10
    type: GRB
    name: "Master LED"
    id: masterled1
    effects:
      - addressable_color_wipe:
      - addressable_color_wipe:
          name: Color Wipe Effect Red
          colors:
            - red: 100%
              green: 0%
              blue: 0%
              num_leds: 1
            - red: 0%
              green: 0%
              blue: 0%
              num_leds: 1
          add_led_interval: 250ms
          reverse: False
      - flicker:
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 1.5%
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/master_led1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing masterled1 (board: nodemcu; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
SCons import failed. Unable to find engine files in:
  /data/cache/platformio/packages/tool-scons/..
  /data/cache/platformio/packages/tool-scons/scons-local-4.3.0
Traceback (most recent call last):
  File "/data/cache/platformio/packages/tool-scons/scons.py", line 89, in <module>
    import SCons.Script
ModuleNotFoundError: No module named 'SCons.Script'
========================== [FAILED] Took 0.55 seconds ==========================
```

OttoWinter, 2022-03-09:
Ok that seems like an error in platformio (/the data in the platformio cache got corrupted).

Could you delete the esphome addon's files in the HA interface and re-add it (the config files are stored in HA's `/config` folder, so deleting the addon retains the config files, but cached files are removed)

ebdavison, 2022-03-09:
Thank you very much for the guidance.  The config now complies properly.

closed 2022-03-09 by ebdavison