esphome/issues #3108: Home Assistant API keeps closing connection

MrAJB, 2022-03-02:
### The problem

The Home Assistant API keeps closing the connection, see the log output. Also, the integration is not detected in Home Assistant. When manually adding the integration, all entities are unavailable.

What I tried so far:
- Setting WiFi channel width to auto, 20MHz or 40MHz (based on another issue I found)> no effect
- Specifying an API password (or not) > no effect
- Making a very, very barebone new device (see YAML snippet) > no effect
- Defining the eps32 component like below > no effect

esphome:
  name: anothertest

esp32:
  board: esp32dev
  framework:
    type: arduino

### Which version of ESPHome has the issue?

2022.2.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP32

### Board

DEVKIT V1

### Component causing the issue

N/A

### Example YAML snippet

```yaml
esphome:
  name: esp32-test3
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid1
  password: !secret wifi_pass1
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Test3 Fallback Hotspot"
    password: !secret fallback_wifi_pass

captive_portal:

sensor:
  - platform: uptime
    name: "ESP32-Test Uptime Sensor"
    update_interval: 20s
  - platform: wifi_signal
    name: "ESP32-Test WiFi Signal Sensor"
    update_interval: 20s
```

### Anything in the logs that might be useful for us?

```txt
[19:28:06][I][app:102]: ESPHome version 2022.2.6 compiled on Mar  2 2022, 16:13:46
[19:28:06][C][wifi:491]: WiFi:
[19:28:06][C][wifi:353]:   Local MAC: 24:0A:C4:31:C2:9C
[19:28:06][C][wifi:354]:   SSID: [redacted]
[19:28:06][C][wifi:355]:   IP Address: 192.168.178.23
[19:28:06][C][wifi:357]:   BSSID: [redacted]
[19:28:06][C][wifi:358]:   Hostname: 'esp32-test3'
[19:28:06][C][wifi:360]:   Signal strength: -46 dB ▂▄▆█
[19:28:06][C][wifi:364]:   Channel: 7
[19:28:06][C][wifi:365]:   Subnet: 255.255.255.0
[19:28:06][C][wifi:366]:   Gateway: 192.168.178.1
[19:28:06][C][wifi:367]:   DNS1: 192.168.178.1
[19:28:06][C][wifi:368]:   DNS2: 0.0.0.0
[19:28:07][C][logger:233]: Logger:
[19:28:07][C][logger:234]:   Level: DEBUG
[19:28:07][C][logger:235]:   Log Baud Rate: 115200
[19:28:07][C][logger:236]:   Hardware UART: UART0
[19:28:07][C][uptime.sensor:031]: Uptime Sensor 'ESP32-Test Uptime Sensor'
[19:28:07][C][uptime.sensor:031]:   State Class: 'total_increasing'
[19:28:07][C][uptime.sensor:031]:   Unit of Measurement: 's'
[19:28:07][C][uptime.sensor:031]:   Accuracy Decimals: 0
[19:28:07][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[19:28:07][C][captive_portal:144]: Captive Portal:
[19:28:07][C][mdns:084]: mDNS:
[19:28:07][C][mdns:085]:   Hostname: esp32-test3
[19:28:07][C][ota:085]: Over-The-Air Updates:
[19:28:07][C][ota:086]:   Address: esp32-test3.local:3232
[19:28:07][C][ota:089]:   Using Password.
[19:28:07][C][api:138]: API Server:
[19:28:07][C][api:139]:   Address: esp32-test3.local:6053
[19:28:07][C][api:143]:   Using noise encryption: NO
[19:28:07][C][wifi_signal.sensor:009]: WiFi Signal 'ESP32-Test WiFi Signal Sensor'
[19:28:07][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[19:28:07][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[19:28:07][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[19:28:07][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[19:28:10][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 12.65100 s with 0 decimals of accuracy
[19:28:21][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -48.00000 dBm with 0 decimals of accuracy
[19:28:30][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 32.64700 s with 0 decimals of accuracy
[19:28:36][D][api:102]: Accepted ::FFFF:C0A8:B2A0
[19:28:36][W][api.connection:083]: Home Assistant 2022.2.9 (::FFFF:C0A8:B2A0): Connection closed
[19:28:41][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -49.00000 dBm with 0 decimals of accuracy
[19:28:50][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 52.64700 s with 0 decimals of accuracy
[19:29:01][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:29:10][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 72.64700 s with 0 decimals of accuracy
[19:29:21][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:29:30][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 92.64800 s with 0 decimals of accuracy
[19:29:36][D][api:102]: Accepted ::FFFF:C0A8:B2A0
[19:29:36][W][api.connection:083]: Home Assistant 2022.2.9 (::FFFF:C0A8:B2A0): Connection closed
[19:29:41][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:29:50][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 112.64700 s with 0 decimals of accuracy
[19:30:01][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:30:10][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 132.64700 s with 0 decimals of accuracy
[19:30:21][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:30:30][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 152.64999 s with 0 decimals of accuracy
[19:30:36][D][api:102]: Accepted ::FFFF:C0A8:B2A0
[19:30:36][W][api.connection:083]: Home Assistant 2022.2.9 (::FFFF:C0A8:B2A0): Connection closed
[19:30:41][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:30:50][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 172.64700 s with 0 decimals of accuracy
[19:31:01][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -47.00000 dBm with 0 decimals of accuracy
[19:31:10][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 192.64700 s with 0 decimals of accuracy
[19:31:21][D][sensor:125]: 'ESP32-Test WiFi Signal Sensor': Sending state -46.00000 dBm with 0 decimals of accuracy
[19:31:30][D][sensor:125]: 'ESP32-Test Uptime Sensor': Sending state 212.64999 s with 0 decimals of accuracy
[19:31:36][D][api:102]: Accepted ::FFFF:C0A8:B2A0
[19:31:36][W][api.connection:083]: Home Assistant 2022.2.9 (::FFFF:C0A8:B2A0): Connection closed
```

Joppewouts, 2022-03-04:
Just an observation and no solution. 
Weird how the log events
```Home Assistant 2022.2.9 (::FFFF:C0A8:B2A0): Connection closed```
consistently occur 1 min apart from each other

MrAJB, 2022-03-04:
Good observation. I am not aware of any disconnection timer anywhere, but I'm also far from sure what is happening under the hood in HA and ESPhome, so there might very well be something like that involved.

pieterij, 2022-03-04:
I observe exactly the same behaviour. Every minute a disconnect. I am running the same version of ESPhome and 2022.3 version of HA

Joppewouts, 2022-03-05:
Could be useful to increase the log level to 

```VERY_VERBOSE```

In your yaml that would be
```diff
# Enable logging
logger:
+  level: VERY_VERBOSE
```

pieterij, 2022-03-06:
After adding verbose logging here is a part of the log. One thing that 
looks a bit strange is the esp_ble_tracker which seems to take a bit too much time.
The other thing is obviously the disconnect from ESPhome API

[12:04:43][V][component:199]: Component esp32_ble_tracker took a long time for an operation (0.07 s).
[12:04:43][V][component:200]: Components should block for at most 20-30ms.
[12:04:43][VV][api.service:355]: on_ping_request: PingRequest {}
[12:04:43][VV][api.service:043]: send_ping_response: PingResponse {}
[12:04:43][VV][esp32_ble_tracker:497]: Parse Result:
[12:04:43][VV][esp32_ble_tracker:514]:   Address: 4E:E4:EB:DE:16:D6 (RANDOM)
[12:04:43][VV][esp32_ble_tracker:516]:   RSSI: -96
[12:04:43][VV][esp32_ble_tracker:517]:   Name: ''
[12:04:43][VV][esp32_ble_tracker:525]:   Ad Flag: 6
[12:04:43][VV][esp32_ble_tracker:547]: Adv data: 02.01.06
[12:04:43][VV][xiaomi_lywsdcgq:020]: parse_device(): unknown MAC address.
[12:04:43][VV][atc_mithermometer:021]: parse_device(): unknown MAC address.
[12:04:43][VV][atc_mithermometer:021]: parse_device(): unknown MAC address.
INFO 192.168.1.123: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for 192.168.1.123
WARNING Disconnected from API
INFO 192.168.1.123: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to 192.168.1.123
[12:04:58][VV][esp32_ble_tracker:497]: Parse Result:
[12:04:58][VV][esp32_ble_tracker:514]:   Address: 10:91:33:FC:8E:C6 (RANDOM)

MrAJB, 2022-03-08:
Good suggestion Joppe, implemented the ``VERY_VERBOSE`` logging. I'm basically running a very barebone ESP to make sure there is nothing else causing this behavior, the only other elements are an uptime sensor and WiFi signal strength. They seem to be working fine (apart from the fact that they are not visible as entities in HA of course due to this API issue).

```
[10:34:54][D][api:102]: Accepted ::FFFF:C0A8:B2A0
[10:34:54][VV][api.service:319]: on_hello_request: HelloRequest {
  client_info: 'Home Assistant 2022.3.1'
}
[10:34:54][V][api.connection:808]: Hello from client: 'Home Assistant 2022.3.1 (::FFFF:C0A8:B2A0)'
[10:34:54][VV][api.service:013]: send_hello_response: HelloResponse {
  api_version_major: 1
  api_version_minor: 6
  server_info: 'esp32-test3 (esphome v2022.2.6)'
  name: 'esp32-test3'
}
[10:34:54][VV][api.socket:810]: Home Assistant 2022.3.1 (::FFFF:C0A8:B2A0): Connection closed
[10:34:54][W][api.connection:083]: Home Assistant 2022.3.1 (::FFFF:C0A8:B2A0): Connection closed
[10:34:54][V][api:114]: Removing connection to Home Assistant 2022.3.1 (::FFFF:C0A8:B2A0)
```

han9988, 2022-03-15:
I had the same issue as described above. Upgraded today to version x in hope it gets better, but it didn't.
What's new is that now in the ESPHome Dashboard the status of the device changes from online to offline every 30s (about).

My current YAML-Script:
esp32_ble_tracker:

xiaomi_ble:

esphome:
  name: esp32ble-2022-1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
#  level: INFO
#  level: VERY_VERBOSE
#  logs:
#    mqtt.component: DEBUG
#    mqtt.client: ERROR

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxx"
    password: "xxx"
    
  domain: .mydomain.local

captive_portal:

sensor:
  - platform: xiaomi_lywsd03mmc
    mac_address: "xxx"
    bindkey: "xxx"
    temperature:
      name: "LYWSD03MMC Temperature"
    humidity:
      name: "LYWSD03MMC Humidity"
    battery_level:
      name: "LYWSD03MMC Battery Level"

  - platform: xiaomi_lywsd03mmc
    mac_address: "xxx"
    bindkey: "xxx"
    temperature:
      name: "LYWSD03MMC2 Temperature"
    humidity:
      name: "LYWSD03MMC2 Humidity"
    battery_level:
      name: "LYWSD03MMC2 Battery Level"

  - platform: xiaomi_mhoc401
    mac_address: "xxx"
    bindkey: "xxx"
    temperature:
      name: "MHOC401 Temperature"
    humidity:
      name: "MHOC401 Humidity"
    battery_level:
      name: "MHOC401 Battery Level"

mqtt:
  broker: '192.168.0.131'
  username: !secret mqtt_username
  password: !secret mqtt_password
  discovery: false
  discovery_retain: false
  birth_message:
  will_message:

davidsat, 2022-03-17:
Hi, 
I have the same issue (disconnect every minute, devices not discovered) I believe this started about three weeks ago. Please help!

[16:03:42][I][app:102]: ESPHome version 2022.4.0-dev compiled on Mar 17 2022, 15:52:28
[16:03:42][C][wifi:491]: WiFi:
[16:03:42][C][wifi:353]:   Local MAC: 3C:61:05:D0:A3:D6
[16:03:42][C][wifi:354]:   SSID: [redacted]
[16:03:42][C][wifi:355]:   IP Address: 192.168.1.73
[16:03:42][C][wifi:356]:   BSSID: [redacted]
[16:03:42][C][wifi:358]:   Hostname: 'leslie-controller'
[16:03:42][C][wifi:360]:   Signal strength: -63 dB ▂▄▆█
[16:03:42][C][wifi:364]:   Channel: 6
[16:03:42][C][wifi:365]:   Subnet: 255.255.255.0
[16:03:42][C][wifi:366]:   Gateway: 192.168.1.1
[16:03:42][C][wifi:367]:   DNS1: 192.168.1.1
[16:03:42][C][wifi:368]:   DNS2: 0.0.0.0
[16:03:42][C][logger:233]: Logger:
[16:03:42][C][logger:234]:   Level: DEBUG
[16:03:42][C][logger:235]:   Log Baud Rate: 115200
[16:03:42][C][logger:236]:   Hardware UART: UART0
[16:03:42][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Leslie D1'
[16:03:43][C][gpio.binary_sensor:016]:   Pin: GPIO5
[16:03:43][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Leslie D2'
[16:03:43][C][gpio.binary_sensor:016]:   Pin: GPIO4
[16:03:43][C][pulse_counter:148]: Pulse Counter 'Leslie Horn Pulse Counter'
[16:03:43][C][pulse_counter:148]:   State Class: 'measurement'
[16:03:43][C][pulse_counter:148]:   Unit of Measurement: 'pulses/min'
[16:03:43][C][pulse_counter:148]:   Accuracy Decimals: 2
[16:03:43][C][pulse_counter:148]:   Icon: 'mdi:pulse'
[16:03:43][C][pulse_counter:149]:   Pin: GPIO5
[16:03:43][C][pulse_counter:150]:   Rising Edge: INCREMENT
[16:03:43][C][pulse_counter:151]:   Falling Edge: DISABLE
[16:03:43][C][pulse_counter:152]:   Filtering pulses shorter than 13 µs
[16:03:43][C][pulse_counter:153]:   Update Interval: 5.0s
[16:03:43][C][pulse_counter:148]: Pulse Counter 'Leslie Woofer Pulse Counter'
[16:03:43][C][pulse_counter:148]:   State Class: 'measurement'
[16:03:43][C][pulse_counter:148]:   Unit of Measurement: 'pulses/min'
[16:03:43][C][pulse_counter:148]:   Accuracy Decimals: 2
[16:03:43][C][pulse_counter:148]:   Icon: 'mdi:pulse'
[16:03:43][C][pulse_counter:149]:   Pin: GPIO4
[16:03:43][C][pulse_counter:150]:   Rising Edge: INCREMENT
[16:03:43][C][pulse_counter:151]:   Falling Edge: DISABLE
[16:03:43][C][pulse_counter:152]:   Filtering pulses shorter than 13 µs
[16:03:43][C][pulse_counter:153]:   Update Interval: 5.0s
[16:03:43][C][display.tm1637:132]: TM1637:
[16:03:43][C][display.tm1637:133]:   Intensity: 7
[16:03:43][C][display.tm1637:134]:   Inverted: 1
[16:03:43][C][display.tm1637:135]:   Length: 4
[16:03:43][C][display.tm1637:136]:   CLK Pin: GPIO12
[16:03:43][C][display.tm1637:137]:   DIO Pin: GPIO14
[16:03:43][C][display.tm1637:138]:   Update Interval: 1.0s
[16:03:43][C][captive_portal:088]: Captive Portal:
[16:03:43][C][mdns:084]: mDNS:
[16:03:43][C][mdns:085]:   Hostname: leslie-controller
[16:03:43][C][ota:085]: Over-The-Air Updates:
[16:03:43][C][ota:086]:   Address: leslie-controller.local:8266
[16:03:43][C][api:138]: API Server:
[16:03:43][C][api:139]:   Address: leslie-controller.local:6053
[16:03:43][C][api:143]:   Using noise encryption: NO
[16:03:45][D][pulse_counter:162]: 'Leslie Horn Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:03:45][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:03:45][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:03:45][D][sensor:124]: 'Leslie Woofer Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:03:50][D][pulse_counter:162]: 'Leslie Horn Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:03:50][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:03:50][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:03:50][D][sensor:124]: 'Leslie Woofer Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:03:55][D][pulse_counter:162]: 'Leslie Horn Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:03:55][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:03:55][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[... 36 lines cut ...]
[16:04:40][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:04:40][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:04:40][D][sensor:124]: 'Leslie Woofer Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:04:45][D][pulse_counter:162]: 'Leslie Horn Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:04:45][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:04:45][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:04:45][D][sensor:124]: 'Leslie Woofer Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:04:50][D][pulse_counter:162]: 'Leslie Horn Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:04:50][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:04:50][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[... cut ...]
[16:05:05][D][pulse_counter:162]: 'Leslie Horn Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:05:05][D][sensor:124]: 'Leslie Horn Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:05:05][D][pulse_counter:162]: 'Leslie Woofer Pulse Counter': Retrieved counter: 0.00 pulses/min
[16:05:05][D][sensor:124]: 'Leslie Woofer Pulse Counter': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[16:05:07][D][api:102]: Accepted 192.168.1.173
[16:05:07][W][api.connection:083]: Home Assistant 2022.3.5 (192.168.1.173): Connection closed

HumanSkunk, 2022-03-24:
Same issue here. I had a working ESPHome board, then updated to the latest version of ESPHome and since then when trying to refresh the board (which succeeds) when it tries to connect with HA it is immediately followed by a connection closed message.

I have as far as I can tell removed all traces of the board within ESPHome and it doesnt appear as a device, but when I reflash I still get the same result. Odd observation though, when I used to flash boards initially with firmware it would appear as a device within HA. It is not doing that currently.

Nadav-K, 2022-03-24:
same issue here...

HumanSkunk, 2022-03-24:
So I set up a new ESP board fresh out the packet and it played ball. I since flashed it with the same code as the last one and it happily talks to HA. I have no idea why HA has blacklisted the board I tried yesterday, it makes no sense. I half expect me to brick a board with a bad solder or a misconnection...

Nadav-K, 2022-03-24:
I hope this helps other people as well.

Opening the HA container log showed some error regarding a name / ip mismatch for the board that keeps detaching.

Once I removed the board completely from the ESPHome integration, it was accepted as a new board with no issue.

HumanSkunk, 2022-03-24:
Yea my issue was I never had it as a device added within ESPHome like they normally do. So I think when I tried to remove it there was something residual that was causing it try and connect as my previous board. Which really shouldn't make a difference anyways.

MrAJB, 2022-03-25:
> I hope this helps other people as well.
> 
> Opening the HA container log showed some error regarding a name / ip mismatch for the board that keeps detaching. I wondered where that history is kept, as I tend to mostly play around with the yaml config for the board itself.
> 
> Once I removed the board device completely from the ESPHome integration, it was accepted as a new board with no issue.

Can you be a bit more specific, Nadav-K? What do you mean with 'removed the board completely from the ESPHome integration'?

Nadav-K, 2022-03-25:
> Can you be a bit more specific, Nadav-K? What do you mean with 'removed the board completely from the ESPHome integration'?

Sure - in HA i went to Configuration --> Devices and Services --> under the ESPHome tile I had a previous configuration used **for the same board**. I then chose that device -> 3 dots menu --> Delete.

Now that device is erased from HA, and so the MAC address of this board is not longer associated to an existing device -> it's recognized as new device. All errors previously shown in the HA logs no longer appear, and the logs of the board itself (viewing through ESPHome) are clean as well - no disconnections.
Hope this helps.

MrAJB, 2022-03-25:
I gave it a shot and it seems to resolve the issue. The board is now up and running for the past few minutes. Will leave it over the weekend and see if it remains table. Thanks Nadav-K!

pxFIN, 2022-04-06:
With singles like BMP680 / ESP32 and PMSA003 / ESP32 ... i didnt have a problem

pxFIN, 2022-04-06:
Examples of logs

[15:55:17][C][api:143]:   Using noise encryption: NO
[15:55:24][D][dht:048]: Got Temperature=0.5°C Humidity=97.0%
[15:55:24][D][sensor:125]: 'DHT22 Lämpötila': Sending state 0.50000 °C with 1 decimals of accuracy
[15:55:24][D][sensor:125]: 'DHT22 kosteus': Sending state 97.00000 % with 0 decimals of accuracy
[15:55:24][D][bmp280.sensor:149]: Got temperature=0.5°C pressure=984.3hPa
[15:55:24][D][sensor:125]: '280bmp_lämpötila': Sending state 0.52000 °C with 1 decimals of accuracy
[15:55:25][D][sensor:125]: '280bmp_ilmanpaine': Sending state 984.26080 hPa with 1 decimals of accuracy
[15:55:39][D][dht:048]: Got Temperature=0.5°C Humidity=97.4%
[15:55:39][D][sensor:125]: 'DHT22 Lämpötila': Sending state 0.50000 °C with 1 decimals of accuracy
[15:55:39][D][sensor:125]: 'DHT22 kosteus': Sending state 97.40000 % with 0 decimals of accuracy
[15:55:40][D][bmp280.sensor:149]: Got temperature=0.5°C pressure=984.3hPa
[15:55:40][D][sensor:125]: '280bmp_lämpötila': Sending state 0.53000 °C with 1 decimals of accuracy
[15:55:40][D][sensor:125]: '280bmp_ilmanpaine': Sending state 984.28436 hPa with 1 decimals of accuracy
[15:55:54][D][dht:048]: Got Temperature=0.5°C Humidity=97.7%
[15:55:54][D][sensor:125]: 'DHT22 Lämpötila': Sending state 0.50000 °C with 1 decimals of accuracy
[15:55:54][D][sensor:125]: 'DHT22 kosteus': Sending state 97.70000 % with 0 decimals of accuracy
[15:55:54][D][bmp280.sensor:149]: Got temperature=0.5°C pressure=984.3hPa
[15:55:54][D][sensor:125]: '280bmp_lämpötila': Sending state 0.50000 °C with 1 decimals of accuracy
[15:55:54][D][sensor:125]: '280bmp_ilmanpaine': Sending state 984.27155 hPa with 1 decimals of accuracy
[15:56:09][D][dht:048]: Got Temperature=0.5°C Humidity=97.5%
[15:56:09][D][sensor:125]: 'DHT22 Lämpötila': Sending state 0.50000 °C with 1 decimals of accuracy
[15:56:09][D][sensor:125]: 'DHT22 kosteus': Sending state 97.50000 % with 0 decimals of accuracy
[15:56:09][D][bmp280.sensor:149]: Got temperature=0.4°C pressure=984.3hPa
[15:56:09][D][sensor:125]: '280bmp_lämpötila': Sending state 0.44000 °C with 1 decimals of accuracy
[15:56:09][D][sensor:125]: '280bmp_ilmanpaine': Sending state 984.29810 hPa with 1 decimals of accuracy
[15:56:13][D][api:102]: Accepted ::FFFF:C0A8:FE67
[15:56:13][W][api.connection:085]: ::FFFF:C0A8:FE67: Reading failed: BAD_INDICATOR errno=11

I changed back to normal unencrypted world with plain password + ota

pxFIN, 2022-04-06:
I still keep getting BAD INDICATOR status

Thus i ask the community + developers.. whats going on ?

SlowStopper, 2022-04-08:
I had the same issue for past 2 months, my BLE temperature trackers are working fine (they are visible in ESPhome logs!), but their state is not reported to HA. 

It always goes like this:
[14:32:12][D][api:102]: Accepted ::FFFF:C0A8:1F5
[14:32:12][W][api.connection:083]: Home Assistant 2022.3.5 (::FFFF:C0A8:1F5): Connection closed

What is a bit weird is that the address reported looks like IPv6 address, but the actual connection is via IPv4 (as far as I can tell with tcpdump.

Now it's fixed, here's how:
1. Don't touch anything under "ESPHome" section.
2. In Devices&Configuration, find your board -> three dots -> delete
3. Add the device back (Add integration -> ESPHome)

There's no need to update any Dashboards, it will just start working.

TomNH643, 2022-04-08:
SlowStopper, thank you, solved the problem I was having with API disconnects.

RichardvanGestel, 2022-04-13:
I'm going crazy with this problem. I haven't been able to create new devices for a few days. They always lose the connection after some time. Fortunately, existing devices continue to function normally. HELP?

RichardvanGestel, 2022-04-13:
The issues occur with ESP32 and ESP8266

RichardvanGestel, 2022-04-13:
Example log:
INFO Reading configuration /config/esphome/esp32-test03.yaml...
INFO Starting log output from esp32-test03.local using esphome API
WARNING Can't connect to ESPHome API for esp32-test03.local: Hello timed out
INFO Trying to reconnect to esp32-test03.local in the background
INFO Successfully connected to esp32-test03.local
[23:32:59][I][app:102]: ESPHome version 2022.3.2 compiled on Apr 13 2022, 21:35:27
[23:32:59][C][wifi:491]: WiFi:
[23:32:59][C][wifi:353]:   Local MAC: 4C:11:AE:C8:E2:6C
[23:32:59][C][wifi:354]:   SSID: [redacted]
[23:32:59][C][wifi:355]:   IP Address: 192.168.178.115
[23:32:59][C][wifi:357]:   BSSID: [redacted]
[23:32:59][C][wifi:358]:   Hostname: 'esp32-test03'
[23:32:59][C][wifi:360]:   Signal strength: -61 dB ▂▄▆█
[23:32:59][C][wifi:364]:   Channel: 1
[23:32:59][C][wifi:365]:   Subnet: 255.255.255.0
[23:32:59][C][wifi:366]:   Gateway: 192.168.178.1
[23:32:59][C][wifi:367]:   DNS1: 192.168.178.1
[23:32:59][C][wifi:368]:   DNS2: 0.0.0.0
[23:32:59][C][logger:233]: Logger:
[23:32:59][C][logger:234]:   Level: DEBUG
[23:32:59][C][logger:235]:   Log Baud Rate: 115200
[23:32:59][C][logger:236]:   Hardware UART: UART0
[23:32:59][C][captive_portal:088]: Captive Portal:
[23:32:59][C][mdns:084]: mDNS:
[23:32:59][C][mdns:085]:   Hostname: esp32-test03
[23:32:59][C][ota:085]: Over-The-Air Updates:
[23:32:59][C][ota:086]:   Address: esp32-test03.local:3232
[23:32:59][C][ota:089]:   Using Password.
[23:32:59][W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts
[23:32:59][C][api:138]: API Server:
[23:32:59][C][api:139]:   Address: esp32-test03.local:6053
[23:32:59][C][api:143]:   Using noise encryption: NO
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO esp32-test03.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO esp32-test03.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
WARNING Can't connect to ESPHome API for esp32-test03.local: Hello timed out
INFO Trying to reconnect to esp32-test03.local in the background
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO esp32-test03.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO esp32-test03.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO esp32-test03.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO esp32-test03.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Ping timed out!
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API
INFO Successfully connected to esp32-test03.local
INFO esp32-test03.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for esp32-test03.local
WARNING Disconnected from API

RichardvanGestel, 2022-04-13:
I have tried everything that was suggested as a solution... and more... but nothing helps...

han9988, 2022-04-14:
Follow-up to my post of march 15th: Turned out the connection problem consisted of two parts:
1. It seems that the esp32 alternates between bluetooth and wifi, hence the reason of connected,disconnected,connected,etc.
2. Resulting of 1. there was a timing problem which went away with another ota I did. The thought behind the new compile was: ble tracker takes longer than expected because of weak rssi strength + dhcp every time it connects to wifi = could be the timing issue. So I converted to static ip and the problem went away and runs stable for two weeks now.

RichardvanGestel, 2022-04-14:
Here it works again. There were a number of old devices in esphome ... that had been on the hardware I was using now ... I cleaned everything up and can now make new devices again. Previously this was not a problem, now apparently it is. So I'm always going to tidy up.

CBPetrovic, 2022-05-15:
I also have been experiencing the same issue, using a Nodemcu-32s, monitoring my home energy usage. All was working fine, then out of the blue, sensors became unavailable. Long story short, and on a hail-mary hunch, I added a this:
#
  power_save_mode: none
#
which for the nodemcu-32s defaults to "light"

So far, it's working again.

CASATAJO, 2022-07-14:
have same issue, follow what SlowStopper recomended, but get same problem, every minute
	[api.connection:080] ESPHome Logs 2022.6.2 (::FFFF:C0A8:164): Connection closed

patrickcollins12, 2022-07-15:
In my case this issue was caused by me simply flashing a new yaml, with a new "friendly_name" to a previous board that was still registered as an active esphome integration in home assistant.

Specifically:
sht40.yaml --> sht40.local --> mac 0C:DC:7E:63:3B:80
temt6000.yaml --> temt6000.local --> mac1 0C:DC:7E:63:3B:80

Both were concurrently registered as integrations to home assistant. 

So the obvious fix was to remove the old device (sht40) from home assistant, then the problem just disappeared. Specifically, home assistant kept the connection alive to the new device and the log entry stopped appearing.

mmakaay, 2022-07-16:
Did you try following the recipe from [this post](https://github.com/esphome/issues/issues/3108#issuecomment-1092819059) ?

When there are multiple devices in HA pointing to the same ESPHome device, this behavior will happen. Looking at the comments, this is what might be happening for most of the posters.

@SlowStopper 
> What is a bit weird is that the address reported looks like IPv6 address, but the actual connection is via IPv4 (as far as I can tell with tcpdump.

That is only a representation issue. When you see an IP address formatted like that, it is IPv4 at the network level.

CASATAJO, 2022-07-20:
Yes I did, that but same problem happens
I think the registry keeps even if deleting the devices from the ESPhome
integration

Eric-N-Be, 2022-07-26:
hi, i have the same problem ...
[W][api.connection:068]: ::FFFF:C0A8:14: Socket operation failed: CONNECTION_CLOSED errno=128

JaskaV, 2022-08-14:
Hi all

Have to comment to this thread because I faced exactly same issue (this connection closed repeatedly). My case was that I had a custom_component in my ESP32 and I earlier used without problems "°C" as a input for set_unit_of_measurement and everything was fine, but suddenly with new "version" of ESPHome this "connection closed repeatedly" started. But when removing the degree mark, meaning changing the set_unit_of_measurement to " C" the problem disappeared.

Might be that the degree mark is not allowed or then there is a bug under the hood...

Steve2017, 2022-08-16:
> 1. Don't touch anything under "ESPHome" section.
> 2. In Devices&Configuration, find your board -> three dots -> delete
> 3. Add the device back (Add integration -> ESPHome)
> 
> There's no need to update any Dashboards, it will just start working.

This fixed a few issues for me - so thank you.

jflattery, 2022-08-22:
I believe #2588 and #3320 are related to this. Unfortunately for me, [SlowStopper](https://github.com/SlowStopper)'s method does not resolve the issue for me.

ssieb, 2022-08-26:
Almost all the issues here are related to changing the name on a device.  Anyone having a similar issue needs to check the **serial** logs and HA logs.  If you don't understand what you see there, then come to the discord server and make a thread with that data.  Only create an issue if you are sure that there's actually an esphome bug.

closed 2022-08-26 by ssieb

1cdo, 2022-11-18:
> Sure - in HA i went to Configuration --> Devices and Services --> under the ESPHome tile I had a previous configuration used **for the same board**. I then chose that device -> 3 dots menu --> [...]

This worked for me as well.  Thank you Nadav-K!

amaisano, 2023-01-16:
None of these suggests have worked for me.

I have an ESP 32, and since enabling ble tracking (bt proxy stuff), it's constantly disconnecting and reconnecting (every 1-2 minutes).

I've tried the latest ESPHome (2023.1 dev) but it didn't help. I also tried removing everything from HA and manually re-adding via IP address, but that did not help either.

My logs show this constantly:

```
INFO everything-presence-one-e01d58.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO everything-presence-one-e01d58.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for everything-presence-one-e01d58.local
WARNING Disconnected from API
INFO Successfully connected to everything-presence-one-e01d58.local
```

And when I use the USB based logs:

```
[D][api:102]: Accepted ::FFFF:192.168.86.36
[D][api.connection:918]: Home Assistant 2023.1.4 (::FFFF:192.168.86.36): Connected successfully
```

whenever it blinks away from HA (shows as unavailable).
```

adamscybot, 2023-02-16:
Thanks to jesserockz / ssieb for helping me through this on discord. This can happen if you have a device flashed ok through web.esphome.io fine, but your install through the ESPHome tab on HA didn't succeed. In my case, the logs during install indicated a failed OTA flash. If you are in this state and view something like logs you will get `errno=11`.

If you do the install whilst also viewing the logs through web.esphome.io you can see the error that is causing the failed flash. In my case the `board` was set to the wrong device instead of the one for my board, which you can find on https://registry.platformio.org/platforms/platformio/espressif32/boards.

ssieb, 2023-02-18:
@amaisano , what else is in the logs around that?  There must a disconnect message of some sort.  Also, it would be better to come ask on Discord instead of commenting on a closed issue.

aheath70, 2023-03-06:
> I had the same issue for past 2 months, my BLE temperature trackers are working fine (they are visible in ESPhome logs!), but their state is not reported to HA.
>
> It always goes like this: [...]

This worked for me. I tried just about everything, include a restore of ESPHOME, and this is what finally fixed it for me. Thanks @SlowStopper ...

aholz, 2023-03-14:
> > 1. Don't touch anything under "ESPHome" section.
> > 2. In Devices&Configuration, find your board -> three dots -> delete
> > 3. Add the device back (Add integration -> ESPHome)
> > 
> > There's no need to update any Dashboards, it will just start working.
> 
> This fixed a few issues for me - so thank you.

For me as well, thanks a lot. I was deleting in EspHome, not on HomeAssistant. I didn't knew how to do it. Your detailed description was the solution!!

mesistostefre, 2023-04-28:
Same issue, so I did remove my Wemos D1 Mini from HA, added again using Add integration, ESPHome, entering the encryption key, installing, etc... 
It is the one and only ESPHome device installed.
WIFI signal is "good enough", but the device keeps reconnecting **once a second.**.. 
[08:42:41][D][api:102]: Accepted 192.168.xxx.xx
[08:42:41][W][api.connection:087]: Home Assistant 2023.4.5 (192.168.xxx.xx): Connection closed
[08:42:41][D][api.connection:961]: Home Assistant 2023.4.5 (192.168.xxx.xx): Connected successfully
[08:42:42][D][api:102]: Accepted 192.168.xxx.xx
[08:42:42][W][api.connection:087]: Home Assistant 2023.4.5 (192.168.xxx.xx): Connection closed
[08:42:42][D][api.connection:961]: Home Assistant 2023.4.5 (192.168.xxx.xx): Connected successfully
[08:42:42][D][api:102]: Accepted 192.168.xxx.xx
[08:42:42][W][api.connection:087]: Home Assistant 2023.4.5 (192.168.xxx.xx): Connection closed
[08:42:43][D][api.connection:961]: Home Assistant 2023.4.5 (192.168.xxx.xx): Connected successfully
...
Definitely don't want to see this thing in my network permanently.
Am running out of ideas...

ssieb, 2023-04-28:
Look in the HA logs and come ask on Discord instead of commenting on a closed issue.

Hohenloherin, 2023-06-03:
I have had a similar problem and thanks to the discussion here I discovered that indeed a typo in the ip-address of my recently attached esp32 has had the same ip-address as another one already existing in my network. The older registered one was not in operation, but the there was a conflict in HA! 
Deletion of these two in HA under "Devices and Services"-" ESPHOME" and then reinstalling with correct issued ip-addresses fixed the mistake. Thanks a lot!

dvhub, 2023-07-02:
I had a similar problem.  I had several different ESPhome devices producing the error message "Connection error occurred: [Errno 104] Connection reset by peer" approximately every 1-2 minutes.  In the debug logs, the error appeared as follows:

WARNING sonoff-s31--2.local: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for sonoff-s31--2.local
WARNING Disconnected from API
INFO Successfully connected to sonoff-s31--2.local

The root cause was related to the configuration of my Ubiquiti UniFi network.  I had disabled the 5 GHz frequency of my wifi IoT network so that a 2.4 GHz device could be added.  I failed to reenable the 5 GHz network and this somehow caused the problem with all of my ESPhome devices.  Once the 5 GHz band was re-enabled, everything was fine again.

apbodrov, 2023-07-27:
Hello everyone, I recently encountered the problem of inaccessible sensors (esp 32 was online and its web interface worked).
On esp32, there was something like 
`Home Assistant Requested disconnect`
in the logs of HA (Core) I saw a strange thing
`Can't connect to ESP Home API for xxx@ yyy: Error connecting to ('yyy', 6053): [Errno 111] Connect call failed ('yyy', 6053) (Socket API Error)`
Where `yyy` is the address of the host machine where there will obviously be an error. My Home Assistant instance works in KVM on Fedora Linux. At the same time, xxx did not resolve on the host machine, and I decided to fix the mdns resolution on the host machine. I haven't fixed the resolution, but for two days everything has been working for me without a single shutdown. Before that, every day the sensors became inaccessible for several minutes for many hours.
I did the following (I don't know which of these is really useful):
1. br0 as the allowed interface for avahi
2. Set `MulticastDNS=resolve` in /etc/systemd/resolve.conf.
3. Create a file /etc/NetworkManager/conf.d/enable-mdns.conf with `connection.mdns=1` set in the `[connection]` section.
4. Restart avahi-daemon and NetworkManager.