esphome/issues #3112: empty {} topic config json in mqtt discovery

twocolors, 2022-03-04:
### The problem

send empty {} json in config topic, all work fine in 2022.2.5

### Which version of ESPHome has the issue?

2022.2.6

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
substitutions:
  device_name: livingroom-ac

esphome:
  name: ${device_name}
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Living Room Ac Hotspot"
    password: !secret ap_password

captive_portal:

# Enable logging
logger:
  level: WARN
  # level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
# api:

ota:
  password: !secret ota_password

web_server:

esp32_ble_tracker:
#  scan_parameters:
#    window: 120ms

mqtt:
  broker: !secret mqtt_host
  topic_prefix: esphome/${device_name}

text_sensor:
  - platform: version
    name: "Version"
  
  - platform: wifi_info
    ip_address:
      name: "Address"

  - platform: template
    id: uptime_human
    name: "Uptime"
```

jerobins, 2022-03-04:
2022.2.6 and 2022.2.7 broke mqtt json publish - json published payload is empty '{}'.  Downgrade to 2022.2.5 resolves.

basschipper, 2022-03-11:
I have the same issue, same platform, same board. Downgrade fixed this for now.

isorin, 2022-03-13:
The problem exists in 2022.2.5 (and before) for multiple sensors coming from the BLE tracker.
It happens if the sensor doesn't publish data for some time after boot (or at least this seems to be the case).

jesserockz, 2022-03-13:
I have just tried to reproduce this and can confirm it happens when `esp32_ble_tracker` is enabled (it is fine without it so maybe the tracker is using too much memory for the json to be built). I am investigating further

isorin, 2022-03-13:
That's great, thanks!
Due to the way it behaves I am leaning more towards a race condition / clash of resources when preparing or publishing the mqtt json but I'm not familiar with the code so it is just a wild guess.

jesserockz, 2022-03-13:
@isorin Please try out esphome/esphome#3289

```yaml
external_components:
  - source: github://pr#3289
    components: ["json"]
```

isorin, 2022-03-13:
Getting these errors:
`[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656
[23:13:35][E][json:034]: Could not allocate memory for JSON document! Requested 10240 bytes, free heap: 38656`

jesserockz, 2022-03-13:
Oh man, now I don't understand why it cant allocate. I followed the chain along and it simply takes the number and parses it directly to `malloc`. Which in this case is returning a `nullptr` because there is not enough memory.
For my esp32 it is working as expected, but Ill try reproduce it more

ssieb, 2022-03-14:
The heap is probably too fragmented.  I think there's another memory check function for what the largest possible allocation is.  That's a big json document though.  Is that a fixed value?

ssieb, 2022-03-14:
`heap_caps_get_largest_free_block()`

ssieb, 2022-03-14:
Sorry, I should have checked the PR before commenting.  That is being used already.

twocolors, 2022-03-14:
i repeated this problem on esp8266 - sonoff mini and version 2022.2.6

All config topics are empty

isorin, 2022-03-14:
I think we are looking at two issues here:
1) the issue in 2022.2.6 in which all config entries are empty
2) the issue in 2022.2.5 (and earlier versions) in which the json fails to allocate memory and random config entries are empty

For the 2nd issue: 
@jesserockz Do you think changing the DefaultAllocator in DynamicJsonDocument to use heap_caps_malloc(size, MALLOC_CAP_DEFAULT) would make a difference?

isorin, 2022-03-14:
Btw. I changed the limit to 4K instead of 10K and I still see 1 error message.

jesserockz, 2022-03-14:
Yeah I was trying to figure out what is happening yesterday and will continue again today. 
It is extremely weird but perhaps the free heap is not giving the proper number.

randybb, 2022-03-14:
@jesserockz thanks! my nspanel is finally able to load json :D

pironic, 2022-03-16:
> @isorin Please try out [esphome/esphome#3289](https://github.com/esphome/esphome/pull/3289)
> 
> ```yaml
> external_components:
>   - source: github://pr#3289
>     components: ["json"]
> ```

Just wanted to point out that this was a viable fix for me.

closed 2022-03-22 by jesserockz

isorin, 2022-03-23:
I think I already did, I don't see any other changes in the last version from what I already tried.
I also tried with 4K instead of 5K and that errored out as well (see above).
My config for ESP32 is big, I have 6 temp/hum sensors (2 Govee + 4 Xiaomi ATC sensors) and it is working fine with ESP native API. I will stick with it for the time being. (At least for ESP32)