esphome/issues #3113: "esphome logs" fails with error "missing 1 required positional argument: password"

M95D, 2022-03-05:
### The problem

Running this command and selecting OTA fails with error:
TypeError: __init__() missing 1 required positional argument: 'password'

Selecting USB serial port works fine without error.

### Which version of ESPHome has the issue?

2022.3.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Board

M5StickC

### Component causing the issue

api client

### Example YAML snippet

```yaml
esphome:
  platform: ESP32
  board: m5stick-c

wifi:
  ssid: <ssid>
  password: <password>
  fast_connect: true
  manual_ip:
    static_ip: <board ip>
    gateway: <router ip>
    subnet: 255.255.255.0
    dns1: <router ip>

logger:
  level: VERBOSE

api:
  port: <port>
  reboot_timeout: 12h

ota:
  port: <port>
  password: "<password>"
```

### Anything in the logs that might be useful for us?

```txt
$ esphome logs <myconfig>.yaml
INFO Reading configuration <myconfig>.yaml...
WARNING Pin 10 (9-10) might already be used by the flash interface in QUAD IO flash mode.
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (M5stack)
  [2] Over The Air (<ip>)
(number): 2
INFO Starting log output from <ip> using esphome API
Traceback (most recent call last):
  File "/home/<user>/.local/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/home/<user>/.local/lib/python3.9/site-packages/esphome/__main__.py", line 825, in main
    return run_esphome(sys.argv)
  File "/home/<user>/.local/lib/python3.9/site-packages/esphome/__main__.py", line 812, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/home/<user>/.local/lib/python3.9/site-packages/esphome/__main__.py", line 362, in command_logs
    return show_logs(config, args, port)
  File "/home/<user>/.local/lib/python3.9/site-packages/esphome/__main__.py", line 284, in show_logs
    return run_logs(config, port)
  File "/home/<user>/.local/lib/python3.9/site-packages/esphome/components/api/client.py", line 71, in run_logs
    asyncio.run(async_run_logs(config, address))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/<user>/.local/lib/python3.9/site-packages/esphome/components/api/client.py", line 24, in async_run_logs
    cli = APIClient(
TypeError: __init__() missing 1 required positional argument: 'password'
```

M95D, 2022-03-05:
I tried these and none of them fix the problem:
- removing OTA password
- esphome clean
- git pull

OttoWinter, 2022-03-09:
How did you install esphome? It looks to me like you have an outdated `aioesphomeapi` version installed (esphome pins the version, so it should only happen if you installed it in a weird way/dev setup).

M95D, 2022-03-09:
I'm using Gentoo Linux. I'm sure there is a better way to install the git master, but I could find no documentation on how to do that. This is what I did:

**pip3 install --user** (this installs into /home/user/.local/lib/python<ver>/site-packages/esphome)
**git clone** into **/build/esphome/** (so I can have the latest git master instead of the stable version that pip3 installs)
**mv /home/user/.../esphome** to some other place
**ln -s /build/esphome /home/user/.../esphome** (to make it work from /build/esphome)
Then I updated several times with git pull and when I was prompted to by the esphome and platformio using the commands indicated at the time.

closed 2022-11-07 by github-actions[bot]