esphome/issues #3117: Invalid Unit

Herbycheck, 2022-03-06:
### The problem

I am trying to set up a [SlimmeLezer+](https://www.zuidwijk.com/product/slimmelezer-plus/) to read data from my smart meter (DSMR 5), but i get an error when trying to read my consumed energy.

If i only request the DSMR Version or Identification it works fine, but as soon as i add the energy_delivered_tariff1 sensor all readings become unavailable.

### Which version of ESPHome has the issue?

2022.2.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.1

### What platform are you using?

ESP8266

### Board

SlimmeLezer+

### Component causing the issue

dsmr

### Anything in the logs that might be useful for us?

```txt
[E][dsmr:265]: 1-0:1.8.0(017225650*Wh)
                    ^
Invalid unit
```

zuidwijk, 2022-03-09:
1-0:1.8.0 is meter reading for Luxembourg. Are you from Luxembourg?

Herbycheck, 2022-03-09:
No I am from Austria and i am using the `energy_delivered_tariff1` sensor and a decryption key.

schlenschi, 2022-03-28:
> No I am from Austria and i am using the `energy_delivered_tariff1` sensor and a decryption key. Would it help if I posted my configuration?

Hello! I have exactly the same issuse with 

[E][dsmr:265]: 1-0:1.8.0(017225650*Wh)
                    ^
Invalid unit

I'am from Austria too! Did you fixed the Problem?
I think we don't need to use the decryption-key in the "Luxenburg-Section". I disabled it agian, and the Issue stays the same. Its not a Problem of reading, it's just a Problem of interpration, i guess...

Herbycheck, 2022-03-28:
Sadly i haven't found an answer yet. I think the problem is that the Luxembourg smart meters don't use Wh as their energy unit and that it is assumed that you are in Luxembourg if you use a decryption key.

I also have to use an encryption key or I get an error.

Herbycheck, 2022-03-28:
I found the issue. The dsmr sensor does not recognize Watt Hours as a unit (I think after trying to understand [the code](https://github.com/esphome/esphome/blob/dev/esphome/components/dsmr/sensor.py)).

schlenschi, 2022-04-01:
And how did you fix that error? Where is the issue? @Herbycheck

HugoTheBoss, 2022-10-31:
There is also a problem with:
OBIS 3.7.0 (reactive_power_delivered) and  4.7.0 (reactive_power_returned) both needs a unit of kvar but my Smart Meter delivers var 1-0:3.7.0(000000000*var)
DEFINE_FIELD(reactive_power_delivered, FixedValue, ObisId(1, 0, 3, 7, 0), FixedField, units::kvar, units::kvar);
and
OBIS 3.8.0 (total_imported_energy) and 4.8.0 (total_exported_energy) needs a unit kvarh but my Smart Meter delivers varh
1-0:3.8.0(000038847*varh), 1-0:4.8.0(005986252*varh)
DEFINE_FIELD(total_exported_energy, FixedValue, ObisId(1, 0, 4, 8, 0), FixedField, units::kvarh, units::kvarh);

Can you please add a fix for this too?

Mosfet31994, 2024-04-29:
20:13:54 | [E] | [dsmr:265] | 1-0:32.7.0(0000.231*kV)                     ^ Invalid unit
20:13:55 | [E] | [dsmr:265] | 1-0:32.7.0(0000.230*kV)                     ^ Invalid unit

Please help

Herbycheck, 2024-04-29:
Im sorry i botched this fix together a long time ago and dont really want to maintain it

closed 2024-04-29 by Herbycheck