esphome/issues #3123: NTP only uses the first server

M95D, 2022-03-09:
### The problem

Adding more than one NTP server is futile. Esphome only uses the first.
```
root@GRAPHRT:~# tcpdump -i br-lan port 123
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br-lan, link-type EN10MB (Ethernet), capture size 262144 bytes
08:52:06.645636 IP GRAPHT2.lan.57862 > GRAPHRT.123: NTPv4, Client, length 48
08:52:19.647165 IP GRAPHT2.lan.65233 > GRAPHRT.123: NTPv4, Client, length 48
08:52:19.665991 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:52:25.669558 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:52:34.672243 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:52:49.673318 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:53:16.676807 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:53:43.680915 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:54:10.685244 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:54:37.687337 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:55:04.691656 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:55:31.696907 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:55:58.699294 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:56:25.703301 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:56:52.706770 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:57:19.711066 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:57:46.714242 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:58:13.719742 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:58:40.724123 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:59:07.725250 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
08:59:34.728671 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
09:00:01.735174 IP GRAPHT2.lan.65234 > GRAPHRT.123: NTPv4, Client, length 48
```

"Synchronized system clock" never appears.
"Time? / Time is not set." never appears. on_time program never runs. on_boot, if uncommented, also never runs.

### Which version of ESPHome has the issue?

2022.3.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

M5StickC

### Component causing the issue

NTP

### Example YAML snippet

```yaml
esphome:
  name: bug-test
  platform: ESP32
  board: m5stick-c
#  on_boot:
#    priority: -100
#    then:
#    - logger.log: "Boot script waiting for time..."
#    - wait_until:
#        time.has_time:
#    - logger.log: "Boot script got time!"

wifi:
  ssid: <my_ssid>
  password: <my_pass>
  manual_ip:
    static_ip: 172.27.143.48
    gateway: 172.27.143.1
    subnet: 255.255.255.0
    dns1: 172.27.143.1

time:
  - platform: sntp
    timezone: "Europe/Bucharest"
    servers:
      - 172.27.143.1
      - pool.ntp.org
    on_time_sync:
      then:
      - logger.log: "Synchronized system clock"
    on_time:
      - seconds: /10
        then:
          - logger.log: "Time?"
          - if:
              condition:
                - time.has_time:
              then:
                - logger.log: "Time is valid."
              else:
                - logger.log: "Time not set."
```

OttoWinter, 2022-03-09:
Ok that would likely be an issue in the esp-idf/lwip framework then; in esphome we just pass the list of servers to lwip and it should handle the servers

ssieb, 2022-10-31:
Apparently the build flag esphome is using isn't working.  `SNTP_MAX_SERVERS` is still `1`.

ssieb, 2022-10-31:
The problem is that the `esp8266` component is setting a conflicting flag:
https://github.com/esphome/esphome/blob/dev/esphome/components/esp8266/__init__.py#L197
`cg.add_build_flag("-DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH")`