esphome/issues #3128: LEDC not working with ESP32-C3 and Arduino framework (but with ESP-IDF)

MFlasskamp, 2022-03-11:
### The problem

The LEDC output component produces no output with a monochromatic light. The board I'm using has an ESP32-C3 (NodeMCU ESP-C3-13) and is compiled with Arduino framework (cf. example yaml).

- working if compiled with ESP-IDF
- working on old ESP32 with Arduino
- NOT working on ESP32-S2 with Arduino

No compile errors or errors in the log when light is controlled by Home Assistant.

platformio/espressif32 @ 3.5.0
 - framework-espidf 3.40302.0 (4.3.2) 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2

### Which version of ESPHome has the issue?

2022.2.6

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.3.3

### What platform are you using?

ESP32

### Board

NodeMCU ESP-C3-13

### Component causing the issue

ledc

### Example YAML snippet

```yaml
esphome:
  name: esp32c3-01

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:

########################################################

### Arduino
    type: arduino
    version: dev
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

### IDF
    # type: esp-idf
    # sdkconfig_options:
    #   CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

########################################################

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: true
  manual_ip:
    static_ip: 192.168.1.70
    gateway: 192.168.1.1
    subnet: 255.255.255.0

ota:
  password: !secret ota_pass

api:
  password: !secret api_pass

logger:
  level: DEBUG  # WARN DEBUG

output:
  - id: led_18
    platform: ledc
    pin: GPIO18

light:
  - platform: monochromatic
    name: esp32c3-01 led 18
    output: led_18
```

### Additional information

Framework/toolchain versions:
```
platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
framework-arduinoespressif32 0.0.0+sha.6a7bcab 
toolchain-riscv32-esp 8.4.0+2021r2
```

Matchlighter, 2022-09-02:
Observing this as well

MFlasskamp, 2022-09-16:
The PR above fixes wrong calculation of the LEDC/PWM resolution. Moreover I had to use the most recent releases of [framework ](https://github.com/espressif/arduino-esp32/releases) and [platform](https://github.com/platformio/platform-espressif32/releases). Here is a snippet to give it a try:

```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: 2.0.4
    platform_version: 5.1.1

external_components:
  - source: github://pr#3820
    components: [ ledc ]
```

Matchlighter, 2022-09-18:
@MFlasskamp Hmm. Not working for me. It compiles and uploads, but I end up with a bootloop:
```
[13:29:45]Build:Feb  7 2021
[13:29:45]rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[13:29:45]Saved PC:0x4004d1f8
```

bls220, 2022-09-25:
This worked for me.
```
  board: adafruit_feather_esp32s3_nopsram
  variant: esp32s3
  framework:
    type: arduino
    version: dev
    platform_version: ^5.1.1
```

ssieb, 2022-09-25:
> variant: esp32s3

That's an S3, not a C3.

MFlasskamp, 2022-09-26:
Great! And my PR fixes the esp32s2, too. The current esphome code has the LED PWM timer resolution hard coded to 20 bit. This is only valid for the old esp32. The esp32s2 needs 18 bit, esp32c3 18 bit and esp32s3 14 bit.

closed 2022-12-22 by jesserockz

kozerscy, 2023-02-09:
Why it's not released yet? What's the point in closing issues as long as they are not released?

MFlasskamp, 2023-02-09:
The pull request https://github.com/esphome/esphome/pull/3820 is merged into the 2023.2.0 betas: https://github.com/esphome/esphome/releases

kozerskil, 2023-02-09:
Great, thank you!