esphome/issues #3129: Invalid username or password via home assistant egress page

maretodoric, 2022-03-12:
### The problem

In Home Assistant ESPHome configuration (under supervisor) I've configured host port 8082 for web interface. That way i can get to ESPHome dashboard outside of home assistant by hitting it with home assistant IP address but port 8082 instead of 8123.

Everything worked fine until up at some point, not sure since what version, i can no longer login to that page using home assistant credentials , even though page itself says: "Login by entering your Home Assistant login credentials."

I've tested credentials in incognito mode on home assistant itself and credentials are fine. But ESPHome will still not recognize them.

### Which version of ESPHome has the issue?

2022.2.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2022.3.3

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
2022-03-12 02:46:19,149 WARNING 401 POST /login (0.0.0.0) 22.40ms
2022-03-12 02:46:19,204 INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 3.19ms
2022-03-12 02:46:19,216 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838 (0.0.0.0) 3.46ms
2022-03-12 02:46:19,229 INFO 200 GET /static/css/esphome-2.css?hash=b79b4733 (0.0.0.0) 3.29ms
2022-03-12 02:46:19,238 INFO 200 GET /static/css/vendor/materialize/materialize.min.css?hash=ec1df3ba (0.0.0.0) 4.98ms
2022-03-12 02:46:19,255 INFO 200 GET /static/js/vendor/jquery/jquery.min.js?hash=12108007 (0.0.0.0) 11.25ms
2022-03-12 02:46:19,261 INFO 200 GET /static/js/vendor/jquery-ui/jquery-ui.min.js?hash=cab3392d (0.0.0.0) 15.10ms
2022-03-12 02:46:19,265 INFO 200 GET /static/js/vendor/jquery-validate/jquery.validate.min.js?hash=aad785d4 (0.0.0.0) 17.26ms
2022-03-12 02:46:19,272 INFO 200 GET /static/js/vendor/materialize/materialize.min.js?hash=4be20daf (0.0.0.0) 21.32ms
2022-03-12 02:46:19,330 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2 (0.0.0.0) 4.56ms
```

elkarto, 2022-03-22:
Same issue here.

Dashboard is accessible using ingress (from HA gui) but not accessible directly using the exposed port.

Perhaps Two-Factor Authentication related?

camnop, 2022-04-18:
Same issue here.

Dashboard is accessible using ingress (from HA gui) but not accessible directly using the exposed port.

cranxi, 2022-04-29:
Same issue here, also not sure at what version this started but I can't login using HA credentials on the ESPhome dashboard outside HA.

baldisos, 2022-04-30:
Same issue here. I want to access ESPHome directly over a reverse proxy, and i can't login because of that issue.

rmarskell, 2022-07-04:
I'm also experiencing this issue. No response after four months and four minor versions. 😣
```
 Add-on version: 2022.6.2
 System: Home Assistant OS 8.2  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2022.6.7
 Home Assistant Supervisor: 2022.05.3
```

maretodoric, 2022-07-04:
Wow it has been 4 months already? C'mon , someone should help us out :D

Spirituss, 2022-08-24:
Same is here. Can't login with HA user/pwd to esphome:6052 address.

chertvl, 2022-08-24:
Plus one. Same here.

closed 2022-09-06 by jesserockz

jesserockz, 2022-09-06:
This has been fixed in 2022.8.3