esphome/issues #3130: Component was marked as failed

paulonHA, 2022-03-12:
### The problem

What is causing a component to fail? I am trying to connect a bme/bmp280 (which is btw looking exactly as the SHT3x-D) sensor over a D1 mini to esphome. Unfortunately I am getting an error message when flashing [E][component:112]: Component bme280.sensor was marked as failed and also [E][bme280.sensor:147]: Communication with BME280 failed! 
I tried already 2 sensors to make sure that they are not broken. How can I fix this? I also tried bmp and sht3x-d to connect but I get everytime the same error message.

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.3.1

### What platform are you using?

ESP8266

### Board

D1 mini

### Example YAML snippet

```yaml
sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x77
    update_interval: 10s
i2c:
   - sda: D2
     scl: D1
     scan: true
     id: bus_a
```

### Anything in the logs that might be useful for us?

```txt
INFO Successfully compiled program.
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:91:51:51:6f:11
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00065fff...
Compressed 417328 bytes to 290299...
Wrote 417328 bytes (290299 compressed) at 0x00000000 in 7.2 seconds (effective 465.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[10:43:32]nrr\x92`p\xf2n0r\x8cbbn\xe2nb\xc4\xf2nn'l`\x90nn$`nr\x8e\x92\x92nrl`r\x92\x92nrl`\x9c\x8c$$rl`\xfc\x82n\x92`[I][logger:214]: Log initialized
[10:43:32][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[10:43:32][I][app:029]: Running through setup()...
[10:43:32][I][i2c.arduino:161]: Performing I2C bus recovery
[10:43:32][C][bme280.sensor:082]: Setting up BME280...
[10:43:32][E][component:112]: Component bme280.sensor was marked as failed.
[10:43:32][C][wifi:037]: Setting up WiFi...
[10:43:32][D][wifi:373]: Starting scan...
[10:43:38][D][wifi:388]: Found networks:
[10:43:38][I][wifi:431]: - 'PASA' [redacted]▂▄▆█
[10:43:38][D][wifi:433]:     Channel: 6
[10:43:38][D][wifi:434]:     RSSI: -78 dB
[10:43:38][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:43:38][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:43:38][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:43:38][I][wifi:248]: WiFi Connecting to 'PASA'...
[10:43:39][I][wifi:505]: WiFi Connected!
[10:43:39][C][wifi:353]:   Local MAC: 40:91:51:51:6F:11
[10:43:39][C][wifi:354]:   SSID: [redacted]
[10:43:40][C][wifi:355]:   IP Address: 192.168.178.60
[10:43:40][C][wifi:356]:   BSSID: [redacted]
[10:43:40][C][wifi:358]:   Hostname: 'office-test2'
[10:43:40][C][wifi:360]:   Signal strength: -78 dB ▂▄▆█
[10:43:40][C][wifi:364]:   Channel: 6
[10:43:40][C][wifi:365]:   Subnet: 255.255.255.0
[10:43:40][C][wifi:366]:   Gateway: 192.168.178.1
[10:43:40][C][wifi:367]:   DNS1: 192.168.178.1
[10:43:40][C][wifi:368]:   DNS2: 0.0.0.0
[10:43:40][D][wifi:514]: Disabling AP...
[10:43:40][C][ota:085]: Over-The-Air Updates:
[10:43:40][C][ota:086]:   Address: office-test2.local:8266
[10:43:40][C][ota:089]:   Using Password.
[10:43:40][C][api:025]: Setting up Home Assistant API server...
[10:43:40][I][app:062]: setup() finished successfully!
[10:43:40][I][app:102]: ESPHome version 2022.2.6 compiled on Mar 12 2022, 10:32:33
[10:43:40][C][wifi:491]: WiFi:
[10:43:40][C][wifi:353]:   Local MAC: 40:91:51:51:6F:11
[10:43:40][C][wifi:354]:   SSID: [redacted]
[10:43:40][C][wifi:355]:   IP Address: 192.168.178.60
[10:43:40][C][wifi:356]:   BSSID: [redacted]
[10:43:40][C][wifi:358]:   Hostname: 'office-test2'
[10:43:40][C][wifi:360]:   Signal strength: -78 dB ▂▄▆█
[10:43:40][C][wifi:364]:   Channel: 6
[10:43:40][C][wifi:365]:   Subnet: 255.255.255.0
[10:43:40][C][wifi:366]:   Gateway: 192.168.178.1
[10:43:40][C][wifi:367]:   DNS1: 192.168.178.1
[10:43:40][C][wifi:368]:   DNS2: 0.0.0.0
[10:43:40][C][logger:233]: Logger:
[10:43:40][C][logger:234]:   Level: DEBUG
[10:43:40][C][logger:235]:   Log Baud Rate: 115200
[10:43:40][C][logger:236]:   Hardware UART: UART0
[10:43:40][C][i2c.arduino:038]: I2C Bus:
[10:43:40][C][i2c.arduino:039]:   SDA Pin: GPIO4
[10:43:40][C][i2c.arduino:040]:   SCL Pin: GPIO5
[10:43:40][C][i2c.arduino:041]:   Frequency: 50000 Hz
[10:43:40][C][i2c.arduino:044]:   Recovery: bus successfully recovered
[10:43:40][I][i2c.arduino:054]: Results from i2c bus scan:
[10:43:40][I][i2c.arduino:060]: Found i2c device at address 0x76
[10:43:40][C][bme280.sensor:143]: BME280:
[10:43:40][C][bme280.sensor:144]:   Address: 0x77
[10:43:40][E][bme280.sensor:147]: Communication with BME280 failed!
[10:43:40][C][bme280.sensor:156]:   IIR Filter: OFF
[10:43:40][C][bme280.sensor:157]:   Update Interval: 10.0s
[10:43:40][C][bme280.sensor:159]:   Temperature 'BME280 Temperature'
[10:43:40][C][bme280.sensor:159]:     Device Class: 'temperature'
[10:43:40][C][bme280.sensor:159]:     State Class: 'measurement'
[10:43:40][C][bme280.sensor:159]:     Unit of Measurement: '°C'
[10:43:40][C][bme280.sensor:159]:     Accuracy Decimals: 1
[10:43:40][C][bme280.sensor:160]:     Oversampling: 16x
[10:43:40][C][bme280.sensor:161]:   Pressure 'BME280 Pressure'
[10:43:40][C][bme280.sensor:161]:     Device Class: 'pressure'
[10:43:40][C][bme280.sensor:161]:     State Class: 'measurement'
[10:43:40][C][bme280.sensor:161]:     Unit of Measurement: 'hPa'
[10:43:40][C][bme280.sensor:161]:     Accuracy Decimals: 1
[10:43:40][C][bme280.sensor:162]:     Oversampling: 16x
[10:43:40][C][bme280.sensor:163]:   Humidity 'BME280 Humidity'
[10:43:40][C][bme280.sensor:163]:     Device Class: 'humidity'
[10:43:40][C][bme280.sensor:163]:     State Class: 'measurement'
[10:43:40][C][bme280.sensor:163]:     Unit of Measurement: '%'
[10:43:40][C][bme280.sensor:163]:     Accuracy Decimals: 1
[10:43:40][C][bme280.sensor:164]:     Oversampling: 16x
[10:43:40][C][captive_portal:144]: Captive Portal:
[10:43:40][C][mdns:084]: mDNS:
[10:43:40][C][mdns:085]:   Hostname: office-test2
[10:43:40][C][ota:085]: Over-The-Air Updates:
[10:43:40][C][ota:086]:   Address: office-test2.local:8266
[10:43:40][C][ota:089]:   Using Password.
[10:43:40][C][api:138]: API Server:
[10:43:40][C][api:139]:   Address: office-test2.local:6053
[10:43:40][C][api:143]:   Using noise encryption: NO
[10:43:50][D][api:102]: Accepted 192.168.178.51
[10:43:50][W][api.connection:083]: Home Assistant 2022.3.1 (192.168.178.51): Connection closed
```

jsuanet, 2022-03-12:
```
[10:43:40][I][i2c.arduino:054]: Results from i2c bus scan:
[10:43:40][I][i2c.arduino:060]: Found i2c device at address 0x76
[10:43:40][C][bme280.sensor:143]: BME280:
[10:43:40][C][bme280.sensor:144]:   Address: 0x77
```

The i2c scan says it found a device at address 0x76. In the yaml you specified address 0x77 for the bme sensor, what happens if you change that to 0x76

paulonHA, 2022-03-12:
Thanks a lot. For whatever reason this was not working yesterday when I tried. I probably made a mistake somewhere. Thanks again

closed 2022-03-12 by paulonHA