esphome/issues #3132: display on/off colors inconsistently inverted

quentinmit, 2022-03-12:
### The problem

The different display-related components in ESPHome seem to disagree about whether "on" is black or white, which makes it hard to reason about what's going to happen and even harder to write generic code. I've been running into this when trying to use the `inkplate6` component.

For example, the `image` component translates black pixels in `BINARY` images into `1` which is then rendered using `COLOR_ON` (white), but if you change the image type to `GRAYSCALE` suddenly black pixels become `COLOR_OFF` (black). This means that `BINARY` images draw inverted by default on `inkplate6` displays.

I suspect this is a side effect of some of the early displays that ESPHome supported being OLEDs, where you commonly want to draw white-on-black. I'm not sure how to best fix this with a minimum of backwards incompatibility, but here would be my two suggestions:

1. Treat all displays as "normal" UI displays, with a white background and black foreground default for all drawing operations. Add an `inverted_colors` configuration option to displays that inverts all the colors at draw time, defaulting to true for OLED displays so that "black" renders as white.
2. Add new methods to `DisplayBuffer` called `get_foreground_color()` and `get_background_color()` which return the correct colors for the given display type, and add new color constants (`COLOR_AUTO_FG` and `COLOR_AUTO_BG`?)  which can be used as the new default values for the methods on `DisplayBuffer` to allow the defaults to draw correctly on different types of display. Since we don't (yet?) support RGBW displays or images, we can use the W component to encode the auto constants.

I'm willing to contribute the PR to do either of these things if you can advise on which one would be preferable. I think the latter might be slightly better, since it would be more backwards compatible (existing code specifying `COLOR_ON` or `COLOR_OFF` would not change behavior), but I do think it would be a larger change.

### Which version of ESPHome has the issue?

2022.3.0-dev (based off 1a8f8adc2a2b0393cd0639d0cdf3b8088df41201)

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Board

inkplate6

### Component causing the issue

display

guillempages, 2022-06-19:
I've also encountered this issue on my inkplate; if I use the draw methods (rectangle, filled_rectangle, text, ...), the colors behave as I expect (COLOR_ON = white; COLOR_OFF = black), but if I use a BINARY image, the colors are inverted (black is displayed as white and the other way around).
I also think that the issue is caused by supporting both OLED and E-Ink displays, that have different default "pixel on" behaviours, but it would be nice if at least it would work consistently *within a single device* ;-)
The issue can be worked around by specifying the "color_on" and "color_off" parameters in the "draw" method, but it would help if it would work out-of-the-box.

nielsnl68, 2022-10-28:
nock to unstall