esphome/issues #3136: stepper with uln2003 goes only clockwise (wemos D1 mini pro)

ezeerfrm, 2022-03-14:
### The problem

Hi,

I have a wemos D1 mini pro with an uln2003 to drive a stepper motor (28BYj-48).

It works well but the motor is only spinning clockwise (and doesn't spin couterclockwise) no matter if I send positive or negative values. The value logged are the one from HA (with correct positive and negative values). I also tried to write directly the target like _target: 250 or target: -250_ (instead of the lambda) but it doesn't work either.

My wiring is good because the set up works if I upload code from arduino platform (using accellstepper library).

Here is my code:

api:
  services:
    - service: control_stepper
      variables:
        target: int
      then:
        - stepper.report_position:
            id : stepper_vmc
            position: 0
        - stepper.set_target:
            id: stepper_vmc
            target: !lambda 'return target;'
        - logger.log:
            format: "The target value is %d"
            args : ['target']

#Etat de la connection
binary_sensor:
  - platform: status
    name: "espvmc_Status"
    

    
stepper:
  - platform: uln2003
    id: stepper_vmc
    pin_a: D0
    pin_b: D1
    pin_c: D2
    pin_d: D3
    max_speed: 50 steps/s
    sleep_when_done: true
    acceleration: inf
    deceleration: inf

Do yo uhave any ideas of what could go wrong?

### Which version of ESPHome has the issue?

2021.12.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.12.7

### What platform are you using?

ESP8266

### Board

wemos d1 mini pro

### Component causing the issue

stepper

### Example YAML snippet

```yaml
api:
  services:
    - service: control_stepper
      variables:
        target: int
      then:
        - stepper.report_position:
            id : stepper_vmc
            position: 0
        - stepper.set_target:
            id: stepper_vmc
            target: !lambda 'return target;'
        - logger.log:
            format: "The target value is %d"
            args : ['target']

#Etat de la connection
binary_sensor:
  - platform: status
    name: "espvmc_Status"
    

    
stepper:
  - platform: uln2003
    id: stepper_vmc
    pin_a: D0
    pin_b: D1
    pin_c: D2
    pin_d: D3
    max_speed: 50 steps/s
    sleep_when_done: true
    acceleration: inf
    deceleration: inf
```

finity69x2, 2022-04-20:
I noticed the same thing. Motor only goes one direction with either positive or negative target values.

finity69x2, 2022-04-22:
Not sure if this will help you or not but I solved the issue for mine.

I was doing some troubleshooting this issue and I realized that two wires from the nodemcu to the ULN2003 needed reversed.

My config is:

```
stepper:
  - platform: uln2003
    id: ${id} 
    pin_a: D3 #IN4 - green
    pin_b: D4 #IN3 - blue
    pin_c: D2 #IN2 - yellow
    pin_d: D1 #IN1 - orange
    max_speed: 100 steps/s
    sleep_when_done: true
```

I originally had pin_a connected to IN3 and pin_b connected to IN4 but that did the same as thie issue describes (only turned CCW when viewed from shaft end).

once I swapped them the motor will now go both directions.

You could try that and see if it works.

closed 2022-09-18 by github-actions[bot]