esphome/issues #3138: Custom UART device code from website is broken

rfvermut, 2022-03-14:
### The problem

I copied the code from https://esphome.io/custom/uart.html to my sample project and getting
```
src\ModbusGPIO.h: In member function 'virtual void ModbusGPIOComponent::loop()':
src\ModbusGPIO.h:12:30: error: 'readString' was not declared in this scope
     String line = readString();
                              ^
src\ModbusGPIO.h:13:22: error: 'parseInt' was not declared in this scope
     int i = parseInt();
                      ^
*** [.pioenvs\irrigator\src\main.cpp.o] Error 1
```

on functions explained as `// Use Arduino API to read data, for example`

My understanding is that sample code wouldn't work because UARTDevice is not exposing `readString` or `parseInt` and never did. I'm I right?

### Which version of ESPHome has the issue?

2022.1.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

d1_mini

DysinLarshee, 2022-06-08:
Same problem here, still can not find a way to be able to read Uart data from ESPHome

mc0110, 2022-10-11:
Hi,
I can't use the example from the documentation either. What are we doing wrong or is there a bug that needs to be solved first?

hlzr, 2023-01-14:
I'm having the same issue.

ssieb, 2023-01-15:
I can't find any indication that `readString` was ever supported.  It shouldn't be used anyway.  You should really make a proper component anyway instead of using this "custom" method.
If the docs need fixing, you should create an issue in the docs repo, this isn't an esphome bug.

rfvermut, 2023-01-15:
> If the docs need fixing, you should create an issue in the docs repo, this isn't an esphome bug.

Meanwhile in docs repo:
[image]

ssieb, 2023-01-15:
Ok... :man_shrugging:

rfvermut, 2023-01-15:
Frankly that probably is an easy fix. Include is missing, or functions need to be declared, or referenced from a class. I just don't know code good enough to propose one.

AndrewXuereb, 2023-01-28:
Because this cost me entirely too long of late night madness, as I had based a custom module on this code, AFAIK, the fix is to change:
```
    String line = readString();
    int i = parseInt();```
```
to
```
    String line = Serial.readString();
    int i = Serial.parseInt();
```
Seems to make it compile fine, at least.

ssieb, 2023-01-29:
But it's probably wrong for what you want to do.  And `readString()` should not be used in esphome.

rfvermut, 2023-01-30:
> But it's probably wrong for what you want to do. And `readString()` should not be used in esphome.

Do you know what SHOULD be used? You sound like you know the esphome ways. Let's make this doc snippet useful together.

As I understand the intention of the code is to read something from UART in simple way. And (quote) `pretty much use almost all Arduino-based code`. 

Presumably whatever works for Arduino `Serial` is supposed to work for `UART`.

So either the line above is wrong and we need to change the text into "use low level read()/write() to access uart data" or "parseInt()" should work somehow in the code example.