esphome/issues #3139: Lillygo T5 4.7" Touchscreen component doesn't work on battery power

Cadair, 2022-03-15:
### The problem

The touchscreen component works well on the lilygo T5 when it is connected to USB power, however, when connected to a 3.7 V battery the touchscreen does not work and `[E][lilygo_t5_47.touchscreen:060]: Failed to communicate!` is logged for each touch.

The device also fails to boot with the touchscreen connected unless it's connected to USB.

This was first reported [by @Plawasan](https://github.com/esphome/esphome/pull/3084#issuecomment-1032839601)

### Which version of ESPHome has the issue?

2022.2.6

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.2.9

### What platform are you using?

ESP32

### Board

Lillygo T5 4.7"

### Component causing the issue

touchscreen

### Example YAML snippet

```yaml
substitutions:
  esp_name: ESP Display #Device Name
  esp_hostname: touchdisplay
  ha_climate_component: climate.central_heating

esphome:
  name: ${esp_hostname}
  platformio_options:
    upload_speed: 1500000

esp32:
  board: esp32dev
  framework:
    type: arduino

spi:
  clk_pin: 18
  mosi_pin: 23

external_components:
  - source: github://ashald/esphome@lilygo-t5-47
    components:
      - lilygo_t5_47

touchscreen:
  - platform: lilygo_t5_47
    id: lilygo_touchscreen
    interrupt_pin: GPIO13
    on_touch:
      - logger.log:
          format: Touch at (%d, %d)
          args: ["touch.x", "touch.y"]

i2c:
  sda: 15
  scl: 14
  scan: True
  
logger:
api:
ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

binary_sensor:
  - platform: touchscreen
    touchscreen_id: lilygo_touchscreen
    id: touch_key1
    x_min: 0
    x_max: 100
    y_min: 0
    y_max: 100
    on_press:
      - logger.log: "Touched"

font:
  - file: "SourceSansPro-Regular.ttf"
    id: font_footer
    size: 28

display:
  - platform: lilygo_t5_47
    id: t5_display
    rotation: 90
    update_interval: 30s
    lambda: |-
      #define xres 540 
      #define yres 960
      #define x_pad 10 // border padding
      #define y_pad 10 // border padding      

      it.printf(xres-x_pad, yres-y_pad/2, id(font_footer), TextAlign::BASELINE_RIGHT, "Hello World");
```

kasdk3, 2022-05-31:
more info here. https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/55

Cadair, 2022-06-02:
@jesserockz as the person who wrote the touchscreen component, this is probably something you could fix quickly and it would take me a week. If you don't have time to fix it do you have any pointers on where to start? Thanks!

Cadair, 2022-11-05:
This has largely been sorted over at https://github.com/esphome/feature-requests/issues/1109

closed 2022-11-05 by Cadair